ioBroker/ioBroker.modbus #159: modbus instance doesn't read input register

derlbear, 2021-08-09:
Hi, my modbus instance on a raspberry pi 4 doesn't read all input registers.
The adapter is connected to a systemair ventilation unit.

I entered the input registers...
[image]

... but the modbus instance doesn't pull it...
[image]

.. so the value stays 0
[image]

When read the input register via qModMaster I receive the right value.

What is my mistake?

GermanBluefox, 2021-08-31:
Please try the 3.4.12. It was some issue with sorting.

derlbear, 2021-09-13:
Thank you. That solved my problem.

closed 2021-09-13 by derlbear