ioBroker/ioBroker.modbus #163: Vers. 3.4.11 : Formula SF Einträge werden nicht ausgewertet  x * Math.pow(10, sf['40065'])

Glasfaser-io, 2021-08-20:
In bestehenden Instanzen werden Änderungen in den Formula / SF übernommen und auch angewandt,
nur in **neu erstellten Instanzen** , werden die Formula Einträge nicht genommen .

Dieses wurde hier im Thread festgestellt .

https://forum.iobroker.net/topic/6403/solaredge-adapter-photovoltaikanlage/242

`_address	name	description	unit	type	len	factor	offset	formula	role	room	poll	wp	cw	isScale
40083	I_AC_Leistung	AC-Leistungswert	W	uint16be	1	1	0	x * Math.pow(10, sf['40084'])	value		true	false		false
40084	I_AC_Leistung_SF	AC-Leistung Skalierungsfaktor	SF	int16be	1	1	0		value		true	false		true
40100	I_DC_Leistung	DC-Leistungswert	W	uint16be	1	1	0	x * Math.pow(10, sf['40101'])	value		true	false		false
40101	I_DC_Leistung_SF	DC-Leistung Skalierungsfaktor	SF	int16be	1	1	0		value		true	false		true
40103	I_Temp_Kühler	Kühlkörpertemperatur	°C	uint16be	1	1	0	x * Math.pow(10, sf['40106'])	value		true	false		false
40106	I_Temp_SF	Kühlkörpertemperatur Skalierungsfaktor	SF	int16be	1	1	0		value		true	false		true
40206	M_AC_Power	Total Real Power (sum of active phases)	W	int16be	1	1	0	x * Math.pow(10, sf['40210'])	value		true	false		false
40210	M_AC_Power_SF	AC Real Power Scale Factor	SF	int16be	1	1	0		value		true	false		true`

[image]

**Versions:**  
 - Adapter version: 3.4.11
 - JS-Controller version: 3.3.15
 - Node version: V12.22.4 / NPM 6.14.4
 - Operating system: Linux
 - -Admin: 5.1.25

Glasfaser-io, 2021-08-28:
> 2021-08-28 21:29:45.631  - debug: modbus.5 (25121) Redis Objects: Use Redis connection: 127.0.0.1:9001
2021-08-28 21:29:45.713  - debug: modbus.5 (25121) Objects client ready ... initialize now
2021-08-28 21:29:45.716  - debug: modbus.5 (25121) Objects create System PubSub Client
2021-08-28 21:29:45.718  - debug: modbus.5 (25121) Objects create User PubSub Client
2021-08-28 21:29:45.720  - debug: modbus.5 (25121) Objects client initialize lua scripts
2021-08-28 21:29:45.779  - debug: modbus.5 (25121) Objects connected to redis: 127.0.0.1:9001
2021-08-28 21:29:45.792  - debug: modbus.5 (25121) objectDB connected
2021-08-28 21:29:45.796  - debug: modbus.5 (25121) Redis States: Use Redis connection: 127.0.0.1:9000
2021-08-28 21:29:45.910  - debug: modbus.5 (25121) States create System PubSub Client
2021-08-28 21:29:45.918  - debug: modbus.5 (25121) States create User PubSub Client
2021-08-28 21:29:46.012  - debug: modbus.5 (25121) States connected to redis: 127.0.0.1:9000
2021-08-28 21:29:46.014  - debug: modbus.5 (25121) statesDB connected
2021-08-28 21:29:47.327  - debug: modbus.5 (25121) Plugin sentry Initialize Plugin (enabled=true)
2021-08-28 21:29:48.103  - info: modbus.5 (25121) starting. Version 3.4.11 in /opt/iobroker/node_modules/iobroker.modbus, node: v14.17.5, js-controller: 3.3.15
2021-08-28 21:29:48.416  - debug: modbus.5 (25121) Initialize Objects for disInputs: []
2021-08-28 21:29:48.418  - debug: modbus.5 (25121) Initialize Objects for coils: []
2021-08-28 21:29:48.420  - debug: modbus.5 (25121) Initialize Objects for inputRegs: []
2021-08-28 21:29:48.420  - debug:  modbus.5 (25121) Initialize Objects for holdingRegs:  [{"_address":"40083","name":"I_AC_Leistung","description":"AC-Leistungswert","unit":"W","type":"uint16be","len":1,"factor":1,"offset":0,"formula":"x  * Math.pow(10,  sf['40084'])","role":"value","room":"","poll":true,"wp":false,"cw":false,"isScale":false,"deviceId":1,"address":82,"id":"holdingRegisters.40083_I_AC_Leistung"},{"_address":"40084","name":"I_AC_Leistung_SF","description":"AC-Leistung   Skalierungsfaktor","unit":"SF","type":"int16be","len":1,"factor":1,"offset":0,"formula":"","role":"value","room":"","poll":true,"wp":false,"cw":false,"isScale":true,"deviceId":1,"address":83,"id":"holdingRegisters.40084_I_AC_Leistung_SF"},{"_address":"40100","name":"I_DC_Leistung","description":"DC-Leistungswert","unit":"W","type":"uint16be","len":1,"factor":1,"offset":0,"formula":"x  * Math.pow(10,  sf['40101'])","role":"value","room":"","poll":true,"wp":false,"cw":false,"isScale":false,"deviceId":1,"address":99,"id":"holdingRegisters.40100_I_DC_Leistung"},{"_address":"40101","name":"I_DC_Leistung_SF","description":"DC-Leistung   Skalierungsfaktor","unit":"SF","type":"int16be","len":1,"factor":1,"offset":0,"formula":"","role":"value","room":"","poll":true,"wp":false,"cw":false,"isScale":true,"deviceId":1,"address":100,"id":"holdingRegisters.40101_I_DC_Leistung_SF"},{"_address":"40103","name":"I_Temp_Kühler","description":"Kühlkörpertemperatur","unit":"°C","type":"uint16be","len":1,"factor":1,"offset":0,"formula":"x  * Math.pow(10,  sf['40106'])","role":"value","room":"","poll":true,"wp":false,"cw":false,"isScale":false,"deviceId":1,"address":102,"id":"holdingRegisters.40103_I_Temp_Kühler"},{"_address":"40106","name":"I_Temp_SF","description":"Kühlkörpertemperatur   Skalierungsfaktor","unit":"SF","type":"int16be","len":1,"factor":1,"offset":0,"formula":"","role":"value","room":"","poll":true,"wp":false,"cw":false,"isScale":true,"deviceId":1,"address":105,"id":"holdingRegisters.40106_I_Temp_SF"},{"_address":"40206","name":"M_AC_Power","description":"Total  Real Power (sum of active  phases)","unit":"W","type":"int16be","len":1,"factor":1,"offset":0,"formula":"x  * Math.pow(10,  sf['40210'])","role":"value","room":"","poll":true,"wp":false,"cw":false,"isScale":false,"deviceId":1,"address":205,"id":"holdingRegisters.40206_M_AC_Power"},{"_address":"40210","name":"M_AC_Power_SF","description":"AC  Real Power Scale  Factor","unit":"SF","type":"int16be","len":1,"factor":1,"offset":0,"formula":"","role":"value","room":"","poll":true,"wp":false,"cw":false,"isScale":true,"deviceId":1,"address":209,"id":"holdingRegisters.40210_M_AC_Power_SF"}]
2021-08-28 21:29:48.421  - debug:  modbus.5 (25121) Add holdingRegisters.40083_I_AC_Leistung:  {"_id":"holdingRegisters.40083_I_AC_Leistung","type":"state","common":{"name":"AC-Leistungswert","role":"value","type":"number","read":true,"write":true,"def":0,"unit":"W"},"native":{"regType":"holdingRegs","address":82,"deviceId":1,"type":"uint16be","len":1,"offset":0,"factor":1,"poll":true}}
2021-08-28 21:29:48.423  - debug:  modbus.5 (25121) Add holdingRegisters.40084_I_AC_Leistung_SF:  {"_id":"holdingRegisters.40084_I_AC_Leistung_SF","type":"state","common":{"name":"AC-Leistung   Skalierungsfaktor","role":"value","type":"number","read":true,"write":true,"def":0,"unit":"SF"},"native":{"regType":"holdingRegs","address":83,"deviceId":1,"type":"int16be","len":1,"offset":0,"factor":1,"poll":true}}
2021-08-28 21:29:48.424  - debug:  modbus.5 (25121) Add holdingRegisters.40100_I_DC_Leistung:  {"_id":"holdingRegisters.40100_I_DC_Leistung","type":"state","common":{"name":"DC-Leistungswert","role":"value","type":"number","read":true,"write":true,"def":0,"unit":"W"},"native":{"regType":"holdingRegs","address":99,"deviceId":1,"type":"uint16be","len":1,"offset":0,"factor":1,"poll":true}}
[... cut ...]
2021-08-28 21:30:53.032  - debug: modbus.5 (25121) Calculation result = NaN, type = number
2021-08-28 21:30:53.033  - debug: modbus.5 (25121) Poll holdingRegs DevID(1) address 205 - 5 bytes
2021-08-28 21:30:53.050  - debug: modbus.5 (25121) Scale factor value stored from address 209 = 1123
2021-08-28 21:30:53.050  - debug: modbus.5 (25121) Input Value = -2
2021-08-28 21:30:53.052  - debug: modbus.5 (25121) Formula = x * Math.pow(10, sf['40210'])
2021-08-28 21:30:53.054  - debug: modbus.5 (25121) Calculation result = NaN, type = number

GermanBluefox, 2021-08-31:
Can you try 3.4.14?

Glasfaser-io, 2021-08-31:
Nur ein Wert wird mal gerechnet aber falsch , die anderen bleiben Null : 

.

> 2021-08-31 17:00:22.012  - debug: modbus.5 (8140) [DevID_1/holdingRegs/40206] Input Value = -2
> 2021-08-31 17:00:22.014  - debug: modbus.5 (8140) [DevID_1/holdingRegs/40206] Formula = x * Math.pow(10, sf['40210'])
> 2021-08-31 17:00:22.015  - debug: modbus.5 (8140) [DevID_1/holdingRegs/40206] Calculation result = -**2e+182,** type = 

..

[image]

..

[image]

> 2021-08-31 17:00:11.632  - info: modbus.5 (8140) starting. Version 3.4.14 in /opt/iobroker/node_modules/iobroker.modbus, node: v14.17.5, js-controller: 3.3.15
> 2021-08-31 17:00:14.398  - debug: modbus.5 (8140) Initialize Objects for disInputs: []
> 2021-08-31 17:00:14.400  - debug: modbus.5 (8140) Initialize Objects for coils: []
> 2021-08-31 17:00:14.400  - debug: modbus.5 (8140) Initialize Objects for inputRegs: []
> 2021-08-31 17:00:14.401  - debug:  modbus.5 (8140) Initialize Objects for holdingRegs:  [{"_address":"40083","name":"I_AC_Leistung","description":"AC-Leistungswert","unit":"W","type":"uint16be","len":1,"factor":1,"offset":0,"formula":"x  * Math.pow(10,  sf['40084'])","role":"value","room":"","poll":true,"wp":false,"cw":false,"isScale":false,"deviceId":1,"address":82,"id":"holdingRegisters.40083_I_AC_Leistung"},{"_address":"40084","name":"I_AC_Leistung_SF","description":"AC-Leistung   Skalierungsfaktor","unit":"SF","type":"int16be","len":1,"factor":1,"offset":0,"formula":"","role":"value","room":"","poll":true,"wp":false,"cw":false,"isScale":true,"deviceId":1,"address":83,"id":"holdingRegisters.40084_I_AC_Leistung_SF"},{"_address":"40100","name":"I_DC_Leistung","description":"DC-Leistungswert","unit":"W","type":"uint16be","len":1,"factor":1,"offset":0,"formula":"x  * Math.pow(10,  sf['40101'])","role":"value","room":"","poll":true,"wp":false,"cw":false,"isScale":false,"deviceId":1,"address":99,"id":"holdingRegisters.40100_I_DC_Leistung"},{"_address":"40101","name":"I_DC_Leistung_SF","description":"DC-Leistung   Skalierungsfaktor","unit":"SF","type":"int16be","len":1,"factor":1,"offset":0,"formula":"","role":"value","room":"","poll":true,"wp":false,"cw":false,"isScale":true,"deviceId":1,"address":100,"id":"holdingRegisters.40101_I_DC_Leistung_SF"},{"_address":"40103","name":"I_Temp_Kühler","description":"Kühlkörpertemperatur","unit":"°C","type":"uint16be","len":1,"factor":1,"offset":0,"formula":"x  * Math.pow(10,  sf['40106'])","role":"value","room":"","poll":true,"wp":false,"cw":false,"isScale":false,"deviceId":1,"address":102,"id":"holdingRegisters.40103_I_Temp_Kühler"},{"_address":"40106","name":"I_Temp_SF","description":"Kühlkörpertemperatur   Skalierungsfaktor","unit":"SF","type":"int16be","len":1,"factor":1,"offset":0,"formula":"","role":"value","room":"","poll":true,"wp":false,"cw":false,"isScale":true,"deviceId":1,"address":105,"id":"holdingRegisters.40106_I_Temp_SF"},{"_address":"40206","name":"M_AC_Power","description":"Total  Real Power (sum of active  phases)","unit":"W","type":"int16be","len":1,"factor":1,"offset":0,"formula":"x  * Math.pow(10,  sf['40210'])","role":"value","room":"","poll":true,"wp":false,"cw":false,"isScale":false,"deviceId":1,"address":205,"id":"holdingRegisters.40206_M_AC_Power"},{"_address":"40210","name":"M_AC_Power_SF","description":"AC  Real Power Scale  Factor","unit":"SF","type":"int16be","len":1,"factor":1,"offset":0,"formula":"","role":"value","room":"","poll":true,"wp":false,"cw":false,"isScale":true,"deviceId":1,"address":209,"id":"holdingRegisters.40210_M_AC_Power_SF"}]
> 2021-08-31 17:00:14.402  - debug:  modbus.5 (8140) Add holdingRegisters.40083_I_AC_Leistung:  {"_id":"holdingRegisters.40083_I_AC_Leistung","type":"state","common":{"name":"AC-Leistungswert","role":"value","type":"number","read":true,"write":true,"def":0,"unit":"W"},"native":{"regType":"holdingRegs","address":82,"deviceId":1,"type":"uint16be","len":1,"offset":0,"factor":1,"poll":true}}
> 2021-08-31 17:00:14.403  - debug:  modbus.5 (8140) Add holdingRegisters.40084_I_AC_Leistung_SF:  {"_id":"holdingRegisters.40084_I_AC_Leistung_SF","type":"state","common":{"name":"AC-Leistung   Skalierungsfaktor","role":"value","type":"number","read":true,"write":true,"def":0,"unit":"SF"},"native":{"regType":"holdingRegs","address":83,"deviceId":1,"type":"int16be","len":1,"offset":0,"factor":1,"poll":true}}
> 2021-08-31 17:00:14.403  - debug:  modbus.5 (8140) Add holdingRegisters.40100_I_DC_Leistung:  {"_id":"holdingRegisters.40100_I_DC_Leistung","type":"state","common":{"name":"DC-Leistungswert","role":"value","type":"number","read":true,"write":true,"def":0,"unit":"W"},"native":{"regType":"holdingRegs","address":99,"deviceId":1,"type":"uint16be","len":1,"offset":0,"factor":1,"poll":true}}
> 2021-08-31 17:00:14.404  - debug:  modbus.5 (8140) Add holdingRegisters.40101_I_DC_Leistung_SF:  {"_id":"holdingRegisters.40101_I_DC_Leistung_SF","type":"state","common":{"name":"DC-Leistung   Skalierungsfaktor","role":"value","type":"number","read":true,"write":true,"def":0,"unit":"SF"},"native":{"regType":"holdingRegs","address":100,"deviceId":1,"type":"int16be","len":1,"offset":0,"factor":1,"poll":true}}
> 2021-08-31 17:00:14.405  - debug:  modbus.5 (8140) Add holdingRegisters.40103_I_Temp_Kühler:  {"_id":"holdingRegisters.40103_I_Temp_Kühler","type":"state","common":{"name":"Kühlkörpertemperatur","role":"value","type":"number","read":true,"write":true,"def":0,"unit":"°C"},"native":{"regType":"holdingRegs","address":102,"deviceId":1,"type":"uint16be","len":1,"o [... cut ...]
> 2021-08-31 17:00:22.010  - debug: modbus.5 (8140) [DevID_1/holdingRegs] Poll address 205 DONE
> 2021-08-31 17:00:22.011  - debug: modbus.5 (8140) [DevID_1/holdingRegs/40210] Scale factor value stored from this address = 182
> 2021-08-31 17:00:22.012  - debug: modbus.5 (8140) [DevID_1/holdingRegs/40206] Input Value = -2
> 2021-08-31 17:00:22.014  - debug: modbus.5 (8140) [DevID_1/holdingRegs/40206] Formula = x * Math.pow(10, sf['40210'])
> 2021-08-31 17:00:22.015  - debug: modbus.5 (8140) [DevID_1/holdingRegs/40206] Calculation result = -2e+182, type = number

GermanBluefox, 2021-09-02:
> 2021-08-31 17:00:22.014 - debug: modbus.5 (8140) [DevID_1/holdingRegs/40206] Formula = x * Math.pow(10, sf['40210'])

`-2 * Math.pow(10, 196) = -2e+196` what is wrong?

Glasfaser-io, 2021-09-02:
[image]
.

Er schreibt die Formula in den Datenpunkt 40206 : **-2e+196 W** ,  es wäre aber das Ergebnis  11 W ( Watt ) zu sehen .
.
Wie hier im Beispiel **-12 W** ,( Negativwert  Import ins Stromnetz )

[image]

Apollon77, 2021-09-02:
Was Bluefox meinte: Das ist nicht die Formel ... das ist der Wert! -2* 10hoch 196 wird in der Informatik als -2e+196 dargestellt. Das log beschreibt es genau. der definierte Scaling Faktor ist 196 ... denke ist also in sf['40210'] ... damit kommt genau das raus. Was ist daran jetzt falsch? Hast Du vllt hier Datenfelder vertauscht?

Glasfaser-io, 2021-09-02:
Ich sehe da nichts vertauscht , ist so wie es sein sollte

> 40206	M_AC_Power	Total Real Power (sum of active phases)	W	int16be	1	1	0	x * Math.pow(10, sf['40210'])	value		true	false		false
> 40210	M_AC_Power_SF	AC Real Power Scale Factor	SF	int16be	1	1	0		value		true	false		true

Wie schon in der Issues Eröffnung geschrieben , funktioniert die Formula bei **vorhandenen Modbus Instanzen** , also wenn man sie nachträglich in Formula einträgt wird auch richtig berechnet .

Bei einer **neu erstelllten** Modbus Instanz wird die Formula nicht berücksichtigt .

Jetzt  in der neuen Version / Änderung  wird ein Wert wie oben geschrieben **-2e+196** , also nicht die Wattzahl. 

Ich hoffe ich habe es verständlich nochmal umschrieben !?

Apollon77, 2021-09-02:
Das log sagt aber das der angegebene SScalting Faktor 196 ist und 10 hoch 196 ist halt eine grosse Zahl ...

> 2021-08-31 17:00:20.359 - debug: modbus.5 (8140) [DevID_1/holdingRegs/40210] Scale factor value stored from this address = 184
2021-08-31 17:00:20.360 - debug: modbus.5 (8140) [DevID_1/holdingRegs/40206] Input Value = -2
2021-08-31 17:00:20.362 - debug: modbus.5 (8140) [DevID_1/holdingRegs/40206] Formula = x * Math.pow(10, sf['40210'])
2021-08-31 17:00:20.364 - debug: modbus.5 (8140) [DevID_1/holdingRegs/40206] Calculation result = -2e+184, type = number

Das zeigt doch alles. Dann ist ggf irgendein setting anders eil es scheint ja dann der Wert vom holding register 40210 mit 184 falsch zu sein!

Glasfaser-io, 2021-09-02:
Ich sprech es mal im Forums-Thread an , ob es anderer mal mit seinen Einstellungen testen kann  , wie es sich bei Ihnen verhält .

https://forum.iobroker.net/topic/6403/solaredge-adapter-photovoltaikanlage/253

Glasfaser-io, 2021-09-02:
Liegt wohl jetzt nur noch an meiner Test Instanz .
Ist alles in Ordnung , DANKE  👍 

Hier die Rückmeldung :  

> @soeni sagte in [Solaredge Adapter \(Photovoltaikanlage\)](/post/673238):
> > @glasfaser Hi, habs grad mal von Github instaliert und funktioniert bei mir jetzt. Vielen Dank. Habs erst mal mit dem Testadapter getestet. muss jetzt umziehen auf den richtigen Adapter.
>