ioBroker/ioBroker.modbus #168: Precision problem with GMC2289 electricity meter

HOHHK, 2021-09-02:
I use a GMC2289 electricity meter for power and energy measurement purposes (connected through Modbus-TCP --> TCP-to-RS485-converter-->GMC2289).  This meter incorporates a real-time clock which can be set and read using Modbus registers. Date and time require a 64-bit-unsigned long value (4 modbus registers) for read and write to be written in a block.   From these 64 bits, 56 bits are used (one byte is always 0x00).  

When writing/reading such 56bit wide values  using a script to/from the meter, the values are not stable. I assume this is due to the number conversion to floating point when calling the setState() function. JS supports a 52 bit wide matissa only while the meter requires 56 bits to be stable. Any kind of rounding results in wrong date/time information transmitted to the meter. 

It would be a lot easier if date/time could be packed in a script to e.g. an ArrayBuffer[8] which is then submitted to the Modbus adapter. 

In the Modbus adapter, such buffer could be unpacked and copied to or read from the Modbus registers.

Apollon77, 2021-09-02:
Sounds like we need modbus support for binary-states becaue all other datatypes are not supporting binary data

GermanBluefox, 2021-09-02:
We should implement support of arrays, like string.

Apollon77, 2021-09-02:
and array will then be converted into an array ... should also work