ioBroker/ioBroker.modbus #170: (20621) Calculation: eval(value) not possible: ReferenceError: value is not defined

Rockclimber87, 2021-09-06:
Hey Team,

I am reading out my E3DC battery system by modbus.
In the log file I am getting following warn:

<html><body>

  | (20621) Calculation: eval(value) not possible: ReferenceError: value is not defined
-- | --

</body>
</html>

This happens every second few times.
I am a really beginner in the iobroker system so I have no idea where this comes from.
How can I send the log file here?

Thanks for your help in advance.

Apollon77, 2021-09-06:
Please provide more log. I see two options:

1.) a Modbus formula is used ... then the plaeholder is "x" according to https://github.com/ioBroker/ioBroker.modbus#formula and not "value"
2.) YOu use an Alias with a Calculation function ... then the placeholder is "val" and not "value"

What is it?

Rockclimber87, 2021-09-06:
How can I provide you more info?
Where can I find the log?

Sorry I am a total newbee...

Apollon77, 2021-09-06:
The question is if you used a formula or an alias with a formula and used" value" there somewhere?

Else please enable debug log in modbus adapter (Admin - Instances - Expertmode - set Loglevel to "Debug") and post a more detailed log "around" the place when it happens

Rockclimber87, 2021-09-06:
<html><body>

modbus.0 | 2021-09-06 14:20:17.062 | warn | (27634) Calculation: eval(value) not possible: ReferenceError: value is not defined
-- | -- | -- | --
modbus.0 | 2021-09-06 14:20:17.061 | debug | (27634) Formula = value
modbus.0 | 2021-09-06 14:20:17.060 | debug | (27634) Input Value = -1802
modbus.0 | 2021-09-06 14:20:17.051 | debug | (27634) Poll holdingRegs DevID(1) address 75 - 2 bytes
modbus.0 | 2021-09-06 14:20:17.014 | warn | (27634) Calculation: eval(value) not possible: ReferenceError: value is not defined
modbus.0 | 2021-09-06 14:20:17.013 | debug | (27634) Formula = value
modbus.0 | 2021-09-06 14:20:17.012 | debug | (27634) Input Value = 0
modbus.0 | 2021-09-06 14:20:17.011 | warn | (27634) Calculation: eval(value) not possible: ReferenceError: value is not defined
modbus.0 | 2021-09-06 14:20:17.010 | debug | (27634) Formula = value
modbus.0 | 2021-09-06 14:20:17.009 | debug | (27634) Input Value = 418
modbus.0 | 2021-09-06 14:20:17.008 | warn | (27634) Calculation: eval(value) not possible: ReferenceError: value is not defined
modbus.0 | 2021-09-06 14:20:17.007 | debug | (27634) Formula = value
modbus.0 | 2021-09-06 14:20:17.006 | debug | (27634) Input Value = 719
modbus.0 | 2021-09-06 14:20:17.005 | warn | (27634) Calculation: eval(value) not possible: ReferenceError: value is not defined
modbus.0 | 2021-09-06 14:20:17.003 | debug | (27634) Formula = value
modbus.0 | 2021-09-06 14:20:17.002 | debug | (27634) Input Value = 0
modbus.0 | 2021-09-06 14:20:17.001 | warn | (27634) Calculation: eval(value) not possible: ReferenceError: value is not defined
modbus.0 | 2021-09-06 14:20:17.000 | debug | (27634) Formula = value
modbus.0 | 2021-09-06 14:20:16.999 | debug | (27634) Input Value = 202
modbus.0 | 2021-09-06 14:20:16.998 | warn | (27634) Calculation: eval(value) not possible: ReferenceError: value is not defined
modbus.0 | 2021-09-06 14:20:16.997 | debug | (27634) Formula = value
modbus.0 | 2021-09-06 14:20:16.996 | debug | (27634) Input Value = 193
modbus.0 | 2021-09-06 14:20:16.994 | warn | (27634) Calculation: eval(value) not possible: ReferenceError: value is not defined
modbus.0 | 2021-09-06 14:20:16.993 | debug | (27634) Formula = value
modbus.0 | 2021-09-06 14:20:16.992 | debug | (27634) Input Value = 4
modbus.0 | 2021-09-06 14:20:16.991 | warn | (27634) Calculation: eval(value) not possible: ReferenceError: value is not defined
modbus.0 | 2021-09-06 14:20:16.990 | debug | (27634) Formula = value
modbus.0 | 2021-09-06 14:20:16.989 | debug | (27634) Input Value = 2
modbus.0 | 2021-09-06 14:20:16.988 | warn | (27634) Calculation: eval(value) not possible: ReferenceError: value is not defined
modbus.0 | 2021-09-06 14:20:16.986 | debug | (27634) Formula = value
modbus.0 | 2021-09-06 14:20:16.985 | debug | (27634) Input Value = 98
modbus.0 | 2021-09-06 14:20:16.984 | warn | (27634) Calculation: eval(value) not possible: ReferenceError: value is not defined
modbus.0 | 2021-09-06 14:20:16.981 | debug | (27634) Formula = value
modbus.0 | 2021-09-06 14:20:16.980 | debug | (27634) Input Value = 25349
modbus.0 | 2021-09-06 14:20:16.979 | warn | (27634) Calculation: eval(value) not possible: ReferenceError: value is not defined
modbus.0 | 2021-09-06 14:20:16.978 | debug | (27634) Formula = value
modbus.0 | 2021-09-06 14:20:16.977 | debug | (27634) Input Value = -3784
modbus.0 | 2021-09-06 14:20:16.976 | warn | (27634) Calculation: eval(value) not possible: ReferenceError: value is not defined
modbus.0 | 2021-09-06 14:20:16.974 | debug | (27634) Formula = value
modbus.0 | 2021-09-06 14:20:16.973 | debug | (27634) Input Value = 252
modbus.0 | 2021-09-06 14:20:16.971 | warn | (27634) Calculation: eval(value) not possible: ReferenceError: value is not defined
modbus.0 | 2021-09-06 14:20:16.969 | debug | (27634) Formula = value
modbus.0 | 2021-09-06 14:20:16.967 | debug | (27634) Input Value = 0
modbus.0 | 2021-09-06 14:20:16.965 | warn | (27634) Calculation: eval(value) not possible: ReferenceError: value is not defined
modbus.0 | 2021-09-06 14:20:16.963 | debug | (27634) Formula = value
modbus.0 | 2021-09-06 14:20:16.961 | debug | (27634) Input Value = 2234
modbus.0 | 2021-09-06 14:20:16.959 | warn | (27634) Calculation: eval(value) not possible: ReferenceError: value is not defined
modbus.0 | 2021-09-06 14:20:16.957 | debug | (27634) Formula = value
modbus.0 | 2021-09-06 14:20:16.956 | debug | (27634) Input Value = 258
modbus.0 | 2021-09-06 14:20:16.928 | debug | (27634) Poll holdingRegs DevID(1) address 1 - 100 bytes
modbus.0 | 2021-09-06 14:20:16.924 | debug | (27634) Poll device 1
modbus.0 | 2021-09-06 14:20:01.854 | warn | (27634) Calculation: eval(value) not possible: ReferenceError: value is not defined
modbus.0 | 2021-09-06 14:20:01.851 | debug | (27634) Formula = value
modbus.0 | 2021-09-06 14:20:01.850 | debug | (27634) Input Value = -1810
modbus.0 | 2021-09-06 14:20:01.823 | debug | (27634) Poll holdingRegs DevID(1) address 75 - 2 bytes
modbus.0 | 2021-09-06 14:20:01.780 | warn | (27634) Calculation: eval(value) not possible: ReferenceError: value is not defined
modbus.0 | 2021-09-06 14:20:01.780 | debug | (27634) Formula = value
modbus.0 | 2021-09-06 14:20:01.779 | debug | (27634) Input Value = 0
modbus.0 | 2021-09-06 14:20:01.778 | warn | (27634) Calculation: eval(value) not possible: ReferenceError: value is not defined
modbus.0 | 2021-09-06 14:20:01.778 | debug | (27634) Formula = value
modbus.0 | 2021-09-06 14:20:01.777 | debug | (27634) Input Value = 418
[... cut ...]
modbus.0 | 2021-09-06 14:19:59.233 | debug | (27634) Objects client initialize lua scripts
modbus.0 | 2021-09-06 14:19:59.231 | debug | (27634) Objects create User PubSub Client
modbus.0 | 2021-09-06 14:19:59.229 | debug | (27634) Objects create System PubSub Client
modbus.0 | 2021-09-06 14:19:59.224 | debug | (27634) Objects client ready ... initialize now
modbus.0 | 2021-09-06 14:19:59.115 | debug | (27634) Redis Objects: Use Redis connection: 127.0.0.1:9001

</body>
</html>

This is the log when turn on the debug loglevel.

Apollon77, 2021-09-06:
So yes clear you entered "value" in all formula fields ... just remove that from your config and only set a formula when you need it ... and IF then use "x" as the placeholder for the value as staed in documentation

closed 2021-09-06 by Apollon77

Rockclimber87, 2021-09-06:
[image]

But there is nothing in formula? isn't it?

Apollon77, 2021-09-06:
Enable expert mode please (that "head icon")

Apollon77, 2021-09-06:
Did you maybe a CSV Improit and the values where there?

Rockclimber87, 2021-09-06:
Found it, thanks for your quick support also for a non expert. Thanks for your patience.