ioBroker/ioBroker.modbus #187: Factor Column is missing,

UwesSmartHome, 2021-11-03:
I have used since more as one year the MODBUS Adapter with no issues. I read some data and used for some values the Factor e.g. 0.01 to have the correct value.

Since more as 3 weeks the Factor and Formula column is gone, I can't see it. But the calculation is correct.

[image]

**Versions:**  
 - Adapter version: 3.4.14 (updated but former version the same issue
 - node -v 12.22.6
 - npm -v 6.14.15

 - Operating system: <Buster>

UwesSmartHome, 2021-11-03:
The same for input registers

[image]

ykuendig, 2021-11-05:
yepp -  Factor is missing here too / as well as the Offset Field ;-)

Apollon77, 2021-11-05:
Did you also tried with enabled "Expert mode"?

ykuendig, 2021-11-05:
If You are talking ybout the Expert-Mode on Instance-Overview  -> then Yes

ykuendig, 2021-11-05:
Thanks Apollo:
found the setting:
[image]
Fields are back.

Apollon77, 2021-11-05:
This looks like an old admin ... :-))

ykuendig, 2021-11-05:
5.1.25

UwesSmartHome, 2021-11-09:
Thanks, correct the fields are back

UwesSmartHome, 2021-11-09:
issue solved

closed 2021-11-09 by UwesSmartHome