ioBroker/ioBroker.modbus #202: Invalid Type

peterle-k, 2022-01-04:
**Describe the bug**  
warnings when adapter starts and connection to master

**Screenshots & Logfiles**  
[image]
[image]
[image]

**Versions:**  
 - Adapter version: 3.4.14
 - JS-Controller version: 3.3.22
 - Node version: v 14.8.2
 - Operating system: debian

hope this is useful
lg
Peter

Apollon77, 2022-01-04:
Please stop adapter, delete all the objects , they will be recreated on next start ... error still there?

peterle-k, 2022-01-05:
keine Änderung ... Wiederherstellung der Objekte hat funktioniert, Error unverändert

Denn281, 2022-01-26:
Habe das selbe problem....

<html><body>

modbus.0 | 18374 | 2022-01-26 10:38:23.468 | info | State value to set for "modbus.0.info.connection" has to be type "string" but received type "number"
-- | -- | -- | -- | --
modbus.0 | 18374 | 2022-01-26 10:38:23.451 | warn | Terminated (UNCAUGHT_EXCEPTION): Without reason
modbus.0 | 18374 | 2022-01-26 10:38:23.450 | debug | Plugin sentry destroyed
modbus.0 | 18374 | 2022-01-26 10:38:23.439 | info | terminating
modbus.0 | 18374 | 2022-01-26 10:38:23.422 | error | Exception-Code: EACCES: listen EACCES: permission denied 0.0.0.0:502
modbus.0 | 18374 | 2022-01-26 10:38:23.422 | error | Error:  listen EACCES: permission denied 0.0.0.0:502    at  Server.setupListenHandle [as _listen2] (net.js:1314:21)    at  listenInCluster (net.js:1379:12)    at doListen (net.js:1516:7)    at  processTicksAndRejections (internal/process/task_queues.js:83:21)
modbus.0 | 18374 | 2022-01-26 10:38:23.421 | error | uncaught exception: listen EACCES: permission denied 0.0.0.0:502
modbus.0 | 18374 | 2022-01-26 10:38:23.418 | debug | setting request handler
modbus.0 | 18374 | 2022-01-26 10:38:23.418 | debug | initiating write single register request handler.
modbus.0 | 18374 | 2022-01-26 10:38:23.417 | debug | setting request handler
modbus.0 | 18374 | 2022-01-26 10:38:23.417 | debug | initiating write single coil request handler.
modbus.0 | 18374 | 2022-01-26 10:38:23.417 | debug | setting request handler
modbus.0 | 18374 | 2022-01-26 10:38:23.416 | debug | initiating write multiple registers request handler.
modbus.0 | 18374 | 2022-01-26 10:38:23.414 | debug | setting request handler
modbus.0 | 18374 | 2022-01-26 10:38:23.413 | debug | initiating write multiple coils request handler.
modbus.0 | 18374 | 2022-01-26 10:38:23.412 | debug | setting request handler
modbus.0 | 18374 | 2022-01-26 10:38:23.411 | debug | initiating read input registers request handler.
modbus.0 | 18374 | 2022-01-26 10:38:23.410 | debug | setting request handler
modbus.0 | 18374 | 2022-01-26 10:38:23.409 | debug | initiating read holding registers request handler.
modbus.0 | 18374 | 2022-01-26 10:38:23.407 | debug | setting request handler
modbus.0 | 18374 | 2022-01-26 10:38:23.406 | debug | initiating read discrete inputs request handler.
modbus.0 | 18374 | 2022-01-26 10:38:23.406 | debug | setting request handler
modbus.0 | 18374 | 2022-01-26 10:38:23.405 | debug | initiating read coils request handler.
modbus.0 | 18374 | 2022-01-26 10:38:23.404 | debug | server is listening on port
modbus.0 | 18374 | 2022-01-26 10:38:23.172 | debug | Slave ready to start
modbus.0 | 18374 | 2022-01-26 10:38:23.070 | info | State value to set for "modbus.0.info.connection" has to be type "string" but received type "number"
modbus.0 | 18374 | 2022-01-26 10:38:22.958 | warn | This object will not be created in future versions. Please report this to the developer.
modbus.0 | 18374 | 2022-01-26 10:38:22.958 | warn | Object  modbus.0.holdingRegisters.13_Temp_Solar.VL is invalid: obj.common.write  has an invalid type! Expected "boolean", received  "number"
modbus.0 | 18374 | 2022-01-26 10:38:22.803 | warn | This object will not be created in future versions. Please report this to the developer.
modbus.0 | 18374 | 2022-01-26 10:38:22.803 | warn | Object  modbus.0.holdingRegisters.12_Heizung_VL.UG is invalid: obj.common.write  has an invalid type! Expected "boolean", received  "number"
modbus.0 | 18374 | 2022-01-26 10:38:22.728 | warn | This object will not be created in future versions. Please report this to the developer.
modbus.0 | 18374 | 2022-01-26 10:38:22.728 | warn | Object  modbus.0.holdingRegisters.11_Heizung_VL.OG is invalid: obj.common.write  has an invalid type! Expected "boolean", received  "number"
modbus.0 | 18374 | 2022-01-26 10:38:22.653 | warn | This object will not be created in future versions. Please report this to the developer.
modbus.0 | 18374 | 2022-01-26 10:38:22.653 | warn | Object  modbus.0.holdingRegisters.10_Solltemp_Fußbodenheizung is invalid:  obj.common.write has an invalid type! Expected "boolean", received   "number"
modbus.0 | 18374 | 2022-01-26 10:38:22.580 | warn | This object will not be created in future versions. Please report this to the developer.
modbus.0 | 18374 | 2022-01-26 10:38:22.580 | warn | Object  modbus.0.holdingRegisters.9_Sensorfehler_ is invalid: obj.common.write  has an invalid type! Expected "boolean", received  "number"
modbus.0 | 18374 | 2022-01-26 10:38:22.495 | warn | This object will not be created in future versions. Please report this to the developer.
modbus.0 | 18374 | 2022-01-26 10:38:22.494 | warn | Object  modbus.0.holdingRegisters.8_Temp_Außen is invalid: obj.common.write has  an invalid type! Expected "boolean", received  "number"
modbus.0 | 18374 | 2022-01-26 10:38:22.400 | warn | This object will not be created in future versions. Please report this to the developer.
modbus.0 | 18374 | 2022-01-26 10:38:22.400 | warn | Object  modbus.0.holdingRegisters.7_Temp_Kamin.UG is invalid: obj.common.write  has an invalid type! Expected "boolean", received  "number"
modbus.0 | 18374 | 2022-01-26 10:38:22.326 | warn | This object will not be created in future versions. Please report this to the developer.
modbus.0 | 18374 | 2022-01-26 10:38:22.325 | warn | Object  modbus.0.holdingRegisters.6_Temp_Speicher1.Unten is invalid:  obj.common.write has an invalid type! Expected "boolean", received   "number"
[... cut ...]
modbus.0 | 18374 | 2022-01-26 10:38:21.882 | warn | This object will not be created in future versions. Please report this to the developer.
modbus.0 | 18374 | 2022-01-26 10:38:21.881 | warn | Object  modbus.0.inputRegisters.1_Feigabe_Heizung UG is invalid:  obj.common.write has an invalid type! Expected "boolean", received   "number"
modbus.0 | 18374 | 2022-01-26 10:38:21.825 | info | State value to set for "modbus.0.info.connection" has to be type "string" but received type "number"

</body>
</html>

UncleSamSwiss, 2022-01-28:
Reason for this is the following line: https://github.com/ioBroker/ioBroker.modbus/blob/4abd927a6003a659fcfbc52d3e69adfbd563612a/main.js#L523

`slave` is not a boolean but a string or number (it is 1 if slave mode is enabled).

hennerich, 2022-02-04:
I have the same error. How can I solve the problem. Especially since newly created objects are not read in.

Apollon77, 2022-03-25:
Will be fixed in next version

closed 2022-03-25 by Apollon77