ioBroker/ioBroker.modbus #203: chang value in holding registers after update

mega-alexraab, 2022-01-11:
Good Morning,
i hope someone can help me.
I have a modbus tcp connection between iobroker and eaton hmi/plc.
Read and write coils works good. But to write holding registers i have problem. The value change the state to zero after 1 or 2 seconds.
This situation appears after i update the adapter from version 3.1.6 to 3.4.xx

for further information or open questions please contact me.

I can answer in german and english

Apollon77, 2022-01-11:
Put the Adaoter into debug log mode (Admin - enable Expertmode - Instances - expand the instance - change loglevel) ... maybe this helps what happens there

mega-alexraab, 2022-03-24:
I switched off the function "Write pulse" after that it works.

closed 2022-03-24 by Apollon77