ioBroker/ioBroker.modbus #204: Always get "Illegal Data Address" returned to master when trying to test as slave from modpoll

raintonr, 2022-01-15:
**Describe the bug**
When trying to create a slave instance that can be polled by a remote master... well... just can't get it to work. Obviously I'm doing something wrong but cannot tell what and no clues in logs.

**To Reproduce**  
Install the adapter. Configure as slave. Leave 'Use aliases' checked. Open holding register tab. Add holding register (eg. 40010, unsigned 16 bit integer).

Fire up [modpoll](https://www.modbusdriver.com/modpoll.html) which always shows errors:

```
# modpoll -m tcp -r 40010 x.x.x.x
modpoll 3.10 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2002-2021 proconX Pty Ltd
Visit https://www.modbusdriver.com for Modbus libraries and tools.

Protocol configuration: MODBUS/TCP, FC3
Slave configuration...: address = 1, start reference = 40010, count = 1
Communication.........: x.x.x.x, port 502, t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave... (Ctrl-C to stop)
Illegal Data Address exception response!
-- Polling slave... (Ctrl-C to stop)
Illegal Data Address exception response!
-- Polling slave... (Ctrl-C to stop)
Illegal Data Address exception response!
-- Polling slave... (Ctrl-C to stop)
Illegal Data Address exception response!
-- Polling slave... (Ctrl-C to stop)
^C
```

I even tried register 9, 10, 100, 40009, 40010, etc in case something fishy was going on with register numbering convention but always get the same error.

It would be super useful to have at least a warning spit out of the adapter that says which register it thinks is being asked for and why that is illegal. Even with debug on all you get is...

```2022-01-15 09:11:14.363  - debug: modbus.0 (3035) + Clients connected: x.x.x.x
2022-01-15 09:11:14.475  - debug: modbus.0 (3035) request outside register boundaries.
2022-01-15 09:11:14.489  - debug: modbus.0 (3035) sending tcp data
2022-01-15 09:11:15.448  - debug: modbus.0 (3035) request outside register boundaries.
2022-01-15 09:11:15.449  - debug: modbus.0 (3035) sending tcp data
2022-01-15 09:11:16.448  - debug: modbus.0 (3035) request outside register boundaries.
2022-01-15 09:11:16.450  - debug: modbus.0 (3035) sending tcp data
2022-01-15 09:11:17.448  - debug: modbus.0 (3035) request outside register boundaries.
2022-01-15 09:11:17.450  - debug: modbus.0 (3035) sending tcp data
2022-01-15 09:11:18.448  - debug: modbus.0 (3035) request outside register boundaries.
2022-01-15 09:11:18.450  - debug: modbus.0 (3035) sending tcp data
```

**Expected behavior**  
Expect `modpoll` to show value of register object.

**Versions:**  
 - Adapter version: 3.4.14
 - JS-Controller version: 3.3.22
 - Node version: v12.22.5
 - Operating system: Raspbian GNU/Linux 11 (bullseye)

raintonr, 2022-01-15:
Just as a follow up, I configured this...

[image]

... which leads to this output when starting in debug:

```
2022-01-15 09:37:16.735  - debug: modbus.0 (3808) Initialize Objects for holdingRegs: [{"_address":40010,"name":"Test value","description":"","unit":"","type":"uint16be","len":1,"factor":1,"offset":0,"formula":"","role":"value","room":"","poll":false,"wp":"","cw":false,"isScale":"","address":9,"deviceId":1,"id":"holdingRegisters.40010_Test_value"}]
2022-01-15 09:37:16.736  - debug: modbus.0 (3808) Add holdingRegisters.40010_Test_value: {"_id":"holdingRegisters.40010_Test_value","type":"state","common":{"name":"","role":"value","type":"number","read":false,"write":1,"def":0,"unit":""},"native":{"regType":"holdingRegs","address":9,"deviceId":1,"type":"uint16be","len":1,"offset":0,"factor":1,"poll":false}}
2022-01-15 09:37:16.807  - info: modbus.0 (3808) State value to set for "modbus.0.info.connection" has to be type "string" but received type "number" 
2022-01-15 09:37:16.816  - warn: modbus.0 (3808) Object holdingRegisters.40010_Test_value is invalid: obj.common.write has an invalid type! Expected "boolean", received  "number"
2022-01-15 09:37:16.817  - warn: modbus.0 (3808) This object will not be created in future versions. Please report this to the developer.
2022-01-15 09:37:16.862  - info: modbus.0 (3808) State value to set for "modbus.0.info.connection" has to be type "string" but received type "number" 
2022-01-15 09:37:16.913  - debug: modbus.0 (3808) Slave ready to start
```

Ok - so few warnings but let's ignore them. Seems the register is there. But then I added some logging lines in `lib/jsmodbus/handler/server/ReadHoldingRegisters.js` and found...

```
2022-01-15 09:38:12.440  - debug: modbus.0 (3808) new connection
2022-01-15 09:38:12.444  - debug: modbus.0 (3808) + Clients connected: x.x.x.x
2022-01-15 09:38:12.552  - debug: modbus.0 (3808) fc/start/byteStart/quantity: 3/40009/80018/1
2022-01-15 09:38:12.554  - debug: modbus.0 (3808) mem.length: 0
2022-01-15 09:38:12.555  - debug: modbus.0 (3808) request outside register boundaries.
2022-01-15 09:38:12.565  - debug: modbus.0 (3808) sending tcp data
2022-01-15 09:38:13.535  - debug: modbus.0 (3808) fc/start/byteStart/quantity: 3/40009/80018/1
2022-01-15 09:38:13.537  - debug: modbus.0 (3808) mem.length: 0
2022-01-15 09:38:13.539  - debug: modbus.0 (3808) request outside register boundaries.
2022-01-15 09:38:13.541  - debug: modbus.0 (3808) sending tcp data
2022-01-15 09:38:14.537  - debug: modbus.0 (3808) fc/start/byteStart/quantity: 3/40009/80018/1
2022-01-15 09:38:14.539  - debug: modbus.0 (3808) mem.length: 0
2022-01-15 09:38:14.541  - debug: modbus.0 (3808) request outside register boundaries.
2022-01-15 09:38:14.543  - debug: modbus.0 (3808) sending tcp data
```

So that `this.getHolding()` is returning no registers. Why?

UncleSamSwiss, 2022-01-28:
Just as a sidenote: The "invalid type" warning is discussed here: https://github.com/ioBroker/ioBroker.modbus/issues/202#issuecomment-1024227045

raintonr, 2022-01-29:
OK, so I have discovered how to get this to work as a slave:

- To configure holding register address `40010` in ioBroker to be read from a Master, the 'Poll' option for that specific register needs to be enabled:
[image]
- To be able to use `40010`, 'Use aliases' needs to be checked in General config page:
[image]
- In modpoll this is actually numbers as (holding register) 10: `modpoll -m tcp -r 10 x.x.x.x`
- In ioBroker, when updating the associated object (`modbus.0.holdingRegisters.40010_Test_value` in the example) do not set `ack`. If `ack` is set the update doesn't trigger an update in the Modbus adapter and therefore is not read from any polling masters.

So, yeah, seems like it works, but doco and logging should be more specific as this took quite some trial and error to discover.

rvdgaag, 2022-02-01:
> So, yeah, seems like it works, but doco and logging should be more specific as this took quite some trial and error to discover.

@raintonr Can you please share the general setting screen ofor Modbus in iobroker , I have still issues reading and setting holding registers in the Smart EVSE, reading the connect Eastron meters works fine.

raintonr, 2022-02-01:
> ...I have still issues reading and setting holding registers in the Smart EVSE-2 reading the connected 3-phase Eastron meters works fine.

What we're talking about here will not help you at all then. This issue is regarding IOB acting as a slave and if you are reading from meters so your IOB instance is a master.

closed 2022-06-03 by Apollon77