ioBroker/ioBroker.modbus #205: Verbindung zu Solarmax Solar Umrichter geht nicht

jh2k19, 2022-01-29:
Habe schon etliche unterschiedliche Geräte per Modbus.Adapter an ioB angebunden.
Leider funktioniert das mit einem Solarmax MaxStorrage Ultimate von SolarMax nicht.
Über einen Modbus Scanner Tool kann ich die Datenpunkte problemlos auslesen.
Über den ioB Modus Adapter klappt das trotz aller Bemühungen dieses mal nicht.
Hier mal das Protokoll und die Datenpunkliste.
Könnt ihr mir da bitte helfen?

[image]

[image]

[image]

[image]

[image]

jh2k19, 2022-01-30:
Vom vielen herumprobieren kamen auch mal Werte in den Objekten. Aber es funktioniert einfach nicht richtig, da der Adapter ja immer di Connection trennt. Also es scheint nur ein kleines Problem zu sein, da ja mal Werte kommen. Aber warum sich die Connection sofort wieder trennt weis ich einfach nicht. Auf einen SmartMeter+ von SolarMax (hat natürlich eine andere IP) kann ich dagegen problemlos zugreifen und der Adapter beendet sich nicht automatisch. Was ist das nur?

jh2k19, 2022-01-30:
Wenn ich den Adapter per Cron alle 10 sec. neu starte werden die Werte in die Datenpunkten eingelesen und aktualisiert.

Apollon77, 2022-01-30:
ALso "socket closed with error" besagt eigentlich eher das die gegenstelle den socket schliesst mit einem Fehler ... nicht der Adapter

jh2k19, 2022-01-30:
Danke für den Hinweis.
Ich versuche morgen mal SolarMax zu erreichen mit der Hoffnung, dass die den Fehler dann auf ihrer Seite beheben können.
Melde mich dann wieder.

jh2k19, 2022-01-30:
Ich könnte mir noch vorstellen, dass die den Anschluss am Hauptgerät evtl. nicht durchgehend offen halten, weil gleichzeitig auch andere Geräte  (wie z.B. der zweite Solarumrichter oder das Energie Management Modul) mit dem Hauptgerät komunizieren. Oder ist es wie bei jedem anderen Bus, bei dem alle Geräte jederzeit kommunizieren können?

Apollon77, 2022-01-30:
Naja das kann ggf gut sein ... an sich hätte ich erwartet das mehrere TCP Verbindungen gehen, aber who knows

jh2k19, 2022-02-08:
Also von Solarmax selbst gibt es keine Hilfe. Die Komponenten sind so abgestimmt, dass die Daten in deren Webportal verarbeitet werden können. Damit habe der Endkunde ja alles was er an Anzeige und Statistik benötige. Es sei nicht angedacht, dass der Endkunde selbst Daten weiterverarbeiten kann oder soll. Außerdem rennen denen die Kunden die Bude ein. Personal fehlt hinten und vorne. Erreichbarkeit = 0. Echt Traurig. 

By the Way...

Ich habe festgestellt, dass der Adapter am leben bleibt, wenn ich die Pollzeit < 900ms mache.
Es kommen ein paar Daten mit denen ich arbeiten kann.
(Leistungen Dach, Netz, Haus, Seriennummer, Firwarestand)
Allerdings fehlen solche wie z.B. erzeugte Energie in kWh, Ströme, CosPhi, Temp. usw.

Falls noch jemand was konstruktives hat, einfach her damit.

closed 2022-03-15 by jh2k19