ioBroker/ioBroker.modbus #214: Reading from another slave not working if ioBroker.modbus is activated

yymsc, 2022-02-17:
I have a SolarEdge SE16K inverter and I am able to read its values into ioBroker:
I also have a car charger (cFos PowerBrain) and this charger also needs to read the same values from the inverter. Unfortunately this doesn't seem to work because as soon as the modbus instance is activated in ioBroker the charger can't read the values anymore.
To me it looks like ioBroker.modbus keeps the connection blocked and no other device is able to communicate with the inverter anymore.
Is there a way to get around this? I already tried to change eg. 'Data polling interval', 'Read timeout' etc but no luck...

Apollon77, 2022-02-17:
Modbus uses a "standing connection". Then it seems to me that the network stack from the inverter only allows one active conection ... so honestly ... you need to decide for one

Apollon77, 2022-03-25:
no further infos

closed 2022-03-25 by Apollon77