ioBroker/ioBroker.modbus #221: Modbus RTU Error by read discrete Input

MartinMohr, 2022-03-05:
If I try to read a discrete Input via Modbus RTU I got this Error:
modbus.1 | 2022-03-05 12:34:51.472 | error | Error response: Value is not accepted by slave
modbus.1 | 2022-03-05 12:34:51.470 | error | Error response for FCx2: Illegal Data Value
All other functions works as expected 
The Hardware is tested with modpoll and works fine 

Adapter Version 3.4.17

Apollon77, 2022-03-05:
DIy you plyes around with the settings for the FC stuff (there are some)

MartinMohr, 2022-03-05:
Nicht bewust ... wo finde ich die Einstellungen?

Apollon77, 2022-03-05:
Gemeint sind einstellungen 

* "Mehrere Register schreiben" nicht verwenden bzw
* Nur "Write multiple registers" verwenden

closed 2022-05-13 by GermanBluefox