ioBroker/ioBroker.modbus #225: How to set Bits by an Holding Register?

kkeintzel, 2022-04-03:
for example:
E3/DC S10 - Holding Register 40088 (WallBox_0_CTRL)
for controlling a Wallbox via Modbus-TCP, "Bit 0 to Bit 12" can/must be set with Function 05h (write single coil)

**How** can/must i set this with your iobroker.modbus Adapter

Thanks for advance

GermanBluefox, 2022-05-13:
You cannot write Holding Register with Function Code 05. 
You can only use FC6 or FC16 for that: https://www.simplymodbus.ca/FC06.htm

Do you have link to MODBUS instructions for your device?

kkeintzel, 2022-05-15:
Here the actual MODBUS instruction für E3/DC:

Hopefully you can give me a description  to set the bits for Holding Register 40088 (WallBox_0_CTRL)

thx in advance

[ModBus_E3DC_Speichersysteme_V2.00_2022-03-07.pdf](https://github.com/ioBroker/ioBroker.modbus/files/8694860/ModBus_E3DC_Speichersysteme_V2.00_2022-03-07.pdf)

GermanBluefox, 2022-05-17:
Also das ist typisch Modbus. Man darf alles mit allem machen.
[image]

Laut Doku sind die Adressen über 40000 - Holding register und die Holding Register müssen mit FC06, FC16 geschrieben werden.
Ich denke es geht hier um eine Art interne Mapping, so dass Die Holding Register intern im Gerät mit Coils verbunden sind.

Ich empfehle eine Simulation software zu nehmen (z.B. https://www.modbusdriver.com/modpoll.html) und mit Try&Error rauszufinden was da gemeint ist. 
Vielleicht kann man mit FC05 auf Address 87*8 + bit schreiben. Vielleicht wird sogar FC06 auf 40088 gehen man muss nur Bits maskieren.

Keine Ahnung, aber es ist nicht normal im Gerät.