ioBroker/ioBroker.modbus #235: Nur erstes Leerzeichen wird ersetzt, nachfolgende nicht.

HansPeterMeier, 2022-05-11:
Bei den ID Namen unter Objekte wird das erste Leerzeichen durch einen Unterstrich ersetzt.
Nachfolgende Leerzeichen werden dann nicht mehr ersetzt. Denke es wäre schöner, wenn alle Leerzeichen ersetzt werden, oder keine, falls das anderweitig keine Probleme macht.

[image]
[image]
[image]

 - Adapter version: 4.0.4
 - JS-Controller version: 4.0.23
 - Node version: NODE.JS: V14.19.2
 - Operating system: Debian

Apollon77, 2022-05-11:
Das würde ich aufheben wenn wir mal wieder nen breaking change brauchen :-) Weil es ist einer .....

Apollon77, 2022-05-11:
Ok, I prepare the change

GermanBluefox, 2022-05-13:
Fixed in 5.0.3

closed 2022-06-03 by Apollon77