ioBroker/ioBroker.modbus #236: Can not execute task add for ID undefined: The id is empty

JBSullivan, 2022-05-12:
Heute Adapter Update auf 5.0.0 durchgeführt. Im Anschluss habe ich folgende Fehlermeldung erhalten. Umstellung auf DEBUG, aber es kommen keine weiteren Informationen zu dem Fehler, das man ggf. heraus finden könnte, welche ID den Fehler liefert.

Admin 5.4.8

PlatformBetriebssystem:win32
Architektur:x64
CPUs:4
Geschwindigkeit:1995 MHz
Modell:Intel(R) Core(TM) i3-5005U CPU @ 2.00GHz
RAM:7.9 GB
Node.js:v14.16.0
NPM:6.14.11
Aktive Instanzen:43

`<html>
<body>

modbus.0 | 2022-05-12 10:25:47.795 | info | Can not execute task add for ID undefined: The id is empty! Please provide a valid id.
-- | -- | -- | --
modbus.0 | 2022-05-12 10:25:47.776 | info | Can not execute task add for ID undefined: The id is empty! Please provide a valid id.
modbus.0 | 2022-05-12 10:25:47.727 | info | Can not execute task add for ID undefined: The id is empty! Please provide a valid id.
modbus.0 | 2022-05-12 10:25:47.725 | info | Can not execute task add for ID undefined: The id is empty! Please provide a valid id.

</body>
</html>`

Apollon77, 2022-05-12:
Kannst Du bitte maldie Konfiguration zeigen und am besten Loglevel auf Debug stellen bei der Instanz und dannLog nochmal s posten?

GermanBluefox, 2022-05-13:
Please use 5.0.1. I added additional outputs there to catch the errro

JBSullivan, 2022-05-13:
5.0.1. ist installiert. Außer dem ersten Eintrag ist trotz aktivierter DEBUG Einstellungen keine neue Meldung dazu gekommen.

`<html>
<body>

modbus.0 | 2022-05-13 13:06:17.279 | info | Connected to slave 10.122.60.57
-- | -- | -- | --
modbus.0 | 2022-05-13 13:06:16.177 | info | Can not execute task "add" for ID undefined: The id is empty! Please provide a valid id.
modbus.0 | 2022-05-13 13:06:16.154 | info | Can not execute task "add" for ID undefined: The id is empty! Please provide a valid id.
modbus.0 | 2022-05-13 13:06:16.064 | info | Can not execute task "add" for ID undefined: The id is empty! Please provide a valid id.
modbus.0 | 2022-05-13 13:06:16.063 | info | Can not execute task "add" for ID undefined: The id is empty! Please provide a valid id.

</body>
</html>`

[image]

[image]

Es sind nur Einträge in den Holding Register vorhanden und da werden z.Zt. nur die beiden angehakten DP`s in die Objekte geschrieben.

[image]

Apollon77, 2022-05-13:
@JBSullivan again ... Ist Debug WIRKLICH aktiviert für die Instanz??? Da MUSS mehr kommen. Mal bitte auch im Lohgfile auf Platte schauen

GermanBluefox, 2022-05-13:
Und bitte 5.0.2 installieren.

JBSullivan, 2022-05-13:
Wie du oben an dem Screenshot sehen kannst, steht das Loglevel auf DEBUG. Wenn man das Loglevel ändert, startet die Instanz ja eigentlich neu sofern man das nicht abgewählt hat - oder? ....... scheinbar ist der Instanz Neustart aber nicht so tiefgreifend, das dass Loglevel aktiviert wird. Ich habe jetzt nochmal einen händischen Neustart gemacht habe und nun kommt auch etwas mehr im Log an.

Ist aber auch nicht wirklich das, womit ICH was anfangen kann.

`2022-05-13 14:50:50.511  - [34mdebug[39m: modbus.0 (10340) [DevID_200] Poll start ---------------------
2022-05-13 14:50:53.562  - [34mdebug[39m: modbus.0 (10340) [DevID_1] Poll start ---------------------
2022-05-13 14:50:53.563  - [34mdebug[39m: modbus.0 (10340) [DevID_1/holdingRegs] Poll address 40044 - 16 registers
2022-05-13 14:50:53.566  - [34mdebug[39m: modbus.0 (10340) [DevID_1/holdingRegs] Poll address 40044 DONE
2022-05-13 14:50:53.568  - [34mdebug[39m: modbus.0 (10340) [DevID_1/holdingRegs] Poll address 40109 - 2 registers
2022-05-13 14:50:53.578  - [34mdebug[39m: modbus.0 (10340) [DevID_1/holdingRegs] Poll address 40109 DONE
2022-05-13 14:50:53.580  - [34mdebug[39m: modbus.0 (10340) [DevID_1/holdingRegs] Poll address 40235 - 1 registers
2022-05-13 14:50:53.584  - [34mdebug[39m: modbus.0 (10340) [DevID_1/holdingRegs] Poll address 40235 DONE
2022-05-13 14:50:53.636  - [34mdebug[39m: modbus.0 (10340) [DevID_200] Poll start ---------------------
2022-05-13 14:50:56.689  - [34mdebug[39m: modbus.0 (10340) [DevID_1] Poll start ---------------------`

JBSullivan, 2022-05-13:
So, eben die 5.0.2 installiert - Meldungen kommen immer noch, aber eine grundsätzliche Beeinträchtigung gibt es nicht.

So sieht das DEBUG LOG direkt nach dem Adapter Update und Neustart aus. Danach kommen nur noch die Meldungen wie aus dem Posting hier drüber. Ich hoffe das hilft.

`2022-05-13 15:02:26.776  - [32minfo[39m: host.GLT(GLT) iobroker + iobroker.modbus@5.0.2
removed 4 packages, updated 1 package and audited 2354 packages in 53.095s

2022-05-13 15:02:29.451  - [32minfo[39m: host.GLT(GLT) iobroker 

2022-05-13 15:02:29.452  - [32minfo[39m: host.GLT(GLT) iobroker 145 packages are looking for funding
  run `npm fund` for details

2022-05-13 15:02:29.455  - [32minfo[39m: host.GLT(GLT) iobroker found 174 vulnerabilities (18 low, 54 moderate, 90 high, 12 critical)
  run `npm audit fix` to fix them, or `npm audit` for details

2022-05-13 15:02:29.711  - [32minfo[39m: host.GLT(GLT) iobroker upload [9] modbus.admin C:/iobroker/GLT/node_modules/iobroker.modbus/admin/asset-manifest.json asset-manifest.json application/json
2022-05-13 15:02:29.727  - [32minfo[39m: host.GLT(GLT) iobroker upload [8] modbus.admin C:/iobroker/GLT/node_modules/iobroker.modbus/admin/favicon.ico favicon.ico image/vnd.microsoft.icon
2022-05-13 15:02:29.734  - [32minfo[39m: host.GLT(GLT) iobroker upload [7] modbus.admin C:/iobroker/GLT/node_modules/iobroker.modbus/admin/index_m.html index_m.html text/html
2022-05-13 15:02:29.741  - [32minfo[39m: host.GLT(GLT) iobroker upload [6] modbus.admin C:/iobroker/GLT/node_modules/iobroker.modbus/admin/manifest.json manifest.json application/json
2022-05-13 15:02:29.748  - [32minfo[39m: host.GLT(GLT) iobroker upload [5] modbus.admin C:/iobroker/GLT/node_modules/iobroker.modbus/admin/modbus.png modbus.png image/png
2022-05-13 15:02:29.755  - [32minfo[39m: host.GLT(GLT) iobroker upload [4] modbus.admin C:/iobroker/GLT/node_modules/iobroker.modbus/admin/static/css/main.fb1757a1.css static/css/main.fb1757a1.css text/css
2022-05-13 15:02:29.761  - [32minfo[39m: host.GLT(GLT) iobroker upload [3] modbus.admin C:/iobroker/GLT/node_modules/iobroker.modbus/admin/static/css/main.fb1757a1.css.map static/css/main.fb1757a1.css.map application/json
2022-05-13 15:02:29.768  - [32minfo[39m: host.GLT(GLT) iobroker upload [2] modbus.admin C:/iobroker/GLT/node_modules/iobroker.modbus/admin/static/js/main.bd869655.js static/js/main.bd869655.js application/javascript
2022-05-13 15:02:29.810  - [32minfo[39m: host.GLT(GLT) iobroker upload [1] modbus.admin C:/iobroker/GLT/node_modules/iobroker.modbus/admin/static/js/main.bd869655.js.LICENSE.txt static/js/main.bd869655.js.LICENSE.txt text/plain
2022-05-13 15:02:29.834  - [32minfo[39m: host.GLT(GLT) iobroker upload [0] modbus.admin C:/iobroker/GLT/node_modules/iobroker.modbus/admin/static/js/main.bd869655.js.map static/js/main.bd869655.js.map application/json
2022-05-13 15:02:30.654  - [32minfo[39m: host.GLT(GLT) iobroker Update "system.adapter.modbus.0"
2022-05-13 15:02:31.139  - [32minfo[39m: host.GLT(GLT) iobroker host.GLT(GLT) Adapter "system.adapter.modbus.0" is started
2022-05-13 15:02:31.330  - [32minfo[39m: host.GLT(GLT) "system.adapter.modbus.0" enabled
2022-05-13 15:02:31.528  - [32minfo[39m: host.GLT(GLT) instance system.adapter.modbus.0 started with pid 4500
2022-05-13 15:02:32.513  - [32minfo[39m: host.GLT(GLT) iobroker exit 0
2022-05-13 15:02:33.288  - [34mdebug[39m: modbus.0 (4500) Redis Objects: Use Redis connection: 127.0.0.1:9001
2022-05-13 15:02:33.343  - [34mdebug[39m: modbus.0 (4500) Objects client ready ... initialize now
2022-05-13 15:02:33.354  - [34mdebug[39m: modbus.0 (4500) Objects create System PubSub Client
2022-05-13 15:02:33.356  - [34mdebug[39m: modbus.0 (4500) Objects create User PubSub Client
2022-05-13 15:02:33.469  - [34mdebug[39m: modbus.0 (4500) Objects client initialize lua scripts
2022-05-13 15:02:33.475  - [34mdebug[39m: modbus.0 (4500) Objects connected to redis: 127.0.0.1:9001
2022-05-13 15:02:33.521  - [34mdebug[39m: modbus.0 (4500) Redis States: Use Redis connection: 127.0.0.1:9000
2022-05-13 15:02:33.548  - [34mdebug[39m: modbus.0 (4500) States create System PubSub Client
2022-05-13 15:02:33.549  - [34mdebug[39m: modbus.0 (4500) States create User PubSub Client
2022-05-13 15:02:33.601  - [34mdebug[39m: modbus.0 (4500) States connected to redis: 127.0.0.1:9000
2022-05-13 15:02:33.914  - [34mdebug[39m: modbus.0 (4500) Plugin sentry Initialize Plugin (enabled=true)
2022-05-13 15:02:34.416  - [32minfo[39m: modbus.0 (4500) starting. Version 5.0.2 (non-npm: ioBroker/ioBroker.modbus) in C:/iobroker/GLT/node_modules/iobroker.modbus, node: v14.16.0, js-controller: 4.0.23
2022-05-13 15:02:34.749  - [34mdebug[39m: modbus.0 (4500) Initialize Objects for disInputs: []
2022-05-13 15:02:34.750  - [34mdebug[39m: modbus.0 (4500) Initialize Objects for coils: []
2022-05-13 15:02:34.751  - [34mdebug[39m: modbus.0 (4500) Initialize Objects for inputRegs: []
2022-05-13 15:02:34.751  - [34mdebug[39m: modbus.0 (4500) Initialize Objects for holdingRegs: [{"_address":40020,"deviceId":1,"name":"Md","description":"Gerätetyp","unit":"","type":"string","len":"16","factor":"","offset":"","formula":"","role":"level","room":"","poll":false,"wp":false,"cw":false,"isScale":false,"address":40020,"id":"holdingRegisters.1.40020_Md"},{"_address":40044,"deviceId":1,"name":"Vr","description":"Software Version","unit":"","type":"string","len":16,"factor":1,"offset":0,"formula":"","role": [... cut ...]
2022-05-13 15:02:35.079  - [32minfo[39m: modbus.0 (4500) Can not execute task "add" for ID undefined: The id is empty! Please provide a valid id. ({"name":"add","obj":{"type":"state","common":{"name":"Total Real Power Imported","role":"value","type":"number","read":false,"write":true,"def":0,"unit":"Wh"},"native":{"regType":"holdingRegs","deviceId":200,"type":"uint32be","len":"2","offset":"0","factor":"1","poll":false}}}
2022-05-13 15:02:36.102  - [32minfo[39m: modbus.0 (4500) Connected to slave 10.122.60.57
2022-05-13 15:02:36.104  - [34mdebug[39m: modbus.0 (4500) [DevID_1] Poll start ---------------------`

JBSullivan, 2022-05-13:
OH - eben beim zurück stellen von Debug auf Info kam das erste mal ein Error

`<html>
<body>

modbus.0 | 14104 | 2022-05-13 15:11:26.940 | error | Invalid object: {"type":"state","common":{"name":"Total Real Power Imported","role":"value","type":"number","read":false,"write":true,"def":0,"unit":"Wh"},"native":{"regType":"holdingRegs","deviceId":200,"type":"uint32be","len":"2","offset":"0","factor":"1","poll":false}}
-- | -- | -- | -- | --
modbus.0 | 14104 | 2022-05-13 15:11:26.939 | error | Invalid data holdingRegisters/16: {"_address":40115,"deviceId":200,"name":"TotWhImp","description":"Total Real Power Imported","unit":"Wh","type":"uint32be","len":"2","factor":"1","offset":"0","formula":"x * Math.pow (10, sf ['40123']);","role":"value","room":"","poll":false,"wp":false,"cw":false,"isScale":false,"fullId":"modbus.0.16"}
modbus.0 | 14104 | 2022-05-13 15:11:26.938 | error | Invalid object: {"type":"state","common":{"name":"Total Reale Energy Exported","role":"value","type":"number","read":false,"write":true,"def":0,"unit":"Wh"},"native":{"regType":"holdingRegs","deviceId":200,"type":"uint32be","len":"2","offset":"0","factor":"1","poll":false}}
modbus.0 | 14104 | 2022-05-13 15:11:26.938 | error | Invalid data holdingRegisters/13: {"_address":40107,"deviceId":200,"name":"TotWhExp","description":"Total Reale Energy Exported","unit":"Wh","type":"uint32be","len":"2","factor":"1","offset":"0","formula":"x * Math.pow (10, sf ['40123']);","role":"value","room":"","poll":false,"wp":false,"cw":false,"isScale":false,"fullId":"modbus.0.13"}
modbus.0 | 14104 | 2022-05-13 15:11:26.936 | error | Invalid object: {"type":"state","common":{"name":"Scale factor","role":"value","type":"number","read":false,"write":true,"def":0,"unit":""},"native":{"regType":"holdingRegs","deviceId":200,"type":"int16be","len":"1","offset":"0","factor":"1","poll":false}}
modbus.0 | 14104 | 2022-05-13 15:11:26.936 | error | Invalid data holdingRegisters/6: {"_address":40091,"deviceId":200,"name":"W_SF","description":"Scale factor","unit":"","type":"int16be","len":"1","factor":"1","offset":"0","formula":"","role":"value","room":"","poll":false,"wp":false,"cw":false,"isScale":true,"fullId":"modbus.0.6"}
modbus.0 | 14104 | 2022-05-13 15:11:26.935 | error | Invalid object: {"type":"state","common":{"name":"Total Real Power","role":"value","type":"number","read":false,"write":true,"def":0,"unit":"W"},"native":{"regType":"holdingRegs","deviceId":200,"type":"int16be","len":"1","offset":"0","factor":"1","poll":false}}
modbus.0 | 14104 | 2022-05-13 15:11:26.934 | error | Invalid data holdingRegisters/5: {"_address":40087,"deviceId":200,"name":"W","description":"Total Real Power","unit":"W","type":"int16be","len":"1","factor":"1","offset":"0","formula":"x * Math.pow (10, sf ['40091']);","role":"value","room":"","poll":false,"wp":false,"cw":false,"isScale":false,"fullId":"modbus.0.5"}
modbus.0 | 14104 | 2022-05-13 15:11:26.601 | info | starting. Version 5.0.2 (non-npm: ioBroker/ioBroker.modbus) in C:/iobroker/GLT/node_modules/iobroker.modbus, node: v14.16.0, js-controller: 4.0.23

</body>
</html>`

GermanBluefox, 2022-05-13:
I could reproduce your problem and please try 5.0.3

JBSullivan, 2022-05-13:
Perfekt - Problem scheint gelöst zu sein !!! Keine (Fehler)Meldungen mehr !!! 👍👍👍

closed 2022-05-13 by GermanBluefox