ioBroker/ioBroker.modbus #237: Index of folder for InputRegisters was 0, now it is 1

ThomasS1962, 2022-05-13:
Modbus is running in my system since 2019. I am receiving all data from my solar system via modbus. The data are stored in the History adapter. Today I opened the configuration and added one register. Suddenly all registers are stored in:
modbus.0.inputRegisters.1
Before adding the register, the folder was:
modbus.0.inputRegisters.0

Now I am in trouble because the History adapter does not record, my scripts do not work and Vis does not find the values.

How can I store the registers again at the index .0?
The slave IDs are 1. When I change them to 0, it will not be stored.

Version 4.0.4

Thanks for your help!

[image]
[image]
[image]

Apollon77, 2022-05-13:
Beside the question whats the reason for this .... (@GermanBluefox needs to check that). In history you can enable the logging for the "new" stats ID and then enter the "old" state ID name in the "alias field" in the history datapoint confiug. Then te new values will be logged untder the old name

ThomasS1962, 2022-05-14:
Thank you for your help. Now my logging is working again and I can access the older data. I have modified Vis and the scripts.
Everythink works fine again.

Best regards
Thomas

closed 2022-05-14 by ThomasS1962