ioBroker/ioBroker.modbus #242: gettint "Poll error count" while reading data from cfos wallbox

christianh17, 2022-05-31:
**Describe the bug**  
I try to integrate a cfos wallbox (22 kW, reduced to 11 kW) and get errors "Poll error count".  In consequence the data gets a delayed update (sometimes it works) and there are reconnections after some errors.

**To Reproduce**  
Steps to reproduce the behavior:
nothing special to do, install modbus adapter, configure adapter like cfos dokumentation and have a look at the log.

**Expected behavior**  
Reading the data without error.

**Screenshots & Logfiles**  
Logfile:

2022-05-31 07:42:04.811	debug	[DevID_1/holdingRegs] Poll address 8082 - 3 registers
2022-05-31 07:41:58.355	debug	[DevID_1/holdingRegs] Poll address 8082 DONE
2022-05-31 07:41:58.027	debug	[DevID_1/holdingRegs] Poll address 8082 - 3 registers
2022-05-31 07:41:54.810	debug	[DevID_1/holdingRegs] Poll address 8080 DONE
2022-05-31 07:41:54.802	debug	[DevID_1/holdingRegs] Poll address 8080 - 2 registers
2022-05-31 07:41:54.801	debug	[DevID_1] Poll start ---------------------
2022-05-31 07:41:52.999	warn	Poll error count: 1 code: "App Timeout"
2022-05-31 07:41:48.023	debug	[DevID_1/holdingRegs] Poll address 8080 DONE
2022-05-31 07:41:42.805	debug	[DevID_1/holdingRegs] Poll address 8080 - 2 registers
2022-05-31 07:41:42.802	debug	Initialization of scale factors done!
2022-05-31 07:41:42.797	debug	[DevID_1] Poll start ---------------------

Wireshark capture:
(192.168.17.40 => iobroker and 192.168.17.7 => wallbox)
[image]

Configuration:
[image]
[image]

**Versions:**  
 - Adapter version: 5.03
 - JS-Controller version: 4.0.15
 - Node version: v14.19.3
 - Operating system: synology-docker (buanet/iobroker:latest)

**Additional context**  
I am not sure where to search for the problem. Perhaps it's a problem from the wallbox but I really do not know.

tp1de, 2022-06-08:
Setze mal das Lese- und Schreibintervall auf 0.
Das Datenabfrageintervall steht auf 1000 - also jede Sekunde. 
Lass die Instanz mal mit Protokoll-Ebene "Alles" laufen. Dann siehst Du wie lange ein Poll läuft. Das Abfrageintervall sollte dann deutlich größer sein.

GermanBluefox, 2022-06-15:
Hm. Wallbox antwortet erst nach 6 Sekunden... Paket 9. Ist das normal??
Natürlich wird ioBroker sagen, dass keine ihm antwortet. Wie hoch ist die Timeout Einstellung?

closed 2023-05-09 by stale[bot]