ioBroker/ioBroker.modbus #246: Factor and Scale Factor

stwa2020, 2022-06-04:
When Factor is used it's not working anymore in combination with Scale Factor formula.

Register is 40094 and SF is 40096

**Register 40094 is correct when Factor is set to 0.001 to have kWh**
[image]

[image]

**Register 40094 is not correct when Factor is set to 0.001 to have kWh and Formula for SF is used, it looks like the Factor is ignored:**
[image]

[image]

Apollon77, 2022-06-04:
Exactly ... Formular OR Scaling (aka another internal formula). So if you need a scaling include in your formula

Apollon77, 2022-06-04:
PS: ALso the README tells exactly this. I now made that part bold

closed 2022-06-04 by Apollon77