ioBroker/ioBroker.modbus #247: Lost continuity of data due to Update to V5, replacing blanks by underscore.

gidhap, 2022-06-05:
I did the upgrade to V5, just to be up to date as the system will run a couple of months unattended now.
The release note says: 
" BREAKING: All space characters will be replaced with underscores now in the Objects IDs, not only the first one." 
Which I did read. However as I did not understand this BREAKING as a warning with severe implications on existing objects, I did not expect loosing data continuity due to this renaming. 

What happened:
I have an object 
modbus.1.inputRegisters.1029_PV_Balkon Power
object's was recorded by influxdb and displayed by various Grafana dashboards.
Since the update my respective graphs are empty

It took me a while, where at the end I used _Time Series Admin_, to figure out the object has been renamed to:
modbus.1.inputRegisters.1029_PV_Balkon_Power 
where recording by influxdb has not been activated for the new object.

I did change all my Grafana graphs, kept the old entry to keep data previous to the upgrade, and added the new object.
So at the end I am missing data for a couple of days, which isn't nice but acceptable.
I have to go through all objects now and see if there is further loss of data due to renaming.

I don't write this as an real "BUG" here, as it looks like the upgrade did what it was supposed to - renaming-, however I'd suggest to add a clear BIG warning to the notes for upgrade in case they have such a severe impact. (Or did I miss them)

Finally I want to express my great respect for all the work being done around the ioBroker, your work is amazing.

Apollon77, 2022-06-05:
InfluxDB and all History Adapter have an option for this. Simply activate Influxdb on the new object and set the "old" state id as "Alias" in the influxdb-configuration of the datapoint. then logging will happen under the "alias"=== "old" ID.

Apollon77, 2022-06-05:
PS: The changelog stated the breaking change, Admin should have shown you also an extra "orange" shown medssage in the update dialog also stating the infos with the object name changes ... We can not do more communication ... Maybe just add info especially for influxdb or such ....

closed 2022-06-05 by Apollon77