ioBroker/ioBroker.modbus #252: no actual data in registers sent until restarting adapter

Tom54497, 2022-07-21:
Using version 5.0.4 in slave mode with working connection to the master

To poll an input register works only once and I will get the correct value. I have to restart the adapter to get a new value which is also pollible one time.

Background:
Made a script which feeds the register 30001 with an interval of 60 sec with new data. That data is polled by a master. If new data is filled in register 30001 the master gets the last and old value but not the actualized one.

Apollon77, 2022-08-29:
Please provide a debug log

mitbringsel, 2022-09-01:
It works for me, if I change the value manually without the "Acknowledged" flag or using the script-block "control". If the flag "Acknowledged" is set, the client pulls some old value.

Apollon77, 2022-09-01:
This is exactly correct ... control means "ack=false" which sends data to the device. settings values with ack=true should normally not trigger any control/sending to the device ... any yes then maybe old values from the device are correctly used to overwrite the manual set ack=true values