ioBroker/ioBroker.modbus #255: strange value behaviour

TeslaNator, 2022-08-01:
Everythings fine until I add first input register (A), that I need. Let B an enum and C a temperature. Normally B is 0.
If I add A to the config, B is not 0. Then B has the value of C.

**To Reproduce**  
Take following input registers
_address	name	description	unit	type	len	factor	offset	formula	role	room	cw	isScale
1100	Vorlauftemperatur1	*10	°C	int16be	1	0.1	0		level		false	false
1105	Heizkreispumpe1	bool		uint16be	1	1	0		level		false	false
1106	Mischerstellung1		%	uint16be	1	1	0		level		false	false
1107	Status1	enum		uint16be	1	1	0		level		false	false
1150	Vorlauftemperatur2	*10	°C	int16be	1	0.1	0		level		false	false
1155	Heizkreispumpe2	bool		uint16be	1	1	0		level		false	false
1156	Mischerstellung2		%	uint16be	1	1	0		level		false	false
1157	Status2	enum		uint16be	1	1	0		level		false	false
1900	PuffertemperaturOben	*10	°C	int16be	1	0.1	0		level		false	false
1901	PuffertemperaturUnten	*10	°C	int16be	1	0.1	0		level		false	false
1904	Pufferstatus	enum		uint16be	1	1	0		level		false	false
2400	Kesseltemperatur	*10		int16be	1	0.1	0		level		false	false
2401	KesselStatus	enum		uint16be	1	1	0		level		false	false
2406	Kesselreinigung		%	int16be	1	1	0		level		false	false
2407	AscheboxFuellstand		%	int16be	1	1	0		level		false	false
2408	Außentemperatur	*10	°C	int16be	1	0.1	0		level		false	false

Without 1100 everything works well.  1107 is 0.
If you add 1100, then 1107 has the value of 1155.

The device used is a Solarfocus Pellet Heater.
[German modbus documentation](https://www.solarfocus.com/produkte/Waermepumpe/EcoManager-t_Modbus-TCP-Registerdaten_Anleitung.pdf)

**Versions:**  
 - Adapter version: 5.0.4
 - JS-Controller version: 4.0.4
 - Node version: v12.18.0
 - Operating system: Win7 Pro

Apollon77, 2022-08-29:
Please show your settings ... Do you use alias? Then sometimes the IDs shift by 1 or such ...

TeslaNator, 2022-08-31:
First I have following config:
[image]
[image]

This value is corret.
[image]

Then I add following register:
[image]

The value 1107 is still correct. 1150 too.
[image]

Then I add register 1100.
[image]

Now 1107 is incorrect. It has the value of 1150 and even switch simultanously.
[image]