ioBroker/ioBroker.modbus #257: MODBUS SLAVE v5.0.4 - FC16 request outside holding registers boundaries...

Thomas-Kast, 2022-09-02:
Hallo, trotz intensiver Suche habe ich leider keinen Lösungsansatz gefunden und würde mich über Hilfe freuen. 

Im Einsatz ist der MODBUS-Adapter in v5.0.4 als SLAVE. Funktioniert auch alles, allerdings bekomme ich im Sekundentakt die Warnung:
**FC16 request outside holding registers boundaries: from 0, len 28. Expected max address 0**

Dabei beschreibe ich die Holding-Register per Script... Die dazugehörigen Datenpunkte sehen so aus:

[image]

Die Register im Adapter so:
[image]

Und das Script wie folgt:
[image]
[MODBUS_Script.log](https://github.com/ioBroker/ioBroker.modbus/files/9478199/MODBUS_Script.log)

Da fehlen mir wohl auch noch ein paar Basics 🤷‍♂️ . Könntet Ihr mir bitte einen Hinweis geben, wo ich ansetzen soll?

Vorab schon mal vielen Dank. Gruß, Thomas

an-klam, 2022-09-13:
Schon Mal versucht im Adapter beim Register bei Abfrage den Haken zu setzen?
Habe auch lange gebraucht um es zu finden.
Habe den Hinweis /die Lösung dann über die Suche im issued #204 gefunden.

Thomas-Kast, 2022-09-14:
Danke für Deine Hinweise. Ich habe die Links durchgearbeitet und auch mit dem Poll-Häkchen probiert. Die "outside boundaries" Meldung ist unverändert.

Was mir halt überhaupt nicht einleuchtet... Es wird sekündlich ins log geschrieben. Auch dann, wenn gar kein Master den Slave abfragt. Wer erzeugt denn überhaupt diese fehlerhafte Adressierung?

an-klam, 2023-01-30:
Hast du mal versucht nur ein einen Wert alle 30 Sekunden zu schreiben oder zwischen den werten eine Pause von zb. einer Sekunde einzufügen?
Kommt der Fehler dann immer noch?

Es ist nicht mein code und ich würde mich selbst definitiv noch als Einsteiger einstufen aber wenn ich das richtig sehe kommt die Fehlermedung aus [WriteMultipleRegisters.js](https://github.com/ioBroker/ioBroker.modbus/blob/cc6afd1c291a3ff1265b41e462a84b29c1977ea7/lib/jsmodbus/handler/server/WriteMultipleRegisters.js#L35) (Zeile 35) und wird gesetzt wenn du entweder eine Datenlänge von Null schreiben willst oder wenn du in einen Bereich ausserhalb deiner Register schreiben willst.

Was du auch noch machen könntest wäre: gucken ob du einfach nur den Wert 100 übertragen kannst und wenn ja, dann deinen Wert 1-0-1-8-0-zählerstand-bezug mit einem debug output angucken. Evtl ist die Zahl zu groß oder hat zuviele Nachkommastellen oder ähnliches.
just my 5 cents.