ioBroker/ioBroker.modbus #258: Option for complete Instance configuration share

Chris87-ux, 2022-09-06:
Hi @GermanBluefox,
great work with this addapter, i have 9 instances running with it now and there are mostly working perfect.

Never the less, I get several requests to share my "modbus-lists" which works fine if only Holding or input is used, but if you using all 4 chambers it is quiet dificult, as well don't miss the configuration ( Aliasses activated, device ID, summ to 16Bit,....).

 So, to sum up it would be nice , to share the complete config of a instance.
Thanks again
/Chris

Apollon77, 2022-09-06:
You mean "download it" or what?

Chris87-ux, 2022-09-06:
> You mean "download it" or what?

yes, in some kind of file, that others can upload it and have the same configuration as I had.