ioBroker/ioBroker.modbus #259: Modbus Master V5.0.4 Node.JS 18.7 Not Working

kevinhaybach, 2022-09-11:
Hi,
first of all this was my first GettingStarted and I became really desperate.
I tried to connect my WAGO PFC as client with IOBroker Modbus as Master.
I wa able to send one REAL value at address 0 but was not able to send anything else e.g. a second REAL value at address 2 or instead of the first REAL value a INT at 0.
Furthermore I was not able to receive any types / values.
After two days I downgraded node.js from 18.7 to 16.17 and everything is working fine. 

**Versions:**  
 - Adapter version: 5.0.4
 - JS-Controller version: <js-controller-version> 
 - Node version: 18.7
 - Operating system: Debian  Lite

Apollon77, 2022-09-12:
Thank you for the report. In fact nodejs 18.x is not yet official supported and also not yet LTS ... Were there any infos in log or such?

kevinhaybach, 2022-09-12:
Would be nice if you state somewhere for which version your software is released or tested with.

As I tried like a maniac and as I'm new to iobroker and modbus I didn't spend to much attention to the logs at some point there were errors shown in log but my configuration was wrong as well.

Apollon77, 2022-09-12:
Yes I know documentation is not yet ideal on iobroker but iti is there ... e.g. https://www.iobroker.net/#de/documentation/install/linux.md ... and in the Forum and such

kevinhaybach, 2023-09-23:
Funktioniert jetzt der MODBUS Adapter mit node.JS V18 nachdem nachdem seit März 2023 node.JS 18 für iobroker empfohlen wird?

Apollon77, 2023-09-23:
no issues beside this one report without too many details known

closed 2023-09-23 by Apollon77