ioBroker/ioBroker.modbus #260: Device ID as Slave

an-klam, 2022-09-13:
It seems that the adapter as slave does not make a difference between the device ID´s.

I configured the Adapter as slave and choose device ID 1

[image]

Holding register 11 is set to 45.
I can read the holding register and it seems to work fine **but** the adapter is answering every request comming at right IP and Port. Regardlessly of what is set as device ID (in the request telegram). Here you can see the Modbus Adapter answering to a request comming in as device ID 3.
[image]

Apollon77, 2022-09-14:
Please porvide a debug log and also enable the enhanced debug in settings

an-klam, 2022-09-14:
`
2022-09-14 12:59:52.495 - info: host.raspberrypi "system.adapter.modbus.2" disabled
--
2022-09-14 12:59:52.496 - info: host.raspberrypi stopInstance system.adapter.modbus.2 (force=false, process=true)
2022-09-14 12:59:52.507 - silly: modbus.2 (2342) States system redis pmessage system.adapter.modbus.2.sigKill/system.adapter.modbus.2.sigKill:{"val":-1,"ack":false,"ts":1663153192501,"q":0,"from":"system.host.raspberrypi","lc":1663153192501}
2022-09-14 12:59:52.520 - info: host.raspberrypi stopInstance system.adapter.modbus.2 send kill signal
2022-09-14 12:59:52.508 - info: modbus.2 (2342) Got terminate signal TERMINATE_YOURSELF
2022-09-14 12:59:52.511 - info: modbus.2 (2342) terminating
2022-09-14 12:59:52.513 - debug: modbus.2 (2342) Plugin sentry destroyed
2022-09-14 12:59:52.514 - info: modbus.2 (2342) Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
2022-09-14 12:59:52.619 - silly: modbus.2 (2342) States user redis pmessage modbus.2.*/modbus.2.info.connection:{"val":"","ack":true,"ts":1663153192615,"q":0,"from":"system.adapter.modbus.2","user":"system.user.admin","lc":1663153192615}
2022-09-14 12:59:53.123 - info: host.raspberrypi instance system.adapter.modbus.2 terminated with code 11 (ADAPTER_REQUESTED_TERMINATION)
2022-09-14 12:59:58.519 - info: host.raspberrypi "system.adapter.modbus.2" enabled
2022-09-14 12:59:58.741 - info: host.raspberrypi instance system.adapter.modbus.2 started with pid 2392
2022-09-14 13:00:00.156 - debug: modbus.2 (2392) Redis Objects: Use Redis connection: 127.0.0.1:9001
2022-09-14 13:00:00.298 - debug: modbus.2 (2392) Objects client ready ... initialize now
2022-09-14 13:00:00.301 - debug: modbus.2 (2392) Objects create System PubSub Client
2022-09-14 13:00:00.303 - debug: modbus.2 (2392) Objects create User PubSub Client
2022-09-14 13:00:00.382 - debug: modbus.2 (2392) Objects client initialize lua scripts
2022-09-14 13:00:00.416 - debug: modbus.2 (2392) Objects connected to redis: 127.0.0.1:9001
2022-09-14 13:00:00.421 - silly: modbus.2 (2392) redis psubscribe cfg.o.system.user.*
2022-09-14 13:00:00.555 - silly: modbus.2 (2392) redis psubscribe cfg.o.enum.*
2022-09-14 13:00:00.563 - silly: modbus.2 (2392) objectDB connected
2022-09-14 13:00:00.567 - debug: modbus.2 (2392) Redis States: Use Redis connection: 127.0.0.1:9000
2022-09-14 13:00:00.584 - debug: modbus.2 (2392) States create System PubSub Client
2022-09-14 13:00:00.586 - debug: modbus.2 (2392) States create User PubSub Client
2022-09-14 13:00:00.611 - debug: modbus.2 (2392) States connected to redis: 127.0.0.1:9000
2022-09-14 13:00:00.612 - silly: modbus.2 (2392) statesDB connected
2022-09-14 13:00:00.698 - debug: modbus.2 (2392) Plugin sentry Initialize Plugin (enabled=true)
2022-09-14 13:00:01.180 - info: modbus.2 (2392) starting. Version 5.0.4 in /opt/iobroker/node_modules/iobroker.modbus, node: v14.17.6, js-controller: 4.0.23
2022-09-14 13:00:01.347 - silly: modbus.2 (2392) States system redis pmessage system.adapter.modbus.2.logLevel/system.adapter.modbus.2.logLevel:{"val":"silly","ack":true,"ts":1663153201330,"q":0,"from":"system.adapter.modbus.2","lc":1663153146161}
2022-09-14 13:00:01.543 - debug: modbus.2 (2392) Iterate holdingRegs holdingRegisters: 0 = holdingRegisters.0_L1_V_N
2022-09-14 13:00:01.546 - debug: modbus.2 (2392) Iterate holdingRegs holdingRegisters: 2 = holdingRegisters.2_L2_V_N
2022-09-14 13:00:01.551 - debug: modbus.2 (2392) Iterate holdingRegs holdingRegisters: 4 = holdingRegisters.4_L3_V_N
2022-09-14 13:00:01.558 - debug: modbus.2 (2392) Iterate holdingRegs holdingRegisters: 6 = holdingRegisters.6_L1_L2_V
2022-09-14 13:00:01.560 - debug: modbus.2 (2392) Iterate holdingRegs holdingRegisters: 8 = holdingRegisters.8_L2_L3_V
2022-09-14 13:00:01.561 - debug: modbus.2 (2392) Iterate holdingRegs holdingRegisters: 10 = holdingRegisters.10_L3_L1_V
2022-09-14 13:00:01.561 - debug: modbus.2 (2392) Iterate holdingRegs holdingRegisters: 11 = holdingRegisters.11_identification_code
2022-09-14 13:00:01.562 - debug: modbus.2 (2392) Iterate holdingRegs holdingRegisters: 12 = holdingRegisters.12_L1_A
2022-09-14 13:00:01.563 - debug: modbus.2 (2392) Iterate holdingRegs holdingRegisters: 14 = holdingRegisters.14_L2_A
2022-09-14 13:00:01.563 - debug: modbus.2 (2392) Iterate holdingRegs holdingRegisters: 16 = holdingRegisters.16_L3_A
2022-09-14 13:00:01.564 - debug: modbus.2 (2392) Iterate holdingRegs holdingRegisters: 18 = holdingRegisters.18_L1_W
2022-09-14 13:00:01.565 - debug: modbus.2 (2392) Iterate holdingRegs holdingRegisters: 20 = holdingRegisters.20_L2_W
2022-09-14 13:00:01.571 - debug: modbus.2 (2392) Iterate holdingRegs holdingRegisters: 22 = holdingRegisters.22_L3_W
2022-09-14 13:00:01.572 - debug: modbus.2 (2392) Iterate holdingRegs holdingRegisters: 24 = holdingRegisters.24_L1_VA
2022-09-14 13:00:01.573 - debug: modbus.2 (2392) Iterate holdingRegs holdingRegisters: 26 = holdingRegisters.26_L2_VA
2022-09-14 13:00:01.574 - debug: modbus.2 (2392) Iterate holdingRegs holdingRegisters: 28 = holdingRegisters.28_L3_VA
2022-09-14 13:00:01.574 - debug: modbus.2 (2392) Iterate holdingRegs holdingRegisters: 30 = holdingRegisters.30_L1_var
2022-09-14 13:00:01.575 - debug: modbus.2 (2392) Iterate holdingRegs holdingRegisters: 32 = holdingRegisters.32_L2_var
2022-09-14 13:00:01.576 - debug: modbus.2 (2392) Iterate holdingRegs holdingRegisters: 34 = holdingRegisters.34_L2_var
[... cut ...]
2022-09-14 13:00:02.262 - debug: modbus.2 (2392) initiating write single register request handler.
2022-09-14 13:00:02.263 - debug: modbus.2 (2392) setting request handler
2022-09-14 13:00:04.513 - debug: modbus.2 (2392) new connection
2022-09-14 13:00:04.514 - debug: modbus.2 (2392) + Clients connected: 192.168.100.166
2022-09-14 13:00:04.527 - silly: modbus.2 (2392) States user redis pmessage modbus.2.*/modbus.2.info.connection:{"val":"192.168.100.166","ack":true,"ts":1663153204523,"q":0,"from":"system.adapter.modbus.2","user":"system.user.admin","lc":1663153204523}
`

an-klam, 2022-09-16:
Thats not the reason for the issue but:
i think you made a typo on [modbus-server-tcp.js](https://github.com/ioBroker/ioBroker.modbus/tree/master/lib/jsmodbus/transports)/modbus-server-tcp.js) row 68 and row 109 **untiId** instead of **unitId**

an-klam, 2023-01-30:
By the way, I solved it for me like this:

In the file [modbus-server-tcp.js](https://github.com/ioBroker/ioBroker.modbus/blob/master/lib/jsmodbus/transports/modbus-server-tcp.js#L81)
I have packed line 81 into an If.
So it becomes from
```

                    const pdu = buffer.slice(7, 7 + len - 1);

                    // output the data event and let the
                    // listener to process the pdu

                    fifo.push({request, pdu, socket});

                    flush();

                    buffer = buffer.slice(pdu.length + 7, buffer.length);

```
on:
```

                   const pdu = buffer.slice(7, 7 + len - 1);

                    // emit data event and let the
                    // listener to process the pdu
if (request.unitId==1) {
                    fifo.push({request, pdu, socket});
}
                    flush();

                    buffer = buffer.slice(pdu.length + 7, buffer.length);
                }

```

if your slave has a different ID, of course you have to enter this ID instead of the 1.

LosPimpos, 2023-02-21:
Hallo,

ich habe auch das Problem. Adapter läuft als Slave...bei ID kann ich reinschreiben was ich will, er behält immer die ID 1. Kann das Problem gefixt werden?

mosmutzel, 2023-11-12:
Hi,
i have the same issue. 
[image]

I have setup as Slave with ID2 and answer was on ID1 instead of ID2.