ioBroker/ioBroker.modbus #261: errorCode and exceptionCode descriptions for  {"errorCode":131,"exceptionCode":229} and  ...226}

christi-ko, 2022-09-19:
**Describe the bug**  
I am getting several warning in the log and don't know how i may change the settings to make them not appear. The communication is working as expected.

```
modbus.0 | 2022-09-19 16:01:19.504 | warn | Poll error count: 1 code: {"errorCode":131,"exceptionCode":229}
modbus.0 | 2022-09-19 16:01:16.230 | warn | Poll error count: 1 code: {"errorCode":131,"exceptionCode":226}
```

**To Reproduce**  
It's appearing without changing the standard settings from initial values. 
 What may be special about my setup:
I have a Dimplex LWPM 410 Modbus RTU modul  communicating with an ESP32 Modbus Gateway SW from here:
https://github.com/zivillian/esp32-modbus-gateway via WLAN with the iobroker.

**Expected behavior**  
This setup may have some time delay or connection problems and i want to know what can be changed to make it work without that big amount of warning.

**Screenshots & Logfiles**  
N/A

**Versions:**  
 - Adapter version: v5.0.4
 - JS-Controller version: 4.0.23
 - Node version: v16.17.0
 - Operating system: ubuntu

**Additional context**  
Add any other context about the problem here.

Apollon77, 2022-09-19:
These codes are returned from the device. Does the manuals of it have infos about their meaning? Maybe ask in the linked GitHub project for the potential reaons for the errors?. I would not know how we can support here. Maabe enable debug in the device andncheck what this shows?

Maybe 229 meanas 0xE5 (https://github.com/eModbus/eModbus/blob/master/src/ModbusError.h#L90) ?? 0xE2 could be then https://github.com/eModbus/eModbus/blob/master/src/ModbusError.h#L81

Without more details I see no chance to support here

christi-ko, 2022-09-20:
Hi Apollo, thanks for your answer. This is exactly the point what i needed. I will contact the GitHub project for more hints.

closed 2022-09-20 by christi-ko