ioBroker/ioBroker.modbus #262: timing QUESTION

an-klam, 2022-09-22:
Its more a question than a bug.

I´m trying to emulate a Carlo Gavazzi EM24. (Modbusadapter is Slave)
Correct registers are read and send.
Although the devices datasheed that requests the data (Inverter) says request time up to 500ms is allowed, the inverter does not work. 
Original EM24 answering in under 80ms (typical 20-60ms)
Time between request and response with this modubus adapter is avg. 110ms (117-122ms)

After testing a few things i found out this part takes around 105ms.
Even after stopping all other networktraffic and after stopping all other IoBroker adapters.

```
            this.onData(current.pdu, response => {
                this.log.debug('sending tcp data');
                let pkt = Put()
                    .word16be(current.request.transId)     // transaction id
                    .word16be(current.request.protocolVer) // protocol version
                    .word16be(response.length + 1)         // pdu length
                    .word8(current.request.untiId)         // unit id
                    .put(response)                         // the actual pdu
                    .buffer();
                current.socket.write(pkt);
                this.setState('ready');
            });
```

Is it possible to fast this up?
Maybe that is not the reason but for me its worth a try.

Maybe you can post code here and i´m going to test it on my system.

an-klam, 2022-09-26:
Found the reason but dont know how to solve it correctly.

[slave.js](https://github.com/ioBroker/ioBroker.modbus/blob/master/lib/slave.js)
on line 105 responseDelay is set to 100ms

i manualy set it to 10 and now the response takes 12-18ms.

Now the Inverter is working as expected.

HansPeterMeier, 2022-10-01:
Ne, leider nicht, ich bekomme nicht mal eine Verbindung hin.
Node Red sowie Home Assistant funktionieren.

`<html>
<body>

modbus.2 | 2022-10-01 11:51:53.102 | debug | Closing client on purpose.
-- | -- | -- | --
modbus.2 | 2022-10-01 11:51:52.103 | debug | Cleaning up request fifo.
modbus.2 | 2022-10-01 11:51:52.102 | debug | Socket closed with error
modbus.2 | 2022-10-01 11:51:52.079 | info | Disconnected from slave 10.10.10.65
modbus.2 | 2022-10-01 11:51:52.079 | debug | Closing client on purpose.
modbus.2 | 2022-10-01 11:51:52.078 | info | Connected to slave 10.10.10.65
modbus.2 | 2022-10-01 11:51:43.102 | debug | Closing client on purpose.
modbus.2 | 2022-10-01 11:51:42.102 | debug | Cleaning up request fifo.
modbus.2 | 2022-10-01 11:51:42.102 | debug | Socket closed with error
modbus.2 | 2022-10-01 11:51:42.076 | info | Disconnected from slave 10.10.10.65
modbus.2 | 2022-10-01 11:51:42.076 | debug | Closing client on purpose.
modbus.2 | 2022-10-01 11:51:42.075 | info | Connected to slave 10.10.10.65
modbus.2 | 2022-10-01 11:51:33.203 | debug | Closing client on purpose.
modbus.2 | 2022-10-01 11:51:32.202 | debug | Cleaning up request fifo.
modbus.2 | 2022-10-01 11:51:32.202 | debug | Socket closed with error
modbus.2 | 2022-10-01 11:51:32.072 | info | Disconnected from slave 10.10.10.65
modbus.2 | 2022-10-01 11:51:32.072 | debug | Closing client on purpose.
modbus.2 | 2022-10-01 11:51:32.072 | info | Connected to slave 10.10.10.65
modbus.2 | 2022-10-01 11:51:23.091 | debug | Closing client on purpose.
modbus.2 | 2022-10-01 11:51:22.092 | debug | Cleaning up request fifo.
modbus.2 | 2022-10-01 11:51:22.092 | debug | Socket closed with error
modbus.2 | 2022-10-01 11:51:22.069 | info | Disconnected from slave 10.10.10.65

</body>
</html>`