ioBroker/ioBroker.modbus #263: modbus_cli funktioniert, Adapter nicht

olant, 2022-09-24:
Ich habe den modbus Adapter installiert, um mein Kostal Smart Energy Meter (KSEM) abzufragen. Mit dem Standard-Python-Tool modbus (modbus_cli) kann ich zB Register 40072 einfach abfragen:
```
root@olserv:/modbus# modbus ksem.fritz.box -s 74 40072/H
Parsed 0 registers definitions from 1 files
40072: 156 0x9c
```
Aber der Modbus Adapter setzt hartnäckig alle Datenpunte auf "0".

Ich sehe im debug-Logs, dass der Adapter etewas tut (gekürzt):
```
2022-09-19 22:09:52.660 - debug: modbus.0 (31295) Initialize Objects for holdingRegs: {"_address":40072,"name":"cur_b","description":"","unit":"","type":"uint16be","len":1,"factor":1,"offset":0,"formula":"","role":"value","room":"","poll":true,"wp":false,"cw":false,"isScale":false,"deviceId":0,"address":71,"id":"holdingRegisters.40072_cur_b"},2022-09-19 22:09:52.661 - debug: modbus.0 (31295) Add holdingRegisters.40071_cur_a: 2022-09-19 22:09:52.661 - debug: modbus.0 (31295) Add holdingRegisters.40072_cur_b: {"_id":"holdingRegisters.40072_cur_b","type":"state","common":{"name":"","role":"value","type":"number","read":true,"write":true,"def":0,"unit":""},"native":{"regType":"holdingRegs","address":71,"deviceId":0,"type":"uint16be","len":1,"offset":0,"factor":1,"poll":true}}
2022-09-19 22:09:53.020 - info: modbus.0 (31295) Connected to slave [192.168.188.20](https://l.facebook.com/l.php?u=https%3A%2F%2F192.168.188.20%2F%3Ffbclid%3DIwAR1v2Atqhq6PGQg0psep3dHTUoa1-9VJLyk5S-4rMxDg7-FQuebtZBEkrfg&h=AT1OTxaIhXIA0rVbmlRt-jvFBi0A--agdzsO7_NLpqhJlFAZk1yt920s5zMTp9MB9WrmX8vKCm47hY9YqgsPjnU1B6RfNGHeCsaL4tdJfSPAuRPK0-6t2oFSOIfeoadBY1j7kvKe-MUJOjgZiA&__tn__=-UK-R&c[0]=AT2t0RVF6NBR5EhukCX8ThaHRULKCzAlNMaZvka_CzVBZ4kusOfcG6gZaglt2gyrQdMAs_lLrA-kGti71_VE8g_dp4w_HsGAMYr_rCSDzOTt4SJ3ruUr4GmC0XiC0pFVbVxiqwQf-4vTV_OOQhZb2Vl2t8OpULXeBaM)
2022-09-19 22:09:53.026 - debug: modbus.0 (31295) [DevID_0] Poll start ---------------------
2022-09-19 22:09:53.028 - debug: modbus.0 (31295) Initialization of scale factors done!
2022-09-19 22:09:53.028 - debug: modbus.0 (31295) [DevID_0/holdingRegs] Poll address 70 - 3 registers
2022-09-19 22:09:53.044 - debug: modbus.0 (31295) [DevID_0/holdingRegs] Poll address 70 DONE
```
Was mache ich falsch?

Apollon77, 2022-09-24:
Bitte vollständiges Debug log posten und auch die EInstellungen

olant, 2022-09-25:
Okay, hier das debug-log:
```
2022-09-25 20:39:10.570  - info: host.dagobert "system.adapter.modbus.0" enabled
2022-09-25 20:39:10.738  - info: host.dagobert instance system.adapter.modbus.0 started with pid 22856
2022-09-25 20:39:12.119  - debug: modbus.0 (22856) Redis Objects: Use Redis connection: 127.0.0.1:9001
2022-09-25 20:39:12.236  - debug: modbus.0 (22856) Objects client ready ... initialize now
2022-09-25 20:39:12.239  - debug: modbus.0 (22856) Objects create System PubSub Client
2022-09-25 20:39:12.240  - debug: modbus.0 (22856) Objects create User PubSub Client
2022-09-25 20:39:12.305  - debug: modbus.0 (22856) Objects client initialize lua scripts
2022-09-25 20:39:12.313  - debug: modbus.0 (22856) Objects connected to redis: 127.0.0.1:9001
2022-09-25 20:39:12.353  - debug: modbus.0 (22856) Redis States: Use Redis connection: 127.0.0.1:9000
2022-09-25 20:39:12.374  - debug: modbus.0 (22856) States create System PubSub Client
2022-09-25 20:39:12.375  - debug: modbus.0 (22856) States create User PubSub Client
2022-09-25 20:39:12.425  - debug: modbus.0 (22856) States connected to redis: 127.0.0.1:9000
2022-09-25 20:39:12.570  - debug: modbus.0 (22856) Plugin sentry Initialize Plugin (enabled=true)
2022-09-25 20:39:13.060  - info: modbus.0 (22856) starting. Version 5.0.4 in /opt/iobroker/node_modules/iobroker.modbus, node: v14.20.0, js-controller: 4.0.23
2022-09-25 20:39:13.250  - debug: modbus.0 (22856) Initialize Objects for disInputs: []
2022-09-25 20:39:13.251  - debug: modbus.0 (22856) Initialize Objects for coils: []
2022-09-25 20:39:13.252  - debug: modbus.0 (22856) Initialize Objects for inputRegs: []
2022-09-25 20:39:13.252  - debug: modbus.0 (22856) Initialize Objects for holdingRegs: [{"_address":40071,"name":"cur_a","description":"","unit":"","type":"uint16be","len":1,"factor":1,"offset":0,"formula":"","role":"value","room":"","poll":true,"wp":false,"cw":false,"isScale":false,"deviceId":71,"address":70,"id":"holdingRegisters.40071_cur_a"},{"_address":40072,"name":"cur_b","description":"","unit":"","type":"uint16be","len":1,"factor":1,"offset":0,"formula":"","role":"value","room":"","poll":true,"wp":false,"cw":false,"isScale":false,"deviceId":71,"address":71,"id":"holdingRegisters.40072_cur_b"},{"_address":40073,"name":"cur_c","description":"","unit":"","type":"uint16be","len":1,"factor":1,"offset":0,"formula":"","role":"level","room":"","poll":true,"wp":false,"cw":false,"isScale":false,"address":72,"deviceId":71,"id":"holdingRegisters.40073_cur_c"},{"_address":40078,"name":"volt_b","description":"","unit":"","type":"uint16be","len":1,"factor":0.01,"offset":0,"formula":"","role":"level","room":"","poll":true,"wp":false,"cw":false,"isScale":false,"deviceId":71,"address":77,"id":"holdingRegisters.40078_volt_b"}]
2022-09-25 20:39:13.253  - debug: modbus.0 (22856) Add holdingRegisters.40071_cur_a: {"_id":"holdingRegisters.40071_cur_a","type":"state","common":{"name":"","role":"value","type":"number","read":true,"write":true,"def":0,"unit":""},"native":{"regType":"holdingRegs","address":70,"deviceId":71,"type":"uint16be","len":1,"offset":0,"factor":1,"poll":true}}
2022-09-25 20:39:13.254  - debug: modbus.0 (22856) Add holdingRegisters.40072_cur_b: {"_id":"holdingRegisters.40072_cur_b","type":"state","common":{"name":"","role":"value","type":"number","read":true,"write":true,"def":0,"unit":""},"native":{"regType":"holdingRegs","address":71,"deviceId":71,"type":"uint16be","len":1,"offset":0,"factor":1,"poll":true}}
2022-09-25 20:39:13.254  - debug: modbus.0 (22856) Add holdingRegisters.40073_cur_c: {"_id":"holdingRegisters.40073_cur_c","type":"state","common":{"name":"","role":"level","type":"number","read":true,"write":true,"def":0,"unit":""},"native":{"regType":"holdingRegs","address":72,"deviceId":71,"type":"uint16be","len":1,"offset":0,"factor":1,"poll":true}}
2022-09-25 20:39:13.255  - debug: modbus.0 (22856) Add holdingRegisters.40078_volt_b: {"_id":"holdingRegisters.40078_volt_b","type":"state","common":{"name":"","role":"level","type":"number","read":true,"write":true,"def":0,"unit":""},"native":{"regType":"holdingRegs","address":77,"deviceId":71,"type":"uint16be","len":1,"offset":0,"factor":0.01,"poll":true}}
2022-09-25 20:39:13.569  - info: modbus.0 (22856) Connected to slave 192.168.188.20
2022-09-25 20:39:13.570  - debug: modbus.0 (22856) [DevID_71] Poll start ---------------------
2022-09-25 20:39:13.573  - debug: modbus.0 (22856) Initialization of scale factors done!
2022-09-25 20:39:13.573  - debug: modbus.0 (22856) [DevID_71/holdingRegs] Poll address 70 - 8 registers
2022-09-25 20:39:13.631  - debug: modbus.0 (22856) [DevID_71/holdingRegs] Poll address 70 DONE
2022-09-25 20:39:14.688  - debug: modbus.0 (22856) [DevID_71] Poll start ---------------------
2022-09-25 20:39:14.689  - debug: modbus.0 (22856) [DevID_71/holdingRegs] Poll address 70 - 8 registers
```

Settings:

- TCP
-  Partner 192.168.188.20
- Port 502
- Geräte ID 71
- Typ Master

Ich habe Holding-Register definiert:
- 40071 cur_a unsigned 16bit (big endian) 1 
- 40072 cur_a unsigned 16bit (big endian) 1 
- 40073 cur_a unsigned 16bit (big endian) 1 
- 40078 volt_b unsigned 16bit (big endian) 1 

Die Datenpunkte bleiben alle auf "0" (ack, Qualität 0x00)