ioBroker/ioBroker.modbus #265: falsche Werte bei Phasenleistung

tis-cs, 2022-09-28:
**Describe the bug**  
in den drei Registern für die Phasenleistung wir bei einer Einspeisung durch eine E3/DC S10 E nicht der korrekte Wert angezeigt.
Es sieht so aus, als wenn er bei einem negativen Wert dann 65000 bzgl. des negativen Werts anzeigt, was diese drei Werte somit unbrauchbar macht.

Register:
modbus.0.holdingRegisters.40106_Phasenleistung_in_Watt_L1_L0
modbus.0.holdingRegisters.40107_Phasenleistung_in_Watt_L2_L0
modbus.0.holdingRegisters.40108_Phasenleistung_in_Watt_L3_L0

Beiläufig ist mir aufgefallen, dass der Wert Autarkie ebenfalls falsch darstellt wird (25443%). Dieser müsste aktuell 100% zeigen.

modbus.0.holdingRegisters.40082_Autarkie_Eigenverbrauch

**To Reproduce**  
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '...'
3. Scroll down to '....'
4. See error

**Expected behavior**  
A clear and concise description of what you expected to happen.

**Screenshots & Logfiles**  
[image]

**Versions:**  
 - Adapter version: 5.0.5
 - JS-Controller version: 4.0.23
 - Node version: <node-version> v16.17.1
 - Operating system: Ubuntu 22.04.1 LTS

**Additional context**  
Add any other context about the problem here.

Apollon77, 2022-09-28:
Hm ... das weisst normalerweise darauf hin das die Register Nummern nicht korrekt sind. Ggf wenn Alias genutzt wird müssen die ggf um eins versetzt angegeben werden (also +1) Bitte mal damit experimentieren.

Sonst brauchen wir bitte die Specs der modbus register, ein Debug Log (inklusive aktiviertem erweiterten debug modus in den settings).

tis-cs, 2022-10-04:
Ich habe es nun so konfiguriert, sodass ich endlich auf allen drei Phasen korrekte Werte angezeigt bekomme:

Apollon77, 2022-12-29:
Wenn es für dich tut ist erstmal alles ok - was war jertzt "der Trick"?

closed 2022-12-29 by Apollon77

tis-cs, 2022-12-30:
> Wenn es für dich tut ist erstmal alles ok - was war jertzt "der Trick"?

Ich habe den Typ auf "Signed 16 bit (Big Endian)" gestellt. Länge und Faktor jeweils auf "1"

Apollon77, 2022-12-30:
Also war vorher der falsche Datentyp gewählt und damit wurden die Zahlen falsch interpretiert?

tis-cs, 2022-12-30:
Ich weiß nicht mehr genau, was default war. Sorry

Apollon77, 2022-12-30:
Wnn du magst stelle deine einstellungen gern hier zur verfpgung, dann machen andere user vllt nicht den gleichen fehler :-) https://github.com/ioBroker/modbus-templates