ioBroker/ioBroker.modbus #268: Modbus Client, datawords not updated

nublike, 2022-10-12:
Datawords are only updated ones when starting the modbus instance, then never again.

modbus Version v5.05

what am I doing wrong?
Did try several settings...

[image]

[image]

[image]

[image]

Apollon77, 2022-10-12:
What exactly is not updated? datawords? What you mean? Please provide a delug level log

nublike, 2022-10-13:
Sorry for misunderstanding.
The Values/Objects in ioBroker are shown and updated correctly.
But when I connect with a modbus master (Qmodmaster or other) the read values stay to the old ones from when I start my modbus.2 instance in ioBroker and are never updated.

nublike, 2022-10-13:
`22-10-12 22:08:02.163 - debug: modbus.2 (21510) Redis Objects: Use Redis connection: 127.0.0.1:9001
2022-10-12 22:08:02.193 - debug: modbus.2 (21510) Objects client ready ... initialize now
2022-10-12 22:08:02.194 - debug: modbus.2 (21510) Objects create System PubSub Client
2022-10-12 22:08:02.195 - debug: modbus.2 (21510) Objects create User PubSub Client
2022-10-12 22:08:02.242 - debug: modbus.2 (21510) Objects client initialize lua scripts
2022-10-12 22:08:02.246 - debug: modbus.2 (21510) Objects connected to redis: 127.0.0.1:9001
2022-10-12 22:08:02.269 - debug: modbus.2 (21510) Redis States: Use Redis connection: 127.0.0.1:9000
2022-10-12 22:08:02.281 - debug: modbus.2 (21510) States create System PubSub Client
2022-10-12 22:08:02.282 - debug: modbus.2 (21510) States create User PubSub Client
2022-10-12 22:08:02.341 - debug: modbus.2 (21510) States connected to redis: 127.0.0.1:9000
2022-10-12 22:08:02.449 - debug: modbus.2 (21510) Plugin sentry Initialize Plugin (enabled=true)
2022-10-12 22:08:02.589 - info: modbus.2 (21510) starting. Version 5.0.5 in /opt/iobroker/node_modules/iobroker.modbus, node: v14.20.1, js-controller: 4.0.23
2022-10-12 22:08:02.673 - debug: modbus.2 (21510) Iterate inputRegs inputRegisters: 0 = inputRegisters.3000_Running_State
2022-10-12 22:08:02.673 - debug: modbus.2 (21510) Iterate inputRegs inputRegisters: 9 = inputRegisters.3009_Export_power
2022-10-12 22:08:02.673 - debug: modbus.2 (21510) Iterate inputRegs inputRegisters: 21 = inputRegisters.3021_Battery_power_
2022-10-12 22:08:02.674 - debug: modbus.2 (21510) Iterate inputRegs inputRegisters: 22 = inputRegisters.3022_Battery_level_
2022-10-12 22:08:02.674 - debug: modbus.2 (21510) Iterate inputRegs inputRegisters: 2016 = inputRegisters.5016_Total_DC_Power
2022-10-12 22:08:02.674 - debug: modbus.2 (21510) Iterate inputRegs inputRegisters: 2030 = inputRegisters.5030_Output_Power
2022-10-12 22:08:02.674 - debug: modbus.2 (21510) Iterate holdingRegs holdingRegisters: 0 = holdingRegisters.3000_Running_State
2022-10-12 22:08:02.675 - debug: modbus.2 (21510) Iterate holdingRegs holdingRegisters: 9 = holdingRegisters.3009_Export_power
2022-10-12 22:08:02.675 - debug: modbus.2 (21510) Iterate holdingRegs holdingRegisters: 21 = holdingRegisters.3021_Battery_power_
2022-10-12 22:08:02.675 - debug: modbus.2 (21510) Iterate holdingRegs holdingRegisters: 22 = holdingRegisters.3022_Battery_level_
2022-10-12 22:08:02.675 - debug: modbus.2 (21510) Iterate holdingRegs holdingRegisters: 2016 = holdingRegisters.5016_Total_DC_Power
2022-10-12 22:08:02.675 - debug: modbus.2 (21510) Iterate holdingRegs holdingRegisters: 2030 = holdingRegisters.5030_Output_Power
2022-10-12 22:08:02.676 - debug: modbus.2 (21510) Initialize Objects for disInputs: []
2022-10-12 22:08:02.676 - debug: modbus.2 (21510) Initialize Objects for coils: []
2022-10-12 22:08:02.676 - debug: modbus.2 (21510) Initialize Objects for inputRegs: [{"_address":3000,"name":"Running State","description":"Betriebsstatus","unit":"","type":"uint16be","len":1,"factor":1,"offset":0,"formula":"","role":"value","room":"","cw":true,"isScale":false,"address":3000,"deviceId":1,"id":"inputRegisters.3000_Running_State"},{"_address":3009,"name":"Export power","description":"Aktuelle Leistung am Übergabepunkt des Versorgungsnetzes","unit":"W","type":"int32sw","len":2,"factor":1,"offset":0,"formula":"","role":"value","room":"","cw":true,"isScale":false,"address":3009,"deviceId":1,"id":"inputRegisters.3009_Export_power"},{"_address":3021,"name":"Battery power ","description":"Batterieladeleistung","unit":"W","type":"uint16be","len":1,"factor":1,"offset":0,"formula":"","role":"value","room":"","cw":true,"isScale":false,"address":3021,"deviceId":1,"id":"inputRegisters.3021_Battery_power_"},{"_address":3022,"name":"Battery level ","description":"Batteriekapazität","unit":"%","type":"uint16be","len":1,"factor":0.1,"offset":0,"formula":"","role":"value","room":"","cw":true,"isScale":false,"address":3022,"deviceId":1,"id":"inputRegisters.3022_Battery_level_"},{"_address":5016,"name":"Total DC Power","description":"PV-Leistung aktuell","unit":"W","type":"uint32sw","len":2,"factor":1,"offset":0,"formula":"","role":"value","room":"","cw":true,"isScale":false,"deviceId":1,"address":5016,"id":"inputRegisters.5016_Total_DC_Power"},{"_address":5030,"name":"Output Power","description":"PV Leistung","unit":"W","type":"int32sw","len":2,"factor":1,"offset":0,"formula":"","role":"value","room":"","cw":true,"isScale":false,"deviceId":1,"address":5030,"id":"inputRegisters.5030_Output_Power"}]
2022-10-12 22:08:02.677 - debug: modbus.2 (21510) Add inputRegisters.3000_Running_State: {"_id":"inputRegisters.3000_Running_State","type":"state","common":{"name":"Betriebsstatus","role":"value","type":"number","read":true,"write":true,"def":0,"unit":""},"native":{"regType":"inputRegs","address":3000,"deviceId":1,"type":"uint16be","len":1,"offset":0,"factor":1}}
2022-10-12 22:08:02.677 - debug: modbus.2 (21510) Add inputRegisters.3009_Export_power: {"_id":"inputRegisters.3009_Export_power","type":"state","common":{"name":"Aktuelle Leistung am Übergabepunkt des Versorgungsnetzes","role":"value","type":"number","read":true,"write":true,"def":0,"unit":"W"},"native":{"regType":"inputRegs","address":3009,"deviceId":1,"type":"int32sw","len":2,"offset":0,"factor":1}}
[... cut ...]
2022-10-12 22:08:46.965 - debug: modbus.2 (21510) sending tcp data
2022-10-12 22:08:47.954 - debug: modbus.2 (21510) sending tcp data
2022-10-12 22:08:48.962 - debug: modbus.2 (21510) sending tcp data
2022-10-12 22:08:49.975 - debug: modbus.2 (21510) sending tcp data
2022-10-12 22:08:50.973 - debug: modbus.2 (21510) sending tcp data
2022-10-12 22:08:51.982 - debug: modbus.2 (21510) sending tcp data
2022-10-12 22:08:51.987 - debug: modbus.2 (21510) - Client connected: 192.168.178.132
2022-10-12 22:08:51.988 - debug: modbus.2 (21510) connection closed, socket
`