ioBroker/ioBroker.modbus #289: Poll error count: 1019 code: {"err":"timeout","timeout":5000}

Sneak-L8, 2022-12-11:
**Describe the bug**  
Since two days I'm getting plenty of the following errors:

`modbus.1 | 2022-12-11 21:20:51.606 | info | (COMPACT) setState not processed because States database not connected
modbus.1 | 2022-12-11 21:20:51.606 | info | (COMPACT) Disconnected from slave 192.168.97.46
modbus.1 | 2022-12-11 21:20:50.608 | error | (COMPACT) Reconnect did not help, restart adapter
modbus.1 | 2022-12-11 21:20:50.607 | info | (COMPACT) setState not processed because States database not connected
modbus.1 | 2022-12-11 21:20:50.607 | warn | (COMPACT) Poll error count: 1019 code: {"err":"timeout","timeout":5000}
modbus.1 | 2022-12-11 21:20:50.606 | error | (COMPACT) Client in error state.
modbus.1 | 2022-12-11 21:20:50.605 | error | (COMPACT) Request timed out.
modbus.1 | 2022-12-11 21:20:50.604 | warn | (COMPACT) Error: undefined
`

**To Reproduce**  
Steps to reproduce the behavior:
Restart adapter and wait some minute and the errors will occur.

**Expected behavior**  
No errors should be shown

**Screenshots & Logfiles**  
See above

**Versions:**  
 - Adapter version: 5.0.5, also tried 5.0.11 from git.
 - JS-Controller version: 4.0.23
 - Node version: v16.16.0
 - Operating system: linux

**Additional context**  
No updes were alpplied to js-controller, admin of the adapter in the last weeks.

Apollon77, 2022-12-12:
Please post a more long debug log from such a case happening

Sneak-L8, 2022-12-12:
No problem, here you are:

`<html><body>

modbus.1 | 2022-12-12 10:13:13.105 | info | (COMPACT) setState not processed because States database not connected
-- | -- | -- | --
modbus.1 | 2022-12-12 10:13:13.104 | info | (COMPACT) Disconnected from slave 192.168.97.46
modbus.1 | 2022-12-12 10:13:12.105 | error | (COMPACT) Reconnect did not help, restart adapter
modbus.1 | 2022-12-12 10:13:12.105 | info | (COMPACT) setState not processed because States database not connected
modbus.1 | 2022-12-12 10:13:12.105 | warn | (COMPACT) Poll error count: 1721 code: {"err":"timeout","timeout":5000}
modbus.1 | 2022-12-12 10:13:12.104 | error | (COMPACT) Client in error state.
modbus.1 | 2022-12-12 10:13:12.104 | error | (COMPACT) Request timed out.
modbus.1 | 2022-12-12 10:13:12.103 | warn | (COMPACT) Error: undefined
modbus.1 | 2022-12-12 10:13:07.102 | info | (COMPACT) setState not processed because States database not connected
modbus.1 | 2022-12-12 10:13:07.102 | info | (COMPACT) Connected to slave 192.168.97.46
modbus.1 | 2022-12-12 10:12:07.096 | info | (COMPACT) setState not processed because States database not connected
modbus.1 | 2022-12-12 10:12:07.095 | info | (COMPACT) Disconnected from slave 192.168.97.46
modbus.1 | 2022-12-12 10:12:06.096 | error | (COMPACT) Reconnect did not help, restart adapter
modbus.1 | 2022-12-12 10:12:06.096 | info | (COMPACT) setState not processed because States database not connected
modbus.1 | 2022-12-12 10:12:06.096 | warn | (COMPACT) Poll error count: 1720 code: {"err":"timeout","timeout":5000}
modbus.1 | 2022-12-12 10:12:06.095 | error | (COMPACT) Client in error state.
modbus.1 | 2022-12-12 10:12:06.095 | error | (COMPACT) Request timed out.
modbus.1 | 2022-12-12 10:12:06.094 | warn | (COMPACT) Error: undefined
modbus.1 | 2022-12-12 10:12:01.093 | info | (COMPACT) setState not processed because States database not connected
modbus.1 | 2022-12-12 10:12:01.092 | info | (COMPACT) Connected to slave 192.168.97.46
web.0 | 2022-12-12 10:11:54.858 | info | (COMPACT) <== Disconnect system.user.admin from ::ffff:192.168.97.65 flot
web.0 | 2022-12-12 10:11:54.857 | info | (COMPACT) <== Disconnect system.user.admin from ::ffff:192.168.97.65 flot
web.0 | 2022-12-12 10:11:54.848 | info | (COMPACT) <== Disconnect system.user.admin from ::ffff:192.168.97.65 vis.0
web.0 | 2022-12-12 10:11:54.846 | info | (COMPACT) <== Disconnect system.user.admin from ::ffff:192.168.97.65 flot
modbus.1 | 2022-12-12 10:11:01.088 | info | (COMPACT) setState not processed because States database not connected
modbus.1 | 2022-12-12 10:11:01.087 | info | (COMPACT) Disconnected from slave 192.168.97.46
modbus.1 | 2022-12-12 10:11:00.091 | error | (COMPACT) Reconnect did not help, restart adapter
modbus.1 | 2022-12-12 10:11:00.090 | info | (COMPACT) setState not processed because States database not connected
modbus.1 | 2022-12-12 10:11:00.089 | warn | (COMPACT) Poll error count: 1719 code: {"err":"timeout","timeout":5000}
modbus.1 | 2022-12-12 10:11:00.088 | error | (COMPACT) Client in error state.
modbus.1 | 2022-12-12 10:11:00.087 | error | (COMPACT) Request timed out.
modbus.1 | 2022-12-12 10:11:00.087 | warn | (COMPACT) Error: undefined
modbus.1 | 2022-12-12 10:10:55.085 | info | (COMPACT) setState not processed because States database not connected
modbus.1 | 2022-12-12 10:10:55.085 | info | (COMPACT) Connected to slave 192.168.97.46
modbus.1 | 2022-12-12 10:09:55.075 | info | (COMPACT) setState not processed because States database not connected
modbus.1 | 2022-12-12 10:09:55.075 | info | (COMPACT) Disconnected from slave 192.168.97.46
modbus.1 | 2022-12-12 10:09:54.075 | error | (COMPACT) Reconnect did not help, restart adapter
modbus.1 | 2022-12-12 10:09:54.075 | info | (COMPACT) setState not processed because States database not connected
modbus.1 | 2022-12-12 10:09:54.074 | warn | (COMPACT) Poll error count: 1718 code: {"err":"timeout","timeout":5000}
modbus.1 | 2022-12-12 10:09:54.074 | error | (COMPACT) Client in error state.
modbus.1 | 2022-12-12 10:09:54.074 | error | (COMPACT) Request timed out.
modbus.1 | 2022-12-12 10:09:54.073 | warn | (COMPACT) Error: undefined
modbus.1 | 2022-12-12 10:09:49.072 | info | (COMPACT) setState not processed because States database not connected
modbus.1 | 2022-12-12 10:09:49.072 | info | (COMPACT) Connected to slave 192.168.97.46
web.0 | 2022-12-12 10:09:38.612 | info | (COMPACT) ==> Connected system.user.admin from ::ffff:192.168.97.65
web.0 | 2022-12-12 10:09:38.554 | info | (COMPACT) ==> Connected system.user.admin from ::ffff:192.168.97.65
web.0 | 2022-12-12 10:09:38.510 | info | (COMPACT) ==> Connected system.user.admin from ::ffff:192.168.97.65
web.0 | 2022-12-12 10:09:36.376 | info | (COMPACT) ==> Connected system.user.admin from ::ffff:192.168.97.65
modbus.1 | 2022-12-12 10:08:49.066 | info | (COMPACT) setState not processed because States database not connected
modbus.1 | 2022-12-12 10:08:49.066 | info | (COMPACT) Disconnected from slave 192.168.97.46
modbus.1 | 2022-12-12 10:08:48.066 | error | (COMPACT) Reconnect did not help, restart adapter
modbus.1 | 2022-12-12 10:08:48.066 | info | (COMPACT) setState not processed because States database not connected
modbus.1 | 2022-12-12 10:08:48.065 | warn | (COMPACT) Poll error count: 1717 code: {"err":"timeout","timeout":5000}
modbus.1 | 2022-12-12 10:08:48.065 | error | (COMPACT) Client in error state.
modbus.1 | 2022-12-12 10:08:48.065 | error | (COMPACT) Request timed out.
[... cut ...]
modbus.1 | 2022-12-12 10:02:11.988 | error | (COMPACT) Request timed out.
modbus.1 | 2022-12-12 10:02:11.986 | warn | (COMPACT) Error: undefined
modbus.1 | 2022-12-12 10:02:06.985 | info | (COMPACT) setState not processed because States database not connected
modbus.1 | 2022-12-12 10:02:06.984 | info | (COMPACT) Connected to slave 192.168.97.46
kecontact.0 | 2022-12-12 10:02:06.798 | info | (COMPACT) (re)start charging
kecontact.0 | 2022-12-12 10:02:06.113 | info | (COMPACT) change of photovoltaics automatic from true to 0

</body>
</html>`

Apollon77, 2022-12-12:
It is now longer yes, but stil no debug log :-) (and best add as txt file)

Admin ->Expert mode -> Instances -> expand modbus.0 instance - set loglevel to "debug"

Sneak-L8, 2022-12-12:
Sorry, I should the whole message but just looked at it briefly.
Here's a log with debug and a longer log as debug.log.
<html><body>

modbus.1 | 2022-12-12 12:02:21.334 | debug | (COMPACT) [DevID_3] Poll start ---------------------
modbus.1 | 2022-12-12 12:02:18.921 | debug | (COMPACT) [DevID_3] Poll start ---------------------
modbus.1 | 2022-12-12 12:02:16.506 | debug | (COMPACT) [DevID_3] Poll start ---------------------
modbus.1 | 2022-12-12 12:02:14.075 | debug | (COMPACT) [DevID_3] Poll start ---------------------
modbus.1 | 2022-12-12 12:02:11.659 | debug | (COMPACT) [DevID_3] Poll start ---------------------
modbus.1 | 2022-12-12 12:02:09.260 | debug | (COMPACT) [DevID_3] Poll start ---------------------
modbus.1 | 2022-12-12 12:02:08.352 | info | (COMPACT) setState not processed because States database not connected
modbus.1 | 2022-12-12 12:02:08.350 | info | (COMPACT) Disconnected from slave 192.168.97.46
modbus.1 | 2022-12-12 12:02:07.350 | error | (COMPACT) Reconnect did not help, restart adapter
modbus.1 | 2022-12-12 12:02:07.349 | info | (COMPACT) setState not processed because States database not connected
modbus.1 | 2022-12-12 12:02:07.349 | warn | (COMPACT) Poll error count: 1820 code: {"err":"timeout","timeout":5000}
modbus.1 | 2022-12-12 12:02:07.348 | error | (COMPACT) Client in error state.
modbus.1 | 2022-12-12 12:02:07.348 | error | (COMPACT) Request timed out.
modbus.1 | 2022-12-12 12:02:07.348 | warn | (COMPACT) Error: undefined
modbus.1 | 2022-12-12 12:02:06.872 | debug | (COMPACT) [DevID_3] Poll start ---------------------
modbus.1 | 2022-12-12 12:02:04.477 | debug | (COMPACT) [DevID_3] Poll start ---------------------
modbus.1 | 2022-12-12 12:02:02.345 | info | (COMPACT) setState not processed because States database not connected
modbus.1 | 2022-12-12 12:02:02.344 | info | (COMPACT) Connected to slave 192.168.97.46
modbus.1 | 2022-12-12 12:02:01.906 | debug | (COMPACT) [DevID_3] Poll start ---------------------
modbus.1 | 2022-12-12 12:01:59.486 | debug | (COMPACT) [DevID_3] Poll start ---------------------
modbus.1 | 2022-12-12 12:01:57.065 | debug | (COMPACT) [DevID_3] Poll start ---------------------
modbus.1 | 2022-12-12 12:01:54.651 | debug | (COMPACT) [DevID_3] Poll start ---------------------
modbus.1 | 2022-12-12 12:01:52.250 | debug | (COMPACT) [DevID_3] Poll start ---------------------
modbus.1 | 2022-12-12 12:01:49.855 | debug | (COMPACT) [DevID_3] Poll start ---------------------
modbus.1 | 2022-12-12 12:01:47.454 | debug | (COMPACT) [DevID_3] Poll start ---------------------

</body>
</html>
[debog.log](https://github.com/ioBroker/ioBroker.modbus/files/10207318/debog.log)

Sneak-L8, 2022-12-15:
Without changing anything the problem is gone since midnight.
Plenty of messages up to yesterday 23:59, now they are gone....
I will check if it re-occurs in the next days.

Apollon77, 2022-12-30:
maybe you need to increase the timout or such? And, happened again?

Sneak-L8, 2022-12-30:
No, everything fine since two weeks now. I expect it will re-occurr sometime - whithout any (visible) reason ...

closed 2022-12-30 by Sneak-L8

Apollon77, 2022-12-30:
Oki, so a "bad day of the device" ;-))