ioBroker/ioBroker.modbus #290: Wakeup-Sequenz senden

tis-cs, 2022-12-14:
Ich nutze den Adapter um einen ltron Wärmemengenzähler (Akku) über USB/IR auszulesen.
Ich empfange immer nur dann Daten, wenn ich mit Hilfe von dem mbus-testT Tool (https://github.com/geronet1/mbus-test) mich parallel per Console mit dem selben IR-Kopf verbinde und mit "q" eine Aufwachsequenz sende.
Sonst erhalte ich vom Wärmemengenzähler keine Daten.

Wäre es möglich, dass der modbus Adapter diese Sequenz auch periodisch sendet?

Danke!

Apollon77, 2022-12-30:
Frage: Bust DU hier richtig? mbus !== modbus??? Müsste das nicht zum mbus Adapter?

Apollon77, 2022-12-30:
https://github.com/geronet1/mbus-test/blob/main/main.c#L202

tis-cs, 2022-12-31:
> Frage: Bust DU hier richtig? mbus !== modbus??? Müsste das nicht zum mbus Adapter?

ups, sorry! Vollkommen richtig, war im falschen Kontext und daher auch falscher Adapter :) 
Guten Rutsch!

closed 2022-12-31 by tis-cs

Apollon77, 2022-12-31:
Aber gern bei mbus neu anlegen wenn relevant