ioBroker/ioBroker.modbus #302: Unsigned 32bit Little Endian Byte Swap

woehrer12, 2023-01-04:
**Is your feature request related to a problem? Please describe.**
I have a air condition system and i need a Input u32 with Little Endian Byte Swap

I had try to fix it, but i'm not so confirm with the iobroker adapter devlopment

https://github.com/woehrer12/ioBroker.modbus

Apollon77, 2023-01-04:
In fact your chanes look good ... You should not change package/io-package because this is done automatically by our release scripts. But  I think we can also take over the commits directly ... Will look into it again soon

Buddeli, 2023-05-11:
Hello everyone,
I just tried to read some values out of my new heat pump. Sadly I figured out, that I do need as well the "32bit Little Endian Byte Swap". It would be great if the enhancement would be come soon. Great thanks.

woehrer12, 2023-07-12:
Can you fix this problem?

adavt, 2023-07-19:
I also needed it for the WB-MAP6S meter. It is also advisable to make Unsigned 64bit Little Endian Byte Swap.