ioBroker/ioBroker.modbus #304: update unchanged states as slave

an-klam, 2023-01-06:
Update unchanged states as slave not working.

First everything works as expected.
At the first request of the values there are no problems but when the master requests values that have not yet been changed by my script, then the adapter transfers the value 0.
If between two queries the values were changed, then with the second query the new value is transferred.

Nothing to see in logs. Seems not to be an error for the adapter.

[image]
[image]

Apollon77, 2023-01-06:
Pleas eprovide a debug level log of such an example

an-klam, 2023-01-06:
ok

as you can see in this part data was not update inbetween requestst.
but i dont know how to show you that the second request was answered with value 0.
as i mentioned befor: you cant see it in the logs.
`
modbus.2 | 2023-01-06 17:33:31.134 | debug | sending tcp data
-- | -- | -- | --
modbus.2 | 2023-01-06 17:33:31.133 | debug | FC3 finished read holding register request: from 22, len 8
modbus.2 | 2023-01-06 17:33:31.080 | debug | sending tcp data
modbus.2 | 2023-01-06 17:33:31.079 | debug | FC3 finished read holding register request: from 12, len 10
modbus.2 | 2023-01-06 17:33:30.688 | debug | sending tcp data
modbus.2 | 2023-01-06 17:33:30.687 | debug | FC3 finished read holding register request: from 0, len 6
modbus.2 | 2023-01-06 17:33:30.638 | debug | sending tcp data
modbus.2 | 2023-01-06 17:33:30.636 | debug | FC3 finished read holding register request: from 22, len 8
modbus.2 | 2023-01-06 17:33:30.583 | debug | sending tcp data
modbus.2 | 2023-01-06 17:33:30.582 | debug | FC3 finished read holding register request: from 12, len 10
`

`
host.raspberrypi | 2023-01-06 17:33:41.389 | info | stopInstance system.adapter.modbus.2 send kill signal
-- | -- | -- | --
modbus.2 | 2023-01-06 17:33:41.220 | info | Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
modbus.2 | 2023-01-06 17:33:41.219 | debug | Plugin sentry destroyed
modbus.2 | 2023-01-06 17:33:41.218 | info | terminating
modbus.2 | 2023-01-06 17:33:41.214 | info | Got terminate signal TERMINATE_YOURSELF
host.raspberrypi | 2023-01-06 17:33:41.207 | info | stopInstance system.adapter.modbus.2 (force=false, process=true)
host.raspberrypi | 2023-01-06 17:33:41.206 | info | "system.adapter.modbus.2" disabled
modbus.2 | 2023-01-06 17:33:40.217 | debug | state  Changed ack=false: modbus.2.holdingRegisters.11_identification_code:  {"val":71,"ack":false,"ts":1673022820011,"q":0,"c":"script.js.common.RS485","from":"system.adapter.javascript.0","user":"system.user.admin","lc":1673022820011}
modbus.2 | 2023-01-06 17:33:40.172 | debug | state  Changed ack=false: modbus.2.holdingRegisters.4353_Measuring_System:  {"val":4,"ack":false,"ts":1673022820011,"q":0,"c":"script.js.common.RS485","from":"system.adapter.javascript.0","user":"system.user.admin","lc":1673022820011}
modbus.2 | 2023-01-06 17:33:40.090 | debug | state  Changed ack=false: modbus.2.holdingRegisters.40_W_Summe:  {"val":4837,"ack":false,"ts":1673022820002,"q":0,"c":"script.js.common.RS485","from":"system.adapter.javascript.0","user":"system.user.admin","lc":1673022820002}
modbus.2 | 2023-01-06 17:33:40.090 | debug | state  Changed ack=false: modbus.2.holdingRegisters.74_L3_kWh:  {"val":73790,"ack":false,"ts":1673022820002,"q":0,"c":"script.js.common.RS485","from":"system.adapter.javascript.0","user":"system.user.admin","lc":1673016196389}
modbus.2 | 2023-01-06 17:33:40.082 | debug | state  Changed ack=false: modbus.2.holdingRegisters.10_L3_L1_V:  {"val":38167,"ack":false,"ts":1673022820001,"q":0,"c":"script.js.common.RS485","from":"system.adapter.javascript.0","user":"system.user.admin","lc":1673022820001}
modbus.2 | 2023-01-06 17:33:40.073 | debug | state  Changed ack=false: modbus.2.holdingRegisters.28_L3_VA:  {"val":1875,"ack":false,"ts":1673022820001,"q":0,"c":"script.js.common.RS485","from":"system.adapter.javascript.0","user":"system.user.admin","lc":1673022820001}
modbus.2 | 2023-01-06 17:33:40.065 | debug | state  Changed ack=false: modbus.2.holdingRegisters.22_L3_W:  {"val":1329,"ack":false,"ts":1673022820001,"q":0,"c":"script.js.common.RS485","from":"system.adapter.javascript.0","user":"system.user.admin","lc":1673022816001}
modbus.2 | 2023-01-06 17:33:40.063 | debug | state  Changed ack=false: modbus.2.holdingRegisters.16_L3_A:  {"val":879,"ack":false,"ts":1673022820001,"q":0,"c":"script.js.common.RS485","from":"system.adapter.javascript.0","user":"system.user.admin","lc":1673022820001}
modbus.2 | 2023-01-06 17:33:40.062 | debug | state  Changed ack=false: modbus.2.holdingRegisters.4_L3_V_N:  {"val":22080,"ack":false,"ts":1673022820001,"q":0,"c":"script.js.common.RS485","from":"system.adapter.javascript.0","user":"system.user.admin","lc":1673022820001}
modbus.2 | 2023-01-06 17:33:40.061 | debug | state  Changed ack=false: modbus.2.holdingRegisters.72_L2_kWh:  {"val":75432,"ack":false,"ts":1673022820001,"q":0,"c":"script.js.common.RS485","from":"system.adapter.javascript.0","user":"system.user.admin","lc":1673016388136}
modbus.2 | 2023-01-06 17:33:40.050 | debug | state  Changed ack=false: modbus.2.holdingRegisters.8_L2_L3_V:  {"val":38284,"ack":false,"ts":1673022820001,"q":0,"c":"script.js.common.RS485","from":"system.adapter.javascript.0","user":"system.user.admin","lc":1673022820001}
modbus.2 | 2023-01-06 17:33:40.049 | debug | state  Changed ack=false: modbus.2.holdingRegisters.26_L2_VA:  {"val":2857,"ack":false,"ts":1673022820001,"q":0,"c":"script.js.common.RS485","from":"system.adapter.javascript.0","user":"system.user.admin","lc":1673022820001}
modbus.2 | 2023-01-06 17:33:40.047 | debug | state  Changed ack=false: modbus.2.holdingRegisters.20_L2_W:  {"val":2417,"ack":false,"ts":1673022820001,"q":0,"c":"script.js.common.RS485","from":"system.adapter.javascript.0","user":"system.user.admin","lc":1673022820001}
modbus.2 | 2023-01-06 17:33:40.041 | debug | state  Changed ack=false: modbus.2.holdingRegisters.14_L2_A:  {"val":1341,"ack":false,"ts":1673022820001,"q":0,"c":"script.js.common.RS485","from":"system.adapter.javascript.0","user":"system.user.admin","lc":167 [... cut ...]
modbus.2 | 2023-01-06 17:33:00.750 | debug | Iterate holdingRegs holdingRegisters: 0 = holdingRegisters.0_L1_V_N
modbus.2 | 2023-01-06 17:33:00.546 | info | starting. Version 5.0.5 in /opt/iobroker/node_modules/iobroker.modbus, node: v14.17.6, js-controller: 4.0.23
modbus.2 | 2023-01-06 17:32:59.957 | debug | Plugin sentry Initialize Plugin (enabled=true)
host.raspberrypi | 2023-01-06 17:32:58.011 | info | instance system.adapter.modbus.2 started with pid 21646
host.raspberrypi | 2023-01-06 17:32:57.649 | info | "system.adapter.modbus.2" enabled
`

an-klam, 2023-01-06:
could not reproduce the error with RADZIO. Not even with a query every 10ms.
Probably the other device is responsible.
closed

closed 2023-01-06 by an-klam