ioBroker/ioBroker.modbus #315: Neuer Datentyp für Raw-Output möglich?

octane321, 2023-02-07:
Oft muss man mit halbgaren Protokollen arbeiten und weiß nicht welcher Datentyp der richtig ist.

Da wäre ein RAW Datentyp ganz praktisch der den Inhalt unverschleiert offenbart.

 Danke und VG

RasputinBerry, 2023-02-09:
... ich klinke mich mal ein ...
Ein SMA Wechselrichter verlangt dies hier:
[image]
Wenn man das (oder anderes) eingibt ...
[image]
.... kommt das als Log:
[image]

PLCHome, 2024-01-29:
Hi, danke für die Anfrage. 

@octane321 
Ich habe zum Auslesen meiner Wallbox einmal einen RAW Typen implantiert.
Er scheibt die Register angefangen beim 1.le bis zum n.be als hexadezimalen String in das Objekt.
Also 1. Wort 1 und zweites Wort 2 ergäbe bei zwei Wörtern den Text: 00010002

Die Wallbox sendet 5 Bytes mit der Uhrzeit. Mit einem rawstring mit der Größe 5 und der folgenden Formel bekomme ich die Zeitangabe:
`new Date(Date.UTC(parseInt(x.substring(0, 4),16),parseInt(x.substring(4, 8),16)-1,parseInt(x.substring(8, 12),16),parseInt(x.substring(12, 16),16),parseInt(x.substring(16, 20),16))).toLocaleString()`

@RasputinBerry 
Ich habe ebenfalls die Strings überarbeitet. Da war was kaputt. Des Weiteren gibt es wohl Boxen wie meine, die in jedes Wort ein Zeichen und nicht zwei schreibt. Dafür hab ich jetzt auch ein Typen String16 und String16le implementiert.

Wenn ihr das einmaltesten wollt hier ist das Repository:
git@github.com:PLCHome/ioBroker.modbus.git

Ein pull requests ist gestellt. Ich kann aber nicht sagen, wie lange das dauert.

octane321, 2024-01-30:
Hi @PLCHome,
danke Dir das du dic meldest - das ganze is schon so lange her, das ich gar nicht mehr weiß wozu ich es gebraucht hatte oder wie ich es letztendlich gelöst habe :-D

Aber ich bin mir sicher das es eines Tages nützlich sein wird!

Vielen Dank für deine Mühen

closed 2024-04-12 by Apollon77