ioBroker/ioBroker.modbus #317: Add leading zeros to addresses in ID

KoMa1978, 2023-02-16:
It would be great to have an option to add leading zeros to the address if added to the ID.

Now the ID looks like "modbus.0.holdingRegisters.40072_M_AC_Current_A" which is fine unless there are addresses with less than 5 numbers.

In the Object Browser they are not sorted properly. Example:
[image]

Apollon77, 2023-02-17:
Is this change really worth a breakage for all existing users and state ids or another config option?

KoMa1978, 2023-02-17:
Of course not if it is breaking anything ;-)
But enabling "Adresse nicht in ID aufnehmen" will also break existing datapoints or configurations like Influx archiving.

Adding an option like the following would allow anyone to decide what is preferred:

[image]