ioBroker/ioBroker.modbus #325: virtual datapoint calculated from formula possible?

harvey637, 2023-03-09:
Hi,

reading values of my solaredge does not directly give the "PV_Production", only the parts "AC_Power", "Battery_Power".

The missing value is the sum of both values.

So I want to create a "virtual" datapoint (like holding register in modbus adapter) with a function like
sf['40084'] + sf['202837']

not polling from slave, as this datapoint is unknown there.

Writing a script outside of modbus adapter seems possoble, but inside it is more "natural" and time consitent and less overhead.

Small part of feature request this is the warning about "scaled" datapoint to be used as SF: if this SF is used in another calulation it is ok to save the scaled datapoint IMHO.
`[DevID_1/holdingRegs/40084]  Calculation of a scaleFactor which is based on another scaleFactor  seems strange. Please check the config for address 83 !`

</body>
</html>

real data:
40084 is scaled by 40085 (formula is part of 40084 definition) : 4000 W
102837 is already correct (not scaled): -1000W
I want the sum of "40084_SCALED" plus "102837": expected: 3000 W)

Thank You
Harvey

Apollon77, 2023-03-12:
Please use the javascript adapter for this and reate an own state in 0_userdata.0 for it

closed 2023-03-12 by Apollon77

harvey637, 2023-03-12:
Thank You,
I dont need javascript for anything, so installing an adapter with xxxx MB of RAM for one simple mathmatical addition seems complete overkill. Especially in the view that you CAN do calculation with SF['...'] :-). 
I will experiment with a value that is always zero and put the Math.xxxx to it, see what happens. If not I will simply skip this.

cu
Harvey