ioBroker/ioBroker.modbus #326: Slave ID 0 not possible

UwesSmartHome, 2023-03-16:
**Describe the bug**  
The issue is, Slave ID 0 is not possible, if stored at holding registers it seems ok, but after reopen the Holding register is 1.

**To Reproduce**  
Steps to reproduce the behavior:
1. Go to 'Holding register.'
2. Click on '.Slave ID'
3. Scroll down to '.0...'
4. See error, no error but not stored

**Expected behavior**  
A clear and concise description of what you expected to happen.

**Screenshots & Logfiles**  
If applicable, add screenshots and logfiles to help explain your problem.

**Versions:**  
 - Adapter version: <v5.0.11>
 - 
[image]

 - JS-Controller version: <js-controller-version> 
 - Node version: <node-version> 
 - Operating system: <os-name>

**Additional context**  
Add any other context about the problem here.
[image]

UwesSmartHome, 2023-03-16:
in addition with mbpoll is it working very well

ichderarnd, 2024-09-08:
I encountered a similar problem today:
The following device

https://www.sailergmbh.de/index.php?eID=dumpFile&t=f&f=408&token=07efe5be4215c9fa03e82a274f39caef04f0ff37

uses Client ID 0 for login and ID 1 to query the registers. The login is valid as long as the TCP connection exists.
If I configured a Modbus instance with multiple device IDs. All registers that were created with ID 0 are automatically changed to ID 1 when saving the instance configuration.

UwesSmartHome, 2025-02-22:
I updated my system to latest version.  6.3.2 behavior stays the same

`CPU: 41,79 %
RAM: 46 %
Betriebszeit: 1d16h
Verfügbar: 7.0.6
Installiert: 7.0.6
Ereignisse: ⇥29 / ↦13
Plattform: linux
Betriebssystem: linux
Architektur: arm64
CPUs: 4
Geschwindigkeit: 1500 MHz
Modell: Cortex-A72
RAM: 3.7 GB
System-Betriebszeit: 1 T. 16:09:13
Node.js: v20.18.2
time: 1740228602250
timeOffset: -60
NPM: 10.8.2
Adapter-Anzahl: 555
Datenträgergröße: 28.0 GB
Freier Festplattenspeicher: 20.2 GB
Aktive Instanzen: 19
Pfad: /opt/iobroker/
Betriebszeit: 1 T. 16:09:18`