ioBroker/ioBroker.modbus #339: Holding Register over 60000 and more (for Solaredge Invester)

bitwicht, 2023-04-10:
For our Solaredge System we need the Hoding register over 60000

Example:
62850 SOH Battery 1 State of Health
62852 SOE Battery 1 State of Energy

If i use this register in iobroker we get the error:
"errorCode":131,"exceptionCode":2,"message":"ILLEGAL DATA ADDRESS

How can we use this register?

Adapter Version: 5-0-11

lennycb, 2023-04-18:
I'll join in here because my "problem" is probably the same.

I want to query the following registers and get no or only sporadic values.

10000 --> 50000-50061 (62)
10500 --> 50500-50534 (35)
10700 --> 50700-50715 (16)
10800 --> 50800-50816 (17)
12000 --> 52000-52005 (6)
40700 --> 80700-80707 (8)

They are all holding registers with function code 03.
I use the input via "Use aliases"

The following warning message appears:
```
modbus.2 2023-04-16 14:52:21.090 warn Poll error count: 4 code: "ReadHoldingRegisters: Invalid FC 0"
modbus.2 2023-04-16 14:52:21.060 warn Error in handler for 3: RangeError [ERR_OUT_OF_RANGE]: The value of "offset" is out of range. It must be >= 0 and <= 11. Received 12
```

it counts up until the adapter restarts.

Also strange is that I always have to query the block from the beginning.
So e.g. 50801, 50802, 50803 etc.
For example, if I only want to query 50802 without also querying 50801, I get an "**ILLEGAL DATA ADDRESS**" error.

Here is another recording between the "SolarAdmin" utility and the inverter:
```
Mode: RTU
--
  | Address: 1 (Slave)
  | Function: 3 (Read Holding Registers)
  | Starting Address: 10000
  | Number of Registers: 62
  | Parsed As:
  | Address: 50000-50061 (62)
  | CRC:53099 (OK)
  |  
  | Packet 262: MODBUS Request (packet size: 129, data size: 125), 2023-04-16 18:26:18,1324920 +0,0001691
  |  
  |  
  | Mode: RTU
  | Address: 1 (Slave)
  | Function: 3 (Read Holding Registers)
  | CRC:63775 (OK)
  |  
  | Packet 264: MODBUS Request (packet size: 8, data size: 4), 2023-04-16 18:26:19,0118781 +0,8793484
  |  
  |  
  | Mode: RTU
  | Address: 1 (Slave)
  | Function: 3 (Read Holding Registers)
  | Starting Address: 10500
  | Number of Registers: 35
  | Parsed As:
  | Address: 50500-50534 (35)
  | CRC:19854 (OK)
  |  
  | Packet 416: MODBUS Request (packet size: 75, data size: 71), 2023-04-16 18:26:19,5737223 +0,0000029
  |  
  |  
  | Mode: RTU
  | Address: 1 (Slave)
  | Function: 3 (Read Holding Registers)
  | CRC:3086 (OK)
  |  
  | Packet 418: MODBUS Request (packet size: 8, data size: 4), 2023-04-16 18:26:20,0402869 +0,4665461
  |  
  |  
  | Mode: RTU
  | Address: 1 (Slave)
  | Function: 3 (Read Holding Registers)
  | Starting Address: 10700
  | Number of Registers: 16
  | Parsed As:
  | Address: 50700-50715 (16)
  | CRC:35941 (OK)
  |  
  | Packet 492: MODBUS Request (packet size: 37, data size: 33), 2023-04-16 18:26:20,5339410 +0,0000032
  |  
  |  
  | Mode: RTU
  | Address: 1 (Slave)
  | Function: 3 (Read Holding Registers)
  | CRC:37498 (OK)
  |  
  | Packet 494: MODBUS Request (packet size: 8, data size: 4), 2023-04-16 18:26:21,0716168 +0,5376566
  |  
  |  
  | Mode: RTU
  | Address: 1 (Slave)
  | Function: 3 (Read Holding Registers)
  | Starting Address: 10800
  | Number of Registers: 17
  | Parsed As:
  | Address: 50800-50816 (17)
  | CRC:36305 (OK)
  |  
  | Packet 572: MODBUS Request (packet size: 39, data size: 35), 2023-04-16 18:26:21,5359289 +0,0000033
  |  
  |  
  | Mode: RTU
  | Address: 1 (Slave)
  | Function: 3 (Read Holding Registers)
  | CRC:29514 (OK)
  |  
  | Packet 574: MODBUS Request (packet size: 8, data size: 4), 2023-04-16 18:26:22,1028002 +0,5668509
  |  
  |  
  | Mode: RTU
  | Address: 1 (Slave)
  | Function: 3 (Read Holding Registers)
  | Starting Address: 12000
  | Number of Registers: 6
  | Parsed As:
  | Address: 52000-52005 (6)
  | CRC:52502 (OK)
  |  
  | Packet 608: MODBUS Request (packet size: 17, data size: 13), 2023-04-16 18:26:22,5131010 +0,0000030
  |  
  |  
  | Mode: RTU
  | Address: 1 (Slave)
  | Function: 3 (Read Holding Registers)
  | CRC:34591 (OK)
  |  
  | Packet 610: MODBUS Request (packet size: 8, data size: 4), 2023-04-16 18:26:23,3681224 +0,8550030
  |  
  |  
  | Mode: RTU
  | Address: 1 (Slave)
  | Function: 3 (Read Holding Registers)
  | Starting Address: 40700
  | Number of Registers: 8
  | Parsed As:
  | Address: 80700-80707 (8)
  | CRC:43988 (OK)
  |  
  | Packet 666: MODBUS Request (packet size: 27, data size: 23), 2023-04-16 18:26:23,5232212 +0,0000030
  |  
  |  
  | Mode: RTU
  | Address: 1 (Slave)
  | Function: 3 (Read Holding Registers)
  | CRC:41613 (OK)
  |  
  | Packet 670: MODBUS Request (packet size: 8, data size: 4), 2023-04-16 18:26:24,5926422 +0,0777386

```

kirk6210, 2023-07-08:
I had the same issue with Solaredge and iobroker ModBus Adapter and found a workaround.

Just add 40000 to the address. In my case 61441 = 101441.

cr1979, 2023-08-08:
Same here I try to get the Address 57600 and also get this error: Poll error count: 2 code: "ReadHoldingRegisters: Invalid FC 0".
I also tried what @kirk6210 said with adding 40000 won't also work.

I have following addresses to read:
| Adresse | Größe | R/W | Name                           | Typ           | Einheiten        |
|---------|-------|-----|--------------------------------|---------------|------------------|
| 57600   | 16    | R   | Battery 1 Manufacturer Name    | String[32]    | -                |
| 57616   | 16    | R   | Battery 1 Model                | String[32]    | -                |
| 57632   | 16    | R   | Battery 1 Firmware Version     | String[32]    | -                |
| 57648   | 16    | R   | Battery 1 Serial Number        | String[32]    | -                |
| 57664   | 1     | R   | Battery 1 Device ID           | Uint16        | -                |
| 57665   | 1     | R   | Reserved                       | Uint16        | -                |
| 57666   | 2     | R   | Battery 1 Rated Energy         | Float32       | W*H              |
| 57668   | 2     | R   | Battery 1 Max Charge Continues Power | Float32       | W                |
| 57670   | 2     | R   | Battery 1 Max Discharge Continues Power | Float32       | W                |
| 57672   | 2     | R   | Battery 1 Max Charge Peak Power | Float32       | W                |
| 57674   | 2     | R   | Battery 1 Max Discharge Peak Power | Float32       | W                |
| 57676   | 32    | R   | Reserved                       | Uint16        | -                |
| 57708   | 2     | R   | Battery 1 Average Temperature   | Float32       | °C               |
| 57710   | 2     | R   | Battery 1 Max Temperature       | Float32       | °C               |
| 57712   | 2     | R   | Battery 1 Instantaneous Voltage | Float32       | V                |
| 57714   | 2     | R   | Battery 1 Instantaneous Current | Float32       | A                |
| 57716   | 2     | R   | Battery 1 Instantaneous Power   | Float32       | W                |
| 57718   | 4     | R   | Battery 1 Lifetime Export Energy Counter | Uint64        | W*H              |
| 57722   | 4     | R   | Battery 1 Lifetime Import Energy Counter | Uint64        | W*H              |
| 57726   | 2     | R   | Battery 1 Max Energy           | Float32       | W*H              |
| 57728   | 2     | R   | Battery 1 Available Energy      | Float32       | W*H              |
| 57730   | 2     | R   | Battery 1 State of Health (SOH) | Float32       | N/A              |
| 57732   | 2     | R   | Battery 1 State of Energy (SOE) | Float32       | N/A              |
| 57734   | 2     | R   | Battery 1 Status               | Uint32        | 0-7              |
| 57736   | 2     | R   | Battery 1 Status Internal      | Uint32        | Vendor-defined status codes. |
| 57738   | 8     | R   | Battery 1 Events Log           | Uint16[8]     | Battery vendor's last fault ID |
| 57746   | 8     | R   | Battery 1 Events Log Internal  | Uint16[8]     | -                |
| 57754   | 32    | R   | Reserved                       | Uint16        | N/A              |