ioBroker/ioBroker.modbus #340: Modbus Adapter stürzt ab und lässt weitere Adapter abstürzen (Shelly, MQTT Broker)

DPO99, 2023-04-12:
**Describe the bug**  
Aus mir nicht bekannten Gründen Gerät der Modbus Adpater bzw. ein verbundenes Gerät öfter einmal in ein TimeOut. Die Ursache ist mir nicht klar, da das Gerät (Victron Cerbo) mehrere Stunden/Tage problemlos läuft aber es zwischendurch dann völlig unerwartet zu TimeOuts kommt. Sofern es zu dem Fehler kommt, werden immer auch die Adapter für Shelly und der MQTT Broker mit in den Tod gerissen:

<html>
<body>

shelly.0 | 2023-04-12 22:29:26.521 | info | [MQTT] Device with client id "shellyplus1pm-4855199af974" connected!
-- | -- | -- | --
mqtt.0 | 2023-04-12 22:29:26.520 | info | Client [OpenDTU-12111212] reconnected. Old secret 1681330522782_1269. New secret 1681331366514_6401
shelly.0 | 2023-04-12 22:29:26.438 | info | [MQTT] Device with client id "shellyplus1pm-441793cfe7c0" connected!
shelly.0 | 2023-04-12 22:29:26.431 | info | [MQTT] Device with client id "shellyplus1pm-441793a611d4" connected!
modbus.0 | 2023-04-12 22:29:15.167 | info | Disconnected from slave 192.168.178.76
modbus.0 | 2023-04-12 22:29:14.167 | warn | Poll error count: 1 code: {"err":"timeout","timeout":5000}
modbus.0 | 2023-04-12 22:29:14.167 | error | Client in error state.
modbus.0 | 2023-04-12 22:29:14.166 | error | Request timed out.
modbus.0 | 2023-04-12 22:29:14.166 | warn | Error: undefined

</body>
</html>

**To Reproduce**  
Modbus Gerät in einen TimeOut zwingen

**Expected behavior**  
Selbst wenn der Modbus Adapter abstürzt, darf er keine anderen Adapter, welche keinen Bezug zu ihm haben, mit stoppen.

**Versions:**  
 - Adapter version: 5.0.11
 - JS-Controller version: 4.0.24
 - Node version: v16.19.1 mit NPM 8.19.3
 - Operating system: Raspbian Bullseye

Apollon77, 2023-04-12:
Bitte ein vollständiges Log posten, ich sehe hier nichts was" in den Tod gerissen" wird

DPO99, 2023-04-12:
Anbei. Es ist immer zu erkennen, dass erst Modbus ein Problem hat und danach Shelly und MQTT einen Reconnect durchführen (müssen).

<html>
<body>

2023-04-12 22:15:14.659 - warn: modbus.0 (660912) Error: undefined
--
2023-04-12 22:15:14.660 - error: modbus.0 (660912) Request timed out.
2023-04-12 22:15:14.661 - error: modbus.0 (660912) Client in error state.
2023-04-12 22:15:14.662 - warn: modbus.0 (660912) Poll error count: 1 code: {"err":"timeout","timeout":5000}
2023-04-12 22:15:15.661 - info: modbus.0 (660912) Disconnected from slave 192.168.178.76
2023-04-12 22:15:22.445 - info: mqtt.0 (520204) Client [OpenDTU-12111212] connection closed: timeout
2023-04-12 22:15:22.790 - info: mqtt.0 (520204) Client [OpenDTU-12111212] connected with secret 1681330522782_1269
2023-04-12 22:15:22.805 - info: shelly.0 (885) [MQTT] Device with client id "shellyplus1pm-4855199af974" connected!
2023-04-12 22:15:22.826 - info: mqtt.0 (520204) Client [OpenDTU-12111212] subscribes on "solar/+/cmd/limit_persistent_relative" with regex /solar\.[^.]*\.cmd\.limit_persistent_relative/
2023-04-12 22:15:22.826 - info: mqtt.0 (520204) Client [OpenDTU-12111212] subscribes on "solar/+/cmd/limit_persistent_relative" with regex /mqtt\.0\.solar\.[^.]*\.cmd\.limit_persistent_relative/
2023-04-12 22:15:22.827 - info: mqtt.0 (520204) Client [OpenDTU-12111212] subscribes on "solar/+/cmd/limit_persistent_absolute" with regex /solar\.[^.]*\.cmd\.limit_persistent_absolute/
2023-04-12 22:15:22.827 - info: mqtt.0 (520204) Client [OpenDTU-12111212] subscribes on "solar/+/cmd/limit_persistent_absolute" with regex /mqtt\.0\.solar\.[^.]*\.cmd\.limit_persistent_absolute/
2023-04-12 22:15:22.845 - info: shelly.0 (885) [MQTT] Device with client id "shellyplus1pm-441793a611d4" connected!
2023-04-12 22:15:22.848 - info: mqtt.0 (520204) Client [OpenDTU-12111212] subscribes on "solar/+/cmd/limit_nonpersistent_relative" with regex /solar\.[^.]*\.cmd\.limit_nonpersistent_relative/
2023-04-12 22:15:22.847 - info: shelly.0 (885) [MQTT] Device with client id "shellyplus1pm-441793cfe7c0" connected!
2023-04-12 22:15:22.848 - info: mqtt.0 (520204) Client [OpenDTU-12111212] subscribes on "solar/+/cmd/limit_nonpersistent_relative" with regex /mqtt\.0\.solar\.[^.]*\.cmd\.limit_nonpersistent_relative/
2023-04-12 22:15:22.850 - info: mqtt.0 (520204) Client [OpenDTU-12111212] subscribes on "solar/+/cmd/limit_nonpersistent_absolute" with regex /solar\.[^.]*\.cmd\.limit_nonpersistent_absolute/
2023-04-12 22:15:22.850 - info: mqtt.0 (520204) Client [OpenDTU-12111212] subscribes on "solar/+/cmd/limit_nonpersistent_absolute" with regex /mqtt\.0\.solar\.[^.]*\.cmd\.limit_nonpersistent_absolute/
2023-04-12 22:15:22.851 - info: mqtt.0 (520204) Client [OpenDTU-12111212] subscribes on "solar/+/cmd/power" with regex /solar\.[^.]*\.cmd\.power/
2023-04-12 22:15:22.852 - info: mqtt.0 (520204) Client [OpenDTU-12111212] subscribes on "solar/+/cmd/power" with regex /mqtt\.0\.solar\.[^.]*\.cmd\.power/
2023-04-12 22:15:22.853 - info: mqtt.0 (520204) Client [OpenDTU-12111212] subscribes on "solar/+/cmd/restart" with regex /solar\.[^.]*\.cmd\.restart/
2023-04-12 22:15:22.853 - info: mqtt.0 (520204) Client [OpenDTU-12111212] subscribes on "solar/+/cmd/restart" with regex /mqtt\.0\.solar\.[^.]*\.cmd\.restart/
2023-04-12 22:15:23.026 - info: mqtt.0 (520204) Client [OpenDTU-12111212] publishOnSubscribe send all known states
2023-04-12 22:15:23.028 - info: mqtt.0 (520204) Client [OpenDTU-12111212] publishOnSubscribe send all known states
2023-04-12 22:15:23.048 - info: mqtt.0 (520204) Client [OpenDTU-12111212] publishOnSubscribe send all known states
2023-04-12 22:15:23.051 - info: mqtt.0 (520204) Client [OpenDTU-12111212] publishOnSubscribe send all known states
2023-04-12 22:15:23.053 - info: mqtt.0 (520204) Client [OpenDTU-12111212] publishOnSubscribe send all known states
2023-04-12 22:15:23.055 - info: mqtt.0 (520204) Client [OpenDTU-12111212] publishOnSubscribe send all known states
2023-04-12 22:15:32.558 - info: mqtt.1 (520215) send2Server mqtt.1.solar.116181671322.cmd.limit_nonpersistent_absolute[solar/116181671322/cmd/limit_nonpersistent_absolute]
2023-04-12 22:15:32.559 - info: mqtt.1 (520215) send2Server mqtt.1.solar.114182122166.cmd.power[solar/114182122166/cmd/power]
2023-04-12 22:15:32.562 - info: mqtt.1 (520215) send2Server mqtt.1.solar.114182122166.cmd.limit_nonpersistent_absolute[solar/114182122166/cmd/limit_nonpersistent_absolute]
2023-04-12 22:15:34.706 - info: mqtt.1 (520215) send2Server mqtt.1.solar.116181671322.cmd.limit_nonpersistent_absolute[solar/116181671322/cmd/limit_nonpersistent_absolute]
2023-04-12 22:15:34.710 - info: mqtt.1 (520215) send2Server mqtt.1.solar.114182122166.cmd.power[solar/114182122166/cmd/power]
2023-04-12 22:15:59.605 - info: mqtt.1 (520215) send2Server mqtt.1.solar.116181671322.cmd.limit_nonpersistent_absolute[solar/116181671322/cmd/limit_nonpersistent_absolute]
2023-04-12 22:16:15.885 - info: modbus.0 (660912) Connected to slave 192.168.178.76
2023-04-12 22:16:24.633 - info: mqtt.1 (520215) send2Server mqtt.1.solar.116181671322.cmd.limit_nonpersistent_absolute[solar/116181671322/cmd/limit_nonpersistent_absolute]
2023-04-12 22:16:49.652 - info: mqtt.1 (520215) send2Server mqtt.1.solar.116181671322.cmd.limit_nonpersistent_absolute[solar/116181671322/cmd/limit_nonpersistent_absolute]
[... cut ...]
2023-04-12 22:55:09.732 - info: mqtt.1 (520215) send2Server mqtt.1.solar.116181671322.cmd.limit_nonpersistent_absolute[solar/116181671322/cmd/limit_nonpersistent_absolute]
2023-04-12 22:55:34.753 - info: mqtt.1 (520215) send2Server mqtt.1.solar.116181671322.cmd.limit_nonpersistent_absolute[solar/116181671322/cmd/limit_nonpersistent_absolute]
2023-04-12 22:55:59.830 - info: mqtt.1 (520215) send2Server mqtt.1.solar.116181671322.cmd.limit_nonpersistent_absolute[solar/116181671322/cmd/limit_nonpersistent_absolute]

</body>
</html>

DPO99, 2023-04-12:
Ich glaube ich habe das Problem gefunden. Anscheinend stört etwas mein Wlan und kickt den Raspberry raus....die Störmeldungen passen zu den Modbus abstürzen:

13.04.23
00:15:08
WLAN: Eine Störquelle wurde auf Kanal 6 (Frequenz 2.437 GHz) erkannt.
13.04.23
00:09:07
WLAN: Eine Störquelle wurde auf Kanal 1 (Frequenz 2.412 GHz) erkannt.
12.04.23
23:44:37
WLAN: Eine Störquelle wurde auf Kanal 6 (Frequenz 2.437 GHz) erkannt.
12.04.23
23:18:07
WLAN: Eine Störquelle wurde auf Kanal 1 (Frequenz 2.412 GHz) erkannt.
12.04.23
23:16:38
WLAN: Eine Störquelle wurde auf Kanal 11 (Frequenz 2.462 GHz) erkannt.
12.04.23
22:50:37
WLAN: Eine Störquelle wurde auf Kanal 1 (Frequenz 2.412 GHz) erkannt.
12.04.23
22:29:07
WLAN: Eine Störquelle wurde auf Kanal 11 (Frequenz 2.462 GHz) erkannt.
12.04.23
22:15:07
WLAN: Eine Störquelle wurde auf Kanal 6 (Frequenz 2.437 GHz) erkannt.
12.04.23
22:12:08
WLAN: Eine Störquelle wurde auf Kanal 1 (Frequenz 2.412 GHz) erkannt.
12.04.23
20:44:07
WLAN: Eine Störquelle wurde auf Kanal 6 (Frequenz 2.437 GHz) erkannt.

closed 2023-04-12 by DPO99