ioBroker/ioBroker.modbus #342: Wertebereichsvorgaben der Objekte berücksichtigen

Tombra1889, 2023-04-14:
Hallo, vom Photovoltaik-Wechselrichter werden in der Nacht negative Werte geschickt, wenn der Wechselrichter sich im Standby befindet. 

In den Objekten habe ich eingestellt, dass der Minimalwert 0 sein soll. Leider warnt der Adapter nur, berücksichtigt dies aber nicht:

`State value to set for "modbus.0.inputRegisters.30961_DC_power_input__2" has value "-2147483648" less than min "0"`

Apollon77, 2023-04-16:
> In den Objekten habe ich eingestellt, dass der Minimalwert 0 sein soll.

Wie hast Du das getan? Zeige bitte mal Screenshot

Tombra1889, 2023-04-16:
Hey, ich habe es in den Objekten eingestellt:
[image]

Leider wird der Wert trotz der Minimaldefinition in das Objekt geschrieben:
[image]

Apollon77, 2023-04-16:
Ok, Du hast also das manuell vom Adapter angelegte Objekt angepasst und selbst was reingeschrieben von dem der Adapter nichts weis? Und die Erwartung ist das er das beachtet?

Dann habe ich eine schlechte Nachricht: Der Adapter ist verantwortlich für die angelegten Objekte und es ist ok wenn er von Deinen Änderungen nichts weiss und die daher auch nicht berückhsichtigt. :-)

Es wäre wenn ein Feature Request das der Adapter in den Settings der Modbus Register bestimmte Werte ausfiltern kann, aber das muss anders laufen als so, sorry. Bitte gern feature Request issue anlegen.

Alternativen zum Aussortieren sind kleine JavaScriptchen bzw Alias-Nutzung

closed 2023-04-16 by Apollon77