ioBroker/ioBroker.modbus #365: Signed 64 bit (Big Indian)

vosi2, 2023-07-08:
Hi, I need an new datatype in the modbus driver.
I have an energycounter which sent its power in  Signed 64  datatype. Up to now I use unsigned 64 but in some cases I got an overflow because the power ist negative.

Thanks for your help and your great work in iobroker.

anwa, 2024-01-24:
Hi,
ABB counter B23 212-100 (all with import/export) using signed 64 bit (4 x 16bit)  values for many registers.
Please add signed 64bit datatype...

Thanks

anwa, 2024-01-24:
I checked the source and the used lib (jsmodbus) is supporting int64be and int64le in common.js
It should not be a big issue to support this in the adapter.
I try to understand the code, but i don*t know how i get this in the admin pages... react? no idea.
I'm not that java/java script expert :)

Gugulao, 2024-04-30:
Hi,
the Phönix Contact EEM-EM377 also uses on some registers Signed 64 bit maybe Big Endian.
Please add a signed 64-bit data type or is there perhaps something to test?