ioBroker/ioBroker.modbus #376: Some registers not working after reboot

JiroMusik, 2023-08-02:
Hi guys,
I am currently working on a visualization for my Victron Energy System.
Some registers are working fine, some won't come up, even though they are in the same module/adress range.
Some registers came up for a few minutes but are not pulled after a reboot of my iobroker Container.
In the logs I get these two messages:

```
modbus.0 | 2023-08-02 19:16:01.698 | warn | Poll error count: 2 code: {"errorCode":131,"exceptionCode":2,"message":"ILLEGAL DATA ADDRESS"}

modbus.0 | 2023-08-02 19:16:01.604 | warn | Poll error count: 1 code: {"errorCode":131,"exceptionCode":10,"message":"GATEWAY PATH UNAVAILABLE"}
```

[image]

These are my Units:
Hub4: 100
System: 100
PV INV: 31
Grid: 30
Vebus: 227
EV Charger: 41

Apollon77, 2023-08-03:
The errors come back from the modbus devcie ... so the first question is to check there.
lterantively enable debug log of the adapter (nd also additional debug in adapter settings) and then compare a log where it works and where it do not work ... Also attach here. maybe you can spot a difference ...

JiroMusik, 2023-08-03:
The display indeed shows an error that the module is not found, but it was found like 10 minutes earlier. I am very confused.
Can a restart of the module help with something like that or is it just overloaded with request and cant reach the broker because of bad wifi?

Apollon77, 2023-08-03:
If you ask if it could help ... try it out