ioBroker/ioBroker.modbus #377: ReadHoldingRegisters: Invalid FC 0 at address 57600

cr1979, 2023-08-10:
**Describe the bug**  
I have following TSV entries

_address	name	description	unit	type	len	factor	offset	formula	role	room	poll	wp	cw	isScale
40001	SunSpecID	C_SunSpec_ID		uint32be	2	1	0		state		true	false	false	false
40003	SunSpecDID	C_SunSpec_DID		uint16be	1	1	0		state		true	false	false	false
40004	SunSpecLength	C_SunSpec_Length		uint16be	1	1	0		state		true	false	false	false
40005	Manufacturer	C_Manufacturer		string	16	1	0		state		true	false	false	false
40021	Model	C_Model		string	16	1	0		state		true	false	false	false
40045	Version	C_Version		string	8	1	0		state		true	false	false	false
40053	SerialNumber	C_SerialNumber		string	16	1	0		state		true	false	false	false
40069	DeviceAddress	C_DeviceAddress		uint16be	1	1	0		state		true	false	false	false
40070	PhaseTypes	C_SunSpec_DID		uint16be	1	1	0		state		true	false	false	false
40071	SunSpecLength	C_SunSpec_Length		uint16be	1	1	0		state		true	false	false	false
40072	CurrentTotal	I_AC_Current	A	uint16be	1	1	0	x * Math.pow(10, sf['40076'])	value		true	false	false	false
40073	CurrentL1	I_AC_CurrentA	A	uint16be	1	1	0	x * Math.pow(10, sf['40076'])	value		true	false	false	false
40074	CurrentL2	I_AC_CurrentB	A	uint16be	1	1	0	x * Math.pow(10, sf['40076'])	value		true	false	false	false
40075	CurrentL3	I_AC_CurrentC	A	uint16be	1	1	0	x * Math.pow(10, sf['40076'])	value		true	false	false	false
40076	CurrentSF	I_AC_Current_SF		int16be	1	1	0		state		true	false	false	true
40080	VoltageL1N	I_AC_VoltageAN	V	uint16be	1	1	0	x * Math.pow(10, sf['40083'])	value		true	false	false	false
40081	VoltageL2N	I_AC_VoltageBN	V	uint16be	1	1	0	x * Math.pow(10, sf['40083'])	value		true	false	false	false
40082	VoltageL3N	I_AC_VoltageCN	V	uint16be	1	1	0	x * Math.pow(10, sf['40083'])	value		true	false	false	false
40083	VoltageSF	I_AC_Voltage_SF		int16be	1	1	0		state		true	false	false	true
40084	Power	I_AC_Power	W	uint16be	1	1	0	x * Math.pow(10, sf['40085'])	value		true	false	false	false
40085	PowerSF	I_AC_Power_SF		int16be	1	1	0		state		true	false	false	true
40086	Frequency	I_AC_Frequency	Hz	uint16be	1	1	0	x * Math.pow(10, sf['40087'])	value		true	false	false	false
40087	FrequencySF	I_AC_Frequency_SF		int16be	1	1	0		state		true	false	false	true
40088	VA	I_AC_VA	VA	int16be	1	1	0	x * Math.pow(10, sf['40089'])	value		true	false	false	false
40089	VASF	I_AC_VA_SF		int16be	1	1	0		state		true	false	false	true
40090	VAR	I_AC_VAR	VAR	int16be	1	1	0	x * Math.pow(10, sf['40091'])	value		true	false	false	false
40091	VARSF	I_AC_VAR_SF		int16be	1	1	0		state		true	false	false	true
40092	PF	I_AC_PF	%	int16be	1	1	0	x * Math.pow(10, sf['40093'])	value		true	false	false	false
40093	PFSF	I_AC_PF_SF		int16be	1	1	0		state		true	false	false	true
40094	TotalEnergy	I_AC_Energy_WH	Wh	uint32be	2	1	0	x * Math.pow(10, sf['40096'])	value		true	false	false	false
40096	TotalEnergySF	I_AC_Energy_WH_SF		int16be	1	1	0		state		true	false	false	true
40097	DCCurrent	I_DC_Current	A	uint16be	1	1	0	x * Math.pow(10, sf['40098'])	value		true	false	false	false
40098	DCCurrentSF	I_DC_Current_SF		int16be	1	1	0		state		true	false	false	true
40099	DCVoltage	I_DC_Voltage	V	uint16be	1	1	0	x * Math.pow(10, sf['40100'])	value		true	false	false	false
40100	DCVoltageSF	I_DC_Voltage_SF		int16be	1	1	0		state		true	false	false	true
40101	DCPower	I_DC_Power	W	int16be	1	1	0	x * Math.pow(10, sf['40102'])	value		true	false	false	false
40102	DCPowerSF	I_DC_Power_SF		int16be	1	1	0		state		true	false	false	true
40104	Temp	I_Temp_Sink	°C	int16be	1	1	0	x * Math.pow(10, sf['40107'])	state		true	false	false	false
40107	TempSF	I_Temp_SF		int16be	1	1	0		value		true	false	false	true
40108	Status	I_Status		uint16be	1	1	0		state		true	false	false	false
40109	StatusVendor	I_Status_Vendor		uint16be	1	1	0		state		true	false	false	false
40140	MeterModel	M_C_Model		string	16	1	0		state		true	false	false	false
40164	MeterVersion	M_C_Version		string	8	1	0		state		true	false	false	false
40172	MeterSerialNumber	M_C_SerialNumber		string	16	1	0		state		true	false	false	false
40188	DeviceAddress	C_DeviceAddress		uint16be	1	1	0		state		true	false	false	false
40191	MeterCurrentTotal	M_AC_Current	A	int16be	1	1	0	x * Math.pow(10, sf['40195'])	value		true	false	false	false
40192	MeterCurrentL1	M_AC_Current_A	A	int16be	1	1	0	x * Math.pow(10, sf['40195'])	value		true	false	false	false
40193	MeterCurrentL2	M_AC_Current_B	A	int16be	1	1	0	x * Math.pow(10, sf['40195'])	value		true	false	false	false
40194	MeterCurrentL3	M_AC_Current_C	A	int16be	1	1	0	x * Math.pow(10, sf['40195'])	value		true	false	false	false
40195	MeterCurrentSF	M_AC_Current_SF		int16be	1	1	0		state		true	false	false	true
40196	MeterVoltageLN	M_AC_Voltage_L_N	V	int16be	1	1	0	x * Math.pow(10, sf['40204'])	value		true	false	false	false
40197	MeterVoltageL1N	M_AC_Voltage_A_N	V	int16be	1	1	0	x * Math.pow(10, sf['40204'])	value		true	false	false	false
40198	MeterVoltageL2N	M_AC_Voltage_B_N	V	int16be	1	1	0	x * Math.pow(10, sf['40204'])	value		true	false	false	false
40199	MeterVoltageL3N	M_AC_Voltage_C_N	V	int16be	1	1	0	x * Math.pow(10, sf['40204'])	value		true	false	false	false
40204	MeterVoltageSF	M_AC_Voltage_SF		int16be	1	1	0		state		true	false	false	true
40205	MeterFrequency	M_AC_Freq	Hz	int16be	1	1	0	x * Math.pow(10, sf['40206'])	value		true	false	false	false
40206	MeterFrequencySF	M_AC_Freq_SF		int16be	1	1	0		state		true	false	false	true
40207	MeterPowerTotal	M_AC_Power	W	int16be	1	1	0	x * Math.pow(10, sf['40211'])	value		true	false	false	false
40208	MeterPowerL1	M_AC_Power_A	W	int16be	1	1	0	x * Math.pow(10, sf['40211'])	value		true	false	false	false
40209	MeterPowerL2	M_AC_Power_B	W	int16be	1	1	0	x * Math.pow(10, sf['40211'])	value		true	false	false	false
40210	MeterPowerL3	M_AC_Power_C	W	int16be	1	1	0	x * Math.pow(10, sf['40211'])	value		true	false	false	false
40211	MeterPowerSF	M_AC_Power_SF		int16be	1	1	0		state		true	false	false	true
40212	MeterApparentPowerTotal	M_AC_VA	VA	int16be	1	1	0	x * Math.pow(10, sf['40216'])	value		true	false	false	false
40213	MeterApparentPowerL1	M_AC_VA_A	VA	int16be	1	1	0	x * Math.pow(10, sf['40216'])	value		true	false	false	false
40214	MeterApparentPowerL2	M_AC_VA_B	VA	int16be	1	1	0	x * Math.pow(10, sf['40216'])	value		true	false	false	false
40215	MeterApparentPowerL3	M_AC_VA_C	VA	int16be	1	1	0	x * Math.pow(10, sf['40216'])	value		true	false	false	false
[... 22 lines cut ...]

Here is the link to the PDF:
https://www.photovoltaikforum.com/core/attachment/88445-power-control-open-protocol-for-solaredge-inverters-pdf/

**To Reproduce**  
Steps to reproduce the behavior:
1. Go to 'Holding Registers'
2. Click on 'edit TSV.'
3. Scroll down to the bottom
4. add line following line "57600	Manufacturer	C_Manufacturer		string	16	1	0		state		true	false	false	false"
5. start import, save and start modbus adapter
6. Check protocoll

**Expected behavior**  
That the value at address 57600 extracted as a string like the manufacturer at address 40005

**Screenshots & Logfiles**  
modbus.0 | 2023-08-10 07:16:24.667 | warn | Poll error count: 2 code: "ReadHoldingRegisters: Invalid FC 0"
modbus.0 | 2023-08-10 07:16:21.543 | warn | Poll error count: 1 code: "ReadHoldingRegisters: Invalid FC 0"
modbus.0 | 2023-08-10 07:16:21.432 | info | Connected to slave 192.168.0.2

**Versions:**  
 - Adapter version: 5.0.11
 - Node version:  v18.16.0
 - Operating system: linux

Apollon77, 2023-08-11:
Please provide a debug log

cr1979, 2023-08-11:
@Apollon77, here is a debug log file from today: 
[iobroker.2023-08-11.log](https://github.com/ioBroker/ioBroker.modbus/files/12320984/iobroker.2023-08-11.log)

If it's easier for you to write in German, then it's also okay with me :

skipperkm, 2023-10-13:
Sorry ,
Hier noch ein logfile
[iobroker.2023-10-13.log](https://github.com/ioBroker/ioBroker.modbus/files/12890157/iobroker.2023-10-13.log)

UweSauter, 2024-01-14:
Ich stelle diese Verhalten mit 6.0.1 ebenfalls fest. Gibt es etwas konkretes, dass man zum Debuggen tun kann?
[modbus.log](https://github.com/ioBroker/ioBroker.modbus/files/13932456/modbus.log)

UweSauter, 2024-01-14:
Ich habe mal meine Logdatei auf einen Abfragezyklus zusammengestrichen:
[modbus_single_loop.log](https://github.com/ioBroker/ioBroker.modbus/files/13933013/modbus_single_loop.log)

Darin fällt folgendes auf:
```console
$ grep -e 'Poll address' -e warn modbus_single_loop.log 
2024-01-11 22:01:30.221  - debug: modbus.0 (9313) [DevID_1/holdingRegs] Poll address 0 - 100 registers
2024-01-11 22:01:30.334  - debug: modbus.0 (9313) [DevID_1/holdingRegs] Poll address 0 DONE
2024-01-11 22:01:30.345  - debug: modbus.0 (9313) [DevID_1/holdingRegs] Poll address 100 - 9 registers
2024-01-11 22:01:30.417  - debug: modbus.0 (9313) [DevID_1/holdingRegs] Poll address 100 DONE
2024-01-11 22:01:30.424  - debug: modbus.0 (9313) [DevID_1/holdingRegs] Poll address 121 - 100 registers
2024-01-11 22:01:30.493  - debug: modbus.0 (9313) [DevID_1/holdingRegs] Poll address 121 DONE
2024-01-11 22:01:30.509  - debug: modbus.0 (9313) [DevID_1/holdingRegs] Poll address 221 - 22 registers
2024-01-11 22:01:30.573  - debug: modbus.0 (9313) [DevID_1/holdingRegs] Poll address 221 DONE
2024-01-11 22:01:30.590  - debug: modbus.0 (9313) [DevID_1/holdingRegs] Poll address 293 - 17322 registers
2024-01-11 22:01:30.688  - warn: modbus.0 (9313) Poll error count: 1 code: "ReadHoldingRegisters: Invalid FC 0"
```

Für mich sieht das, ohne jetzt den Code verstanden zu haben, so aus als wäre die Berechnung fehlerhaft, wie viele Register ausgelesen bzw. bei welcher Adresse der nächste Block gelesen werden soll.

Die entsprechende Stelle ist in

https://github.com/ioBroker/ioBroker.modbus/blob/3659f18d45b3385b218f15476d25664ab1290667/lib/master.js#L186

In meiner Konfiguration der Instanz sind diese Werte hinterlegt:

- Zahlen runden auf: 2
- Datenabfrageintervall: 1000 ms
- Wartezeit bis zum erneuten Verbinden: 60000 ms
- Wartezeit Lesend: 5000 ms
- Impluszeit: 1000 ms
- Wartezeit: 50 ms
- Max Leseanforderungslänge (Float): 100 Register
- Max Leseanforderungslänge (Booleans): 128 Register
- Leseintervall: 0 ms
- Schreibintervall: 0 ms

Wie zu sehen ist widerspricht `Max Leseanforderungslänge (Float): 100 Register` der Zeile im Log `debug: modbus.0 (9313) [DevID_1/holdingRegs] Poll address 293 - 17322 registers`

UweSauter, 2024-01-14:
Wenn ich meine Instanz stoppe und dafür eine zweite Instanz anlege, die nur die Register ab 57600 abfragen soll bekomme ich eine andere Fehlermeldung:

```console
2024-01-14 21:39:14.583 - info: host.iobroker instance system.adapter.modbus.1 started with pid 67382
2024-01-14 21:39:15.267 - info: modbus.1 (67382) Plugin sentry Sentry Plugin disabled for this process because sending of statistic data is disabled for the system
2024-01-14 21:39:15.338 - info: modbus.1 (67382) starting. Version 6.0.1 in /opt/iobroker/node_modules/iobroker.modbus, node: v20.9.0, js-controller: 5.0.17
2024-01-14 21:39:15.563 - info: modbus.1 (67382) Connected to slave 192.168.55.221
2024-01-14 21:39:15.632 - warn: modbus.1 (67382) Poll error count: 1 code: {"errorCode":131,"exceptionCode":2,"message":"ILLEGAL DATA ADDRESS"}
2024-01-14 21:39:16.723 - warn: modbus.1 (67382) Poll error count: 2 code: {"errorCode":131,"exceptionCode":2,"message":"ILLEGAL DATA ADDRESS"}
2024-01-14 21:39:17.832 - warn: modbus.1 (67382) Poll error count: 3 code: {"errorCode":131,"exceptionCode":2,"message":"ILLEGAL DATA ADDRESS"}
2024-01-14 21:39:18.937 - warn: modbus.1 (67382) Poll error count: 4 code: {"errorCode":131,"exceptionCode":2,"message":"ILLEGAL DATA ADDRESS"}
2024-01-14 21:39:20.076 - warn: modbus.1 (67382) Poll error count: 5 code: {"errorCode":131,"exceptionCode":2,"message":"ILLEGAL DATA ADDRESS"}
2024-01-14 21:39:21.168 - warn: modbus.1 (67382) Poll error count: 6 code: {"errorCode":131,"exceptionCode":2,"message":"ILLEGAL DATA ADDRESS"}
2024-01-14 21:39:21.218 - info: modbus.1 (67382) Disconnected from slave 192.168.55.221
2024-01-14 21:40:21.222 - info: modbus.1 (67382) Connected to slave 192.168.55.221
2024-01-14 21:40:21.307 - warn: modbus.1 (67382) Poll error count: 7 code: {"errorCode":131,"exceptionCode":2,"message":"ILLEGAL DATA ADDRESS"}
2024-01-14 21:40:21.359 - info: modbus.1 (67382) Disconnected from slave 192.168.55.221
```

skipperkm, 2024-01-15:
Hi,

es gibt leider von Solaredge keine Doku darüber wo die Register wirklich liegen.(auch in der aus 12/23 nicht, Oder ich habe sie nicht gefunden). Auch im Netz finde ich die unterschiedlichsten Aussagen darüber wo diese liegen sollen, was möglicherweise an unterschiedlichen Typen liegt. Der Adapter scheint alles richtig zu machen. 
Bei mir funktionieren die folgenden Einstellungen für Batterie 1:

97601	Batt1_Manufacturer	B1_Manufacturer		string	8	1	0				true	false	false	false
97617	Batt1_Model	B1_Model		string	16	1	0				true	false	false	false
97633	Batt1_Firmware	B1_Firmware		string	16	1	0				true	false	false	false
97649	Batt1_Serial	B1_SerialNumber		string	16	1	0				true	false	false	false
97665	Batt1_ID	B1_Device_ID		uint16be	1	1	0				true	false	false	false
97709	Batt1_AverageTemp	B1_Durchschnittstemperatur	°C	floatsw	2	1	0				true	false	false	false
97711	Batt1_MaxTemp	B1_Maximale_Temperatur	°C	floatsw	2	1	0				true	false	false	false
97717	Batt1_Power	B1_Aktuelle_Leistung (Minus = Entladen; Plus = Laden)	W	floatsw	2	1	0				true	false	false	false
97719	Batt1_Lifetime_Export	B1_Gesamt_Exportierte_Energie	Wh	uint64le	4	1	0				true	false	false	false
97723	Batt1_Lifetime_Import	B1_Gesamt_Importierte_Energie	Wh	uint64le	4	1	0				true	false	false	false
97727	Batt1_MaxEnergy	B1_Maximale_Ladung	Wh	floatsw	2	1	0				true	false	false	false
97729	Batt1_AvailableEnergy	B1_Nutzbare_Ladung	Wh	floatsw	2	1	0				true	false	false	false
97731	Batt1_StateOfHealth	B1_Lebenszustand	%	floatsw	2	1	0				true	false	false	false
97733	Batt1_StateOfCharge	B1_Ladezustand	%	floatsw	2	1	0				true	false	false	false
97735	Batt1_Status	B1_Batterie_Status (0 – Off; 1 – Standby; 2 – Init; 3 – Charge; 4 – Discharge; 5 – Fault; 7 - Idle)		uint32sw	2	1	0				true	false	false	false
97737	Batt1_StatusIntern	B1_Vendor_defined_status_codes		uint32sw	2	1	0				true	false	false	false
97739	Batt1_EventLog	B1_Battery_vendor_last_fault_ID		uint16le	1	1	0				true	false	false	false

Wie es für einen zweiten Akku aussieht steht in den Sternen. Ein Dritter soll laut Doku dann über Modbus gar nicht mehr gehen. Es sind wohl Register für 3 Zähler und 2 Akkus vor gesehen.
Ich habe eine "SolarEdge Home Battery 4.6kWh" gekauft in 2023, also relativ aktuell.
Wie gesagt, nach meinen Erkenntnissen hat das alles nichts mit dem Adapter zu tun sondern nur mit mangelnder Dokumentation.
Ich hatte mir parallel Homeassistant installiert und die Solaredge-Integration eingerichtet. Die Werte habe ich dann mit dem Homeassistant-Adapter in iobroker verarbeitet. Hat funktioniert.
Jetzt läuft wieder alles über den Modbus-Adapter.

UweSauter, 2024-01-15:
Hallo @skipperkm,

vielen Dank für die Register, diese funktionieren bei mir auch. Und auch vielen Dank für den Adapter insgesamt.

Hier nochmal als Codeblock für alle, die das einfach per Copy/Paste weiter verwenden wollen:
```console
_address	name	description	unit	type	len	factor	offset	formula	role	room	poll	wp	cw	isScale
97601	B1_Manufacturer			string	8	1	0				true	false	false	false
97617	B1_Model			string	16	1	0				true	false	false	false
97633	B1_Firmware			string	16	1	0				true	false	false	false
97649	B1_Serial			string	16	1	0				true	false	false	false
97665	B1_ID	Device ID		uint16be	1	1	0				true	false	false	false
97667	B1_RatedEnergy	Nennenergie	Wh	floatsw	2	1	0				true	false	false	false
97669	B1_Power_Charge_Cont_Max	Max. kontinuierliche Ladeleistung	W	floatsw	2	1	0				true	false	false	false
97671	B1_Power_Discharge_Cont_Max	Max. kontinuierliche Entladeleistung	W	floatsw	2	1	0				true	false	false	false
97673	B1_Power_Charge_Peak_Max	Max. Ladeleistung	W	floatsw	2	1	0				true	false	false	false
97675	B1_Power_Discharge_Peak_Max	Max. Entladeleistung	W	floatsw	2	1	0				true	false	false	false
97709	B1_Temp_Avg	Durchschnittstemperatur	°C	floatsw	2	1	0				true	false	false	false
97711	B1_Temp_Max	Max. Temperatur	°C	floatsw	2	1	0				true	false	false	false
97713	B1_Voltage	Aktuelle Spannung	V	floatsw	2	1	0				true	false	false	false
97715	B1_Current	Aktueller Strom	A	floatsw	2	1	0				true	false	false	false
97717	B1_Power	Aktuelle Leistung	W	floatsw	2	1	0				true	false	false	false
97719	B1_Energy_Lifetime_Export	Exportierte Gesamtenergie	Wh	uint64le	4	1	0				true	false	false	false
97723	B1_Energy_Lifetime_Import	Importierte Gesamtenergie	Wh	uint64le	4	1	0				true	false	false	false
97727	B1_Energy_Max	Max. Ladung	Wh	floatsw	2	1	0				true	false	false	false
97729	B1_Energy_Avail	Nutzbare Ladung	Wh	floatsw	2	1	0				true	false	false	false
97731	B1_SoH	Lebenszustand	%	floatsw	2	1	0				true	false	false	false
97733	B1_SoC	Ladezustand	%	floatsw	2	1	0				true	false	false	false
97735	B1_Status	Battery operating state: 0 – Off, 1 – Standby, 2 – Init, 3 – Charge, 4 – Discharge, 5 – Fault, 7 - Idle		uint32sw	2	1	0				true	false	false	false
97737	B1_StatusIntern	Vendor-defined status codes.		uint32sw	2	1	0				true	false	false	false
97739	B1_EventLog	Battery vendor's last fault ID		uint16le	1	1	0				true	false	false	false
```

> Der Adapter scheint alles richtig zu machen.

Kannst du das noch etwas näher erläutern? Wie wird die Anzahl der zu lesenden Register berechnet? Wie ich oben ja schon erwähnt habe, denke ich trotzdem noch, dass bei dem zuerst genannten Register 57600 irgendwas schief läuft, sonst würde der Adapter ja nicht versuchen, 17322 Register am Stück auslesen zu wollen. Insbesondere, da das jenseits des eingestellten Maximums, wie viele Register gleichzeitig abgefragt werden sollen, liegt.

skipperkm, 2024-01-16:
Hi,

freut mich das die Register bei dir auch passen. Das ist eine von vielen Varianten die ich im Internet gefunden und ausprobiert habe. Von der internen Abarbeitung im Adapter hab ich keine Ahnung. Ich war auch nur auf das Problem gestoßen. Bei 57600 erhalte ich nach wie vor Fehler. Das scheint bei den Entwicklern aber nicht unbedingt Priorität 1 zu sein. Es gibt wohl wichtigere Dinge zu bearbeiten.
Ich bin noch dran eine Doku vom Hersteller zu bekommen. Die Lage der Register scheint auch unterschiedlich zu sein bei den Produkten.

UweSauter, 2024-01-16:
Für alle, die dies hier finden:

Um das zu konkretisieren, bei mir handelt es sich um eine Anlage mit

- Wechselrichter: SolarEdge SE10K-RWB48
- Batterie: SolarEdge Home Battery 48V 13,8kWh (3 Module BAT-05K48)
- Backup Interface: SolarEdge Home Backup Interface, 3-phasig BI-EU3P

Die oben genannten Register 97601 bis 97739 werden gelesen. Allerdings zeigen manche von ihnen keine plausiblen Werte:
- 97711 Max. Temperatur: 0 °C
- 97713 Aktuelle Spannung: ~820 V (wahrscheinlich ist das die Seite des DC-DC-Wandlers, die dem Wechselrichter zugewandt ist)
- 97719 Exportierte Gesamtenergie: 0 Wh (wobei hier gestern noch ca. 13kWh angezeigt wurden)
- 97723 Importierte Gesamtenergie: 0 Wh
- 97735 Status: 6 (0 bis 5 und 7 sind erklärt…)

Die anderen Register scheinen plausibel.

skipperkm, 2024-01-17:
Hi,

dann hast du offensichtlich die gleiche Anlage wie ich, bis auf den Wechselrichter. Ich hab im Moment nur einen Akku, werde aber noch einen nachrüsten. 
Bekommst du alle 3 Akkus ausgelesen? Ich hab bei meinen Forschungen irgendwo gelesen, das die Solaredge Implementation nur 2 Akkus über modbus zur Verfügung stellt (datentechnisch).
Die anderen Werte sind bei mir ok. Importierte und exportierte Energie sind meiner Meinung nach Tageswerte und keine Gesamtwerte. Da wird bei mir nur was angezeigt wenn wirklich was passiert und nachts auf 0 gesetzt.

skipperkm, 2024-01-17:
@UweSauter 
beim Akku Modus 6 ist "TrickleCharge". Eine Frage noch. Ich habe festgestellt, daß mein Wechselrichter wenn es finstere Nacht und der Akku leer ist so ca. 100W zieht und im Modus "Throttled" steht. Ich bin der Meinung er müßte im Modus 2 (Nachtmodus) sein und ggf nur ca. 5W verbrauchen. Ist das bei Dir auch so ? Habe schon meinen Installateur kontaktiert, der wusste ad hoc auch nicht woran das liegt.

UweSauter, 2024-01-17:
> Bekommst du alle 3 Akkus ausgelesen?

Ich glaube, da liegt ein Missverständnis vor. Mein Wechselrichter [Datenblatt](https://knowledge-center.solaredge.com/sites/kc/files/se-solaredge-home-hub-inverter-three-phase-backup-datasheet-de.pdf) kann überhaupt nur eine Batterie verwalten. Diese kann jedoch aus bis zu fünf Modulen bestehen. Das zeigt sich dann z.B. in den Registern 97667 Nennkapazität und 97727 Max. Kapazität. Aber nicht darin, dass man jedes einzelne Modul separat auslesen kann.

> Ich habe festgestellt, daß mein Wechselrichter wenn es finstere Nacht und der Akku leer ist so ca. 100W zieht und im Modus "Throttled" steht. Ich bin der Meinung er müßte im Modus 2 (Nachtmodus) sein und ggf nur ca. 5W verbrauchen. Ist das bei Dir auch so ?

Wir hier gehen davon aus, dass das letztlich mit dem Backup Interface zu tun hat. Während ein Aufbau ohne Backup Interface nachts tatsächlich in den Tiefschlaf gehen kann und nur die Spannung der PV-Panele überwachen muss um wieder aufzuwachen kann ein Aufbau mit Backup Interface das nicht. Vermutlich weil sonst die Umschaltzeit zwischen Netzspannungsverlust und Einsetzen des Inselbetriebs viel zu lange dauern würde.
Mit Backup Interface ist der Akku ja nicht "leer" wenn man die Akkufüllung aufgebraucht ist sondern es ist immer noch die eingestellte Ladung für den Notbetrieb vorhanden.
Allerdings sind wir auch von den 100 W Dauerverbrauch schockiert.

skipperkm, 2024-01-17:
Hi,
das mit der Batterie macht Sinn. Die WR sin ja auch relativ neue Modelle. Ich glaube erst letztes Jahr auf den Markt gekommen. Hab den Beitrag noch ma rausgesucht - war irgendwie von 2019/20.

Das Thema Verbrauch habe ich auch schon in Richtung Backup überlegt - finde die Last aber auch definitiv zu hoch. Das erhöht meine Grundlast um die Hälfte des Status von vor der Installation.

skipperkm, 2024-01-17:
Ich bin gespannt welche Antwort ich von meinem Installateur bekomme

UweSauter, 2024-01-17:
Falls du noch bessere Info oder andere Register heraus bekommst machst du der Community eine Freude, wenn du sie teilst. Ich bin zwar mit den oben genannten, nicht plausiblen Werten zwar nicht zufrieden. Aber auch hier ist der Installateur schon informiert, dass etwas noch nicht richtig passt.

skipperkm, 2024-01-17:
Mach ich, würde mich ebenso freuen wenn Du etwas herausbekommst und es hier einstellst

UweSauter, 2024-04-03:
Hallo zusammen,

uns sind in letzter Zeit noch zwei Dinge aufgefallen:

1. Die Register 97719 und 97723 springen wild hin und her. Ich bezweifle, dass die oben genannte Einstellungen (4 Words = 8 Byte = 64 Bit pro Register sowie Encoding) korrekt sind. Ich vermute vielmehr, dass die diese 2x8 Byte eigentlich mehrere Datensätze enthalten und außerdem in Big Endian kodiert sind. Auswertungen dazu laufen noch.

Für die Auswertung der insgesamt 16 Bytes zwischen 97719 und 97726 habe ich jetzt für jede der acht Adressen ein eigenes Datenfeld in uint16 Big Endian angelegt und lasse die Daten in meine InfluxDB abspeichern. Damit sollte sich eigentlich auswerten lassen, ob das Encoding passt und welche Register ggf. zusammen einen Wert bilden.

2. Modbus definiert die Adressen mit 16bit, also bis max. 65535. Warum die Adressen im Bereich 97601 bis 97739 trotzdem (irgendwie) funktionieren ist mir nicht klar. Wenn ich allerdings die TCP Kommunikation mit Wireshark betrachte und dann den 16bit Wert aus Register 97719 in dem Datenstrom suche, dann sieht es für mich so aus, als wäre Register 97719 aus der ioBroker Konfiguration auf Register 57718 in den Modbus/TCP Daten gemappt. D.h. dass ein konstanter Wert von 40001 von den Adressen abgezogen wird. Die 40001 allerdings sind die Anfangsadresse des Abschnitts zu Holding Registern…

Wenn ich allerdings den Bereich 57600 bis 57738 äquivalent konfiguriere, dann hängt sich der Modbus Adapter auf…

skipperkm, 2024-04-04:
Hi,

bei mir gehen diese Register immer wieder auf 0, deswegen dachte ich der Zähler sei tagesbezogen, was allerdings für einen energiezähler keinen Sinn macht. 

Gib bitte Bescheid wenn du was rausbekommst :-)

UweSauter, 2024-04-11:
Eine Auswertung der Register 97719-97726 als (u)int64be, also 2 Register / 4 Bytes pro Wert bringt unsinnige Werte, während die Konfiguration mit uint32be sinnvoll aussieht. Allerdings ergeben nur die Register 97719 und 97723 für mich Sinn, die anderen sechs bleiben bei 0.
In 97719 wird die aus der Batterie entnommene Energie in Wh angezeigt, in 97723 die eingespeicherte. Den anderen 6 Registern kann ich keine Bedeutung zuordnen.

Bzgl. der täglichen Nullung der Zähler gehe ich davon aus, dass beim Einschalten des Wechselrichters eine interne Uhr gestartet wird, die die beiden Werte nach jeweils 24h zurücksetzt (und nicht, was sinnvoller wäre, um (lokal) Mitternacht oder 0:00 UTC). Um die Vermutung zu bestätigen muss ich allerdings noch einen Test machen und den Wechselrichter zu einer definierten Zeit aus- und wieder einschalten.

Man könnte natürlich eine eigene Auswertung über diese Register implementieren, indem man über die zeitliche Ableitung (Steigung) der Register integriert und dabei Werte für die Steigung größer oder kleiner der maximalen (Ent-)Ladeleistung ignoriert. Das müsste, wenn man InfluxDB als Datenbank verwendet, sogar als "continuous query" implementierbar sein. Aber andererseits müsste es diese Werte noch an anderer Stelle geben, die Solar Edge bisher nur noch nicht veröffentlicht hat.

skipperkm, 2024-04-11:
Meine Fresse das ist ja ein Käse. Ich kann auch nicht verstehen warum se da so ein Geheimnis draus macht. Ende letzten Jahres kam endlich mal eine aktuelle modbus-Doku raus, die aber den kompletten Batteriesektor wieder nicht beinhaltet.
Ich schau mal was die Umstellung der vartypes bringt die du erforscht hast. 
Vielen Dank schon mal vorab dafür:-)