ioBroker/ioBroker.modbus #389: Remeha Tensio C Heatpump (RTU) -> USR-TCP232-410S (MODBUS TCP Server) -> IOBroker

Lordbooker, 2023-09-29:
Habe jetzt seit einiger Zeit erfolgreich meinen SMA Wechselrichter über den Adapter am Laufen.
Leider bekomme ich meine Wärmepumpe nicht an den Start.

Ich nutze einen USR-TCP232-410S (RS232 / RS485) TCP Server, mit folgender Konfiguration:

[image]

Wenn ich mittels CAS Modbus-Scanner auf den TCP Server zugreife kann ich eine saubere Verbindung herstellen, alle benötigten Register lesen und auch schreiben.

[image]

Leider bekomme ich den Adapter in IOBroker nicht konfiguriert, dass er mir die Register ausliest.
Eine Verbindung kommt zu Stande und wird auch gehalten. Daten erhalte ich jedoch keine.

Die Konfiguration sieht wie folgt aus:

[image]

[image]

[image]

Ich habe die letzten Tage viel probiert, komme aber zu keinem logischen Ergebnis.
Hat hier vielleicht jemand eine Idee, woran es liegen kann?

Lordbooker, 2023-09-29:
Ich habe mir jetzt spaßeshalber noch eine beliebige Modbus Android App aus dem Play Store gezogen.

Auch mit der ist lesen und schreiben problemlos möglich.
Kann dann ja schonmal nicht an den Modbus Server liegen, würde ich vermuten.

Apollon77, 2023-10-02:
Closed because? Reopen if you want that someone has a look and please add a full debug log

Lordbooker, 2023-10-03:
Hey there,
i am working with "Holding Registers" for de the first time and it seems to be a little different to "Input Register".
I solved my problem, using Adresses like "12" and not "40012" an i put den flag at "Abfrage"

[image]

All my tries, i have one or the other active. That was the Problem.

But now, i have another Issue.

There are registers, that i cant read with all other Modbus Tools, but when i put them to IOBroker Adapter, the Adapter won't connect.

Example:

[image]

[image]

It takes the wrong register.
In all Tools i used, it seems to be 21.

But in the adapter there are some registers +3. That i dont understand.

In my case, i need "21".

This register has the value "45".
In all tools but IOBroker. There it will take the value of register  "18"

[image]

[image]

Now if im going to add +3 and pull register 24 i get an error in the log.

[image]

AlexanderRose82, 2025-02-07:
Just if someone wants to help to debug, this adapter is also sold as HF-5122. At least it looks like it from the pics and specs.