ioBroker/ioBroker.modbus #401: Modbus Verhalten nicht klar

kevinhaybach, 2023-10-01:
Hi,

hierbei handelt es sich um einen Doppelpost da in ioBroker / Allgemein niemand geantwortet hat.

Ich hoffe ihr könnt mir helfen das Modbus Verhalten hier zu verstehen.
Ich habe eine Modbus Variable (HoldingRegisters) die gelesen und geschrieben werden soll.
Ich schreibe diese Mittels setState wenn ein Ereignis eintritt, gleichzeitig überwache ich diese Modbus-Variable mit on.
Ein Minimalbeispiel sieht wie folgt aus:

`
on({id: "0_userdata.0.Test.MyBool", change: 'ne'}, function(obj)
 {
 console.log("0_userdata.0.Test.MyBool");
 console.log(getState("0_userdata.0.Test.MyBool"));
 if (getState("0_userdata.0.Test.MyBool").val)
 {
     setState("modbus.1.holdingRegisters.32114_Test", 1);
 }
 else
 {
     setState("modbus.1.holdingRegisters.32114_Test", 0);
  }
  });
    
    
  on({id: "modbus.1.holdingRegisters.32114_Test", change: 'ne'}, function(obj)
  {
      console.log("modbus.1.holdingRegisters.32114_Test");
      console.log(getState("modbus.1.holdingRegisters.32114_Test"));
  });

Die Logausgaben sind wie folgt:
`
2023-09-28 15:33:59.210  - [32minfo[39m: javascript.0 (4971) script.js.common.TestModbus: 0_userdata.0.Test.MyBool
2023-09-28 15:33:59.211  - [32minfo[39m: javascript.0 (4971) script.js.common.TestModbus: {'val':false,'ack':false,'ts':1695908039206,'q':0,'from':'system.adapter.admin.0','user':'system.user.admin','lc':1695908039206}
2023-09-28 15:33:59.224  - [32minfo[39m: javascript.0 (4971) script.js.common.TestModbus: modbus.1.holdingRegisters.32114_Test
2023-09-28 15:33:59.225  - [32minfo[39m: javascript.0 (4971) script.js.common.TestModbus: {'val':0,'ack':false,'ts':1695908039212,'q':0,'c':'script.js.common.TestModbus','from':'system.adapter.javascript.0','user':'system.user.admin','lc':1695908039212}
2023-09-28 15:33:59.380  - [32minfo[39m: javascript.0 (4971) script.js.common.TestModbus: modbus.1.holdingRegisters.32114_Test
2023-09-28 15:33:59.381  - [32minfo[39m: javascript.0 (4971) script.js.common.TestModbus: {'val':1,'ack':true,'ts':1695908039377,'q':0,'from':'system.adapter.modbus.1','user':'system.user.admin','lc':1695908039377}
2023-09-28 15:33:59.427  - [32minfo[39m: javascript.0 (4971) script.js.common.TestModbus: modbus.1.holdingRegisters.32114_Test
2023-09-28 15:33:59.428  - [32minfo[39m: javascript.0 (4971) script.js.common.TestModbus: {'val':0,'ack':true,'ts':1695908039423,'q':0,'from':'system.adapter.modbus.1','user':'system.user.admin','lc':1695908039423}
`

Zeile 6 vom Log verstehe ich nicht. Wieso ist der Value wieder kurzzeitig auf 1?

Mit freundlichen Grüßen
Kevin Haybach

kevinhaybach, 2023-10-01:
Hier ein Link zum Initialen Post, hier funktioniert es anscheinend mit dem Code-Snippet nicht
https://forum.iobroker.net/topic/68680/modbus-verhalten-nicht-klar?_=1696183304310

kevinhaybach, 2023-10-01:
Ich habe mittlerweile rausgefunden das es am Zusammenspiel zwischen der aktuellen PollTime und dem Data Polling interval liegt. Ich hatte das Data Polling interval auf 200ms eingestellt und meine aktuelle PollTime betrug 270ms.

Apollon77, 2023-10-02:
Bitte "ack" flag beachten, voin daher scheint sich hier deine Önderung mit dem polling überschnitten zu haben.

Du sagst "setze auf 0", der adapter verarbeitet das - währenddessen kommt nochmal eine 1 als wert rein, direkt danach dann die besätigung der durchführung. Das ganze wäre mit einem parallelen log vom modbus Adapter zeitlich klar geworden.

closed 2023-10-02 by Apollon77