ioBroker/ioBroker.modbus #403: Read and Write of holding register with different adress

cdaerr, 2023-10-09:
Hi,

my Solar Inverter has different adress for read and write to holding registers. I configured the language paramter for example:
[image]

Now when I change the Language_r datapoint, the modbus Adapter writes to 40188 and changes a different paramter.

Is there a way to handle this kind of configuration properly?

Thanks

Apollon77, 2023-10-10:
> Now when I change the Language_r datapoint,

Why you change it when you know that it shoukd not be written :)

cdaerr, 2023-10-10:
Not with intent of course. This can happen accidently and should be prohibited. 
Setting write flag to false has no effect. Writing to is possible
A nice solution would be different Modbus write and read adresses at instance configuration.

leonundjulie, 2024-01-02:
This threat might answer my problem uploaded in issue https://github.com/ioBroker/ioBroker.modbus/issues/433. **Where is the WRITE Flag**?