ioBroker/ioBroker.modbus #413: Update stable version in repo from 5.0.11 to 6.0.1

GermanBluefox, 2023-11-15:
# Think about update stable version to 6.0.1
**Version**: stable=**5.0.11** (348 days old) => latest=**6.0.1** (15 days old)
**Installs**: stable=**11710** (82.96%), latest=**801** (5.67%), total=**14116**

Click to use [developer portal](https://www.iobroker.dev/adapter/modbus/releases)
Click to [edit](https://github.com/ioBroker/ioBroker.repositories/edit/master/sources-dist-stable.json#L1588)

Note: This is an automatically generated message and not personally authored by bluefox!

mcm1957, 2023-12-04:
@Apollon77 
@GermanBluefox 
Is there any reason to delay an update to stable any longer?

**If there is no response until 15.12.2023 an update could be performed anyway.**

mcm1957, 2023-12-08:
a kind reminder

closed 2023-12-21 by GermanBluefox