ioBroker/ioBroker.modbus #414: faktor 0.1/0,1  is not scaling down

HarryHpd, 2023-11-15:
I am using mdbus adapter in slave mode. Master is "Technische Alternative - CMI".  For scaling down imported values I use factor 0.1. 
But instead of scaling down by factor 10,  modbus instance shows values multiplied by 10.  trying decimal commas instead of decimal point makes no difference.
Trying to solve this issue via formulas is not possible neither. It appears to me, formulas are only available for master instances. This appears to me to be strange as well.

**To Reproduce**  
1. use modbus slave mode instance, none of the check boxes are ticked in "Allgemein" 
2. create register entry for holding register with factor 0.1  or 0,1   / signed 16 Bit Big endian, only "Abfrage" Checkbox is ticked
3. wait for incomming data

**Expected behavior**  
I expect incomming values to be scaled down

**Screenshots & Logfiles**  
If applicable, add screenshots and logfiles to help explain your problem.

**Versions:**  
 Installierte Instanzen:
3
Verfügbare Version:
5.0.11
Installierte Version:
5.0.11

Apollon77, 2023-11-15:
please provide a debiug log. 0.1 should work normally ...

HarryHpd, 2023-11-15:
I hope this is what you want to see. 

Adress Nr 4 is the one you probably want to see.

<html><body>

2023-11-15 12:19:32.450  - warn: modbus.0 (915) FC4 request outside inputs registers boundaries: address 30, len 1. Expected from 0 to 4
--
2023-11-15 12:19:32.450  - debug: modbus.0 (915) sending tcp data
2023-11-15 12:19:37.450  - debug: modbus.0 (915) handling write single register request.
2023-11-15 12:19:37.452  - debug: modbus.0 (915) sending tcp data
2023-11-15 12:19:40.925  - info: host.iobroker stopInstance system.adapter.modbus.0 (force=false, process=true)
2023-11-15 12:19:40.928  - info: modbus.0 (915) Got terminate signal TERMINATE_YOURSELF
2023-11-15 12:19:40.930  - info: modbus.0 (915) terminating
2023-11-15 12:19:40.930  - debug: modbus.0 (915) Plugin sentry destroyed
2023-11-15 12:19:40.930  - info: modbus.0 (915) Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
2023-11-15 12:19:40.971  - info: host.iobroker stopInstance system.adapter.modbus.0 send kill signal
2023-11-15 12:19:41.432  - info: modbus.0 (915) terminating
2023-11-15 12:19:41.463  - info: host.iobroker instance system.adapter.modbus.0 terminated with code 11 (ADAPTER_REQUESTED_TERMINATION)
2023-11-15 12:19:44.038  - info: host.iobroker instance system.adapter.modbus.0 started with pid 930
2023-11-15 12:19:44.767  - debug: modbus.0 (930) Redis Objects: Use Redis connection: 127.0.0.1:9001
2023-11-15 12:19:44.789  - debug: modbus.0 (930) Objects client ready ... initialize now
2023-11-15 12:19:44.791  - debug: modbus.0 (930) Objects create System PubSub Client
2023-11-15 12:19:44.791  - debug: modbus.0 (930) Objects create User PubSub Client
2023-11-15 12:19:44.801  - debug: modbus.0 (930) Objects client initialize lua scripts
2023-11-15 12:19:44.805  - debug: modbus.0 (930) Objects connected to redis: 127.0.0.1:9001
2023-11-15 12:19:44.814  - debug: modbus.0 (930) Redis States: Use Redis connection: 127.0.0.1:9000
2023-11-15 12:19:44.819  - debug: modbus.0 (930) States create System PubSub Client
2023-11-15 12:19:44.820  - debug: modbus.0 (930) States create User PubSub Client
2023-11-15 12:19:44.867  - debug: modbus.0 (930) States connected to redis: 127.0.0.1:9000
2023-11-15 12:19:44.887  - debug: modbus.0 (930) Plugin sentry Initialize Plugin (enabled=true)
2023-11-15 12:19:45.092  - info: modbus.0 (930) starting. Version 5.0.11 in /opt/iobroker/node_modules/iobroker.modbus, node: v18.18.2, js-controller: 5.0.12
2023-11-15 12:19:45.172  - debug: modbus.0 (930) Iterate inputRegs inputRegisters: 0 = inputRegisters.3_test
2023-11-15 12:19:45.173  - debug: modbus.0 (930) Iterate holdingRegs holdingRegisters: 0 = holdingRegisters.2_0-10v
2023-11-15 12:19:45.173  - debug: modbus.0 (930) Iterate holdingRegs holdingRegisters: 1 = holdingRegisters.3_EZ3-Leistung
2023-11-15 12:19:45.173  - debug: modbus.0 (930) Iterate holdingRegs holdingRegisters: 2 = holdingRegisters.4_EZ3-Tagesleistung
2023-11-15 12:19:45.174  - debug: modbus.0 (930) Iterate holdingRegs holdingRegisters: 3 = holdingRegisters.5
2023-11-15 12:19:45.174  - debug: modbus.0 (930) Initialize Objects for disInputs: []
2023-11-15 12:19:45.174  - debug: modbus.0 (930) Initialize Objects for coils: []
2023-11-15 12:19:45.175  - debug:  modbus.0 (930) Initialize Objects for inputRegs:  [{"_address":"3","name":"test","description":"test","unit":"keh","type":"int16be","len":1,"factor":1,"offset":0.1,"formula":"","role":"value.valve","room":"","cw":false,"isScale":"","address":3,"deviceId":2,"id":"inputRegisters.3_test"}]
2023-11-15 12:19:45.175  - debug:  modbus.0 (930) Add inputRegisters.3_test:  {"_id":"inputRegisters.3_test","type":"state","common":{"name":"test","role":"value.valve","type":"number","read":true,"write":true,"def":0,"unit":"keh"},"native":{"regType":"inputRegs","address":3,"deviceId":2,"type":"int16be","len":1,"offset":0.1,"factor":1}}
2023-11-15 12:19:45.175  - debug:  modbus.0 (930) Initialize Objects for holdingRegs:  [{"_address":2,"name":"0-10v","description":"0-10vB","unit":"","type":"int16be","len":1,"factor":1,"offset":0,"formula":"","role":"level","room":"","poll":true,"wp":false,"cw":false,"isScale":false,"deviceId":2,"address":2,"id":"holdingRegisters.2_0-10v"},{"_address":3,"name":"EZ3-Leistung","description":"EZ3-Elektr.   Leistung","unit":"Watt","type":"int16be","len":1,"factor":1,"offset":0,"formula":"x*12","role":"level","room":"","poll":true,"wp":false,"cw":false,"isScale":false,"deviceId":2,"address":3,"id":"holdingRegisters.3_EZ3-Leistung"},{"_address":4,"name":"EZ3-Tagesleistung","description":"EZ3-Tagesleistung","unit":"KWh","type":"int16be","len":1,"factor":0.1,"offset":0,"formula":"","role":"level","room":"","poll":true,"wp":false,"cw":false,"isScale":false,"deviceId":2,"address":4,"id":"holdingRegisters.4_EZ3-Tagesleistung"},{"_address":5,"name":"","description":"","unit":"","type":"int16be","len":1,"factor":1,"offset":0,"formula":"","role":"level","room":"","poll":true,"wp":false,"cw":false,"isScale":false,"deviceId":2,"address":5,"id":"holdingRegisters.5"}]
2023-11-15 12:19:45.175  - debug:  modbus.0 (930) Add holdingRegisters.2_0-10v:  {"_id":"holdingRegisters.2_0-10v","type":"state","common":{"name":"0-10vB","role":"level","type":"number","read":true,"write":true,"def":0,"unit":""},"native":{"regType":"holdingRegs","address":2,"deviceId":2,"type":"int16be","len":1,"offset":0,"factor":1,"poll":true}}
[... cut ...]
2023-11-15 12:20:23.457  - debug: modbus.0 (930) sending tcp data
2023-11-15 12:20:32.450  - warn: modbus.0 (930) FC4 request outside inputs registers boundaries: address 30, len 1. Expected from 0 to 4
2023-11-15 12:20:32.451  - debug: modbus.0 (930) sending tcp data
2023-11-15 12:20:37.449  - debug: modbus.0 (930) handling write single register request.
2023-11-15 12:20:37.450  - debug: modbus.0 (930) sending tcp data
2023-11-15 12:20:40.449  - debug: modbus.0 (930) handling write single register request.

</body>
</html>

HarryHpd, 2023-11-25:
I would still prefere a proper solution to the problem reported. Is this issue still under investigation?

Apollon77, 2023-11-25:
... sure, as soon as someone has time to look into the code for the reason