ioBroker/ioBroker.modbus #443: Einstellungen werden nicht angezeigt

Forestarius, 2024-01-02:
Der Adapter läuft schon lange ohne Probleme

Jedoch habe ich gerade festgestellt, dass ich keine Einstellungen mehr sehe.

Iobroker, gehe auf Instanzen, dann auf Modbusadapter, dort auf Einstellungen und bekomme nichts mehr angezeigt.

[image]

Apollon77, 2024-01-02:
Is the JavaScritp error console showing any error or logs?

Forestarius, 2024-01-03:
Wo finde ich die Console? Unter den Punkt Protokoll laufen keine Fehler auf.

Apollon77, 2024-01-03:
Im Browder meistens Web-Entwickler Tools oder so

Forestarius, 2024-01-04:
Danke :-) 

Das steht in der Console
[image]

Apollon77, 2024-01-04:
Hm ... an nsich nichts relevantes. Komisch. Mach mal ein neues "iob upload modbus"

Forestarius, 2024-01-04:
Hi, habe die Version 6.0.0 installiert und danach wieder die 6.0.1 drüber und es klappt. Lag dann wohl an meinem System

closed 2024-01-04 by Forestarius