ioBroker/ioBroker.modbus #444: Probleme EW11 und IOBroker

Sefina-DS, 2024-01-07:
Hi zusammen,

ich habe einen Deye SUN 12 K und an diesen ist ein EW11 (Elfin) angebunden.
Dieser ist als Server eingestellt.

Die Register habe ich eingerichtet.

Ich bekomme die Daten aus dem Inverter, sodass die Kommunikation funktioniert.

Ich bekomme mehrmals in der Minute Verbindungsabbrüche, sodass das gesamte Log damit voll ist.

Hat, wer 'ne Idee, was ich noch ändern kann?

[image]
[image]
[image]
[image]
[image]

mymeyer, 2024-01-16:
Hallo, ich habe das selbe Problem. Bei mir ist es so das alle 3 Sekunden ein Connect/Disconnect kommt ! Kann mir nicht vorstellen das dieses Verhalten normal sein.

alfiriel, 2024-10-31:
Hallo, gibt es hier ein Update - ich habe das gleiche Problem